OSSystems/meta-browser #39: Libav dependency

mdavis777, 2017-05-15:
I am wondering about the libav dependency for chromium.  I don't think it is required anymore and I have a patch(0dff75b038a246becbb127df61b2e4771075fe5c) to remove it, but I wanted to make sure I wasn't missing anything.

Chromium has had its own copy of ffmpeg since version 44.  
I have built it for x86 and x86_64 with libav removed and have seen no issues.

otavio, 2017-05-15:
I think you are right. One question though is if it is possible to make it use the system one as it reduces the build time and easy security maintenance.

Could you check it?

mdavis777, 2017-05-15:
There is a lot of talk from old bug tickets and mailing lists of it getting support for system ffmpeg.  Then shortly after it breaks again and then comes back. Most recent one I could find is it being fixed in Chrome 56.

The gyp define for it is use_system_ffmpeg.

otavio, 2017-05-15:
In fact it is a problem as you will be using a commercial license without knowing it.

mdavis777, 2017-05-15:
It shouldn't be including any of the commercial portions unless asked to.
There is a build flag in the chromium recipe that is off by default to handle that:
# * proprietary-codecs: (off by default)
#       Enable this option if you want to add support for additional proprietary
#       codecs, most notably MPEG standards (h.264, h.265, MP4, MP3, AAC, MPEG-2 ..)

otavio, 2017-05-15:
Yes but in OE case it means it must also whitelist libav.

If we can force it to use libav from system I'd prefer.

mdavis777, 2017-05-16:
I gave it a test build.  It doesn't appear that Chrome 54 is honoring the use_system_ffmpeg.  It just builds and uses the internal one regardless.  Perhaps after an upgrade to 56+ a PACKAGE_CONFIG can be added to set / not set system ffmpeg.  

Do we want to remove unnecessary libav dependency in the meantime?

otavio, 2017-05-16:
I think it is fine to drop it for now.

otavio, 2017-05-16:
Fixed in #40

closed 2017-05-16 by otavio