OSSystems/meta-browser #42: libxss

denyste, 2017-05-22:
Hi,
I building recipes for Chromium on Yocto Toaster Morty
 2.2.1,  see the message below :
Required build target 'rpi-basic-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['rpi-basic-image', 'libxss']

There is no "libxss" in OpenEmbedded Layer Index:
libxss, no matching recipes in database.

Why is it integrated in this recipe? 
DEPENDS += "xextproto gtk+ libxi libxss"

otavio, 2017-05-22:
This is not related to our layer; check with the RPi maintainer please.

closed 2017-05-22 by otavio

denyste, 2017-05-22:
Ok! I found libxss in the layer "meta-debian", branch Jethro, but currently it is not compatible with the branch Morty.
Thanks.