OSSystems/meta-browser #56: Issue using layer out-of-box for i.MX6

thsle, 2017-07-25:
Hi everyone,

This is not properly speaking an issue on the meta-browser but still I think here is a suitable place to get an answer from people who master it. I am (simply) trying to build a YOCTO image for my NXP Sabre i.MX6QP SD board, that includes Chromium. Here is what I do (from scratch and based on NXP Yocto User's Guide):

```
mkdir fsl-release-bsp_imx-4.1.15-2.1.0_ga/
cd fsl-release-bsp_imx-4.1.15-2.1.0_ga/
repo init -u git://git.freescale.com/imx/fsl-arm-yocto-bsp.git -b imx-4.1-krogoth -m imx-4.1.15-2.1.0.xml
repo sync
DISTRO=fsl-imx-x11 MACHINE=imx6qpsabresd source fsl-setup-release.sh -b build-x11
# add the following two lines in conf/local.conf
# CORE_IMAGE_EXTRA_INSTALL += " chromium libexif"
# LICENSE_FLAGS_WHITELIST = "commercial"
bitbake fsl-image-gui
```

This fails with the following messages:

```
ERROR: Nothing RPROVIDES 'chromium' (but /data/i.MX6_refs/fsl-release-bsp_imx-4.1.15-2.1.0_ga/sources/meta-fsl-bsp-release/imx/meta-sdk/recipes-fsl/images/fsl-image-gui.bb RDEPENDS on or otherwise requires it)
ERROR: chromium was skipped: Recipe is blacklisted: BROKEN: fails to build with gcc-6
NOTE: Runtime target 'chromium' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['chromium']
ERROR: Required build target 'fsl-image-gui' has no buildable providers.
Missing or unbuildable dependency chain was: ['fsl-image-gui', 'chromium']
```

However, `/data/i.MX6_refs/fsl-release-bsp_imx-4.1.15-2.1.0_ga/sources/meta-browser` exists with content of the repo. This is a kind of dead-end to me as asking support from my reseller or NXP, the answer was "we do no support meta-browser layer". Additionally, I could not find any useful help on the web (most of the time similar errors seem to be related to the commercial license I already put on whitelist).

Note that without modifying the `conf/local.conf` file, I produce a working image... but without Chromium. So I guess my procedure is fine, only the way I add the meta-browser layer and chromium feature is wrong.

Can you provide any help? What am I missing in the 6 above command lines?

Thanks,

otavio, 2017-07-25:
meta-freescale's Chromium patches are outdated regarding newer releases. I think if you use krogoth branch of meta-browser it should work.

Nobody spent time to fix it yet and at O.S. Systems we have been busy working on other stuff so until a customer demands it for us it is not something we intend to work for now...

thsle, 2017-07-25:
Hello @otavio, I do not get what you mean by "meta-freescale's Chromium patches"? I don't see the link with meta-freescale. Can you be more precise?
However, I will give a try forcing krogoth branch instead of master in my 4.1.15 environment and provide a feedback here.
Thanks for you quick reply.

otavio, 2017-07-31:
> Hello @otavio, I do not get what you mean by "meta-freescale's Chromium patches"? I don't see the link with meta-freescale. Can you be more precise?

The meta-freescale BSP adds few more patches to enable the i.MX to work with Chromium.

> However, I will give a try forcing krogoth branch instead of master in my 4.1.15 environment and provide a feedback here.

Good.

closed 2017-07-31 by otavio