OSSystems/meta-browser #59: No Pyro branch

embexus, 2017-08-29:
Hello,

which branch of meta-browser is compatible with Yocto Pyro ?

Regards

otavio, 2017-08-29:
@embexus you can use the master branch. As the man power to maintain the layer is small we only branch when it becomes incompatible somehow. Up to now, master works on both pyro and master branches.

closed 2017-08-29 by otavio

embexus, 2017-08-30:
There are missmachtes between the pyro branch of meta-freescale and master branc of meta-browser :
ERROR: No recipes available for:
DIR/meta-freescale/dynamic-layers/browser-layer/recipes-browser/chromium/chromium-wayland_48.0.2548.0.bbappend
 DIR/meta-freescale/dynamic-layers/browser-layer/recipes-browser/chromium/chromium_52.0.2743.76.bbappend

otavio, 2017-08-30:
This is on meta-freescale. Someone need to send the upgrade patches for it as meta-freescale didn't upgrade the bbappend patches for a while and if it is not fixed I will end removing them there.

embexus, 2017-08-30:
It makes sens.
Thank you Otavio.