OSSystems/meta-browser #76: Splitting the original recipe to support more targets and upstreaming Chromium/Wayland recipe.

msisov, 2017-11-27:
Hi!

As you have already known, Igalia has been working on the Chromium browser to add Wayland support utilizing using the newest mus/ozone (ui service) efforts being done in the Chromium project's upstream repositories. At the moment, the upstream version of Chromium supports Wayland on ChromeOS only (so called internal window mode, when mus works with ash window system, which plays a role of a window manager).

The project to run Chromium/mus/Wayland on Linux is currently being done in the downstream branch called ozone-wayland-dev (https://github.com/Igalia/chromium/). At the moment, we met a phase when the Chromium with Wayland support is stable enough (it is mostly affected with the upstream bugs rather than the implementation we have nowadays) and as close to the stock chromium as possible. 

During this project, we have been closely collaborating with Google. Thus, the project to have Chromium with Wayland on Linux is aligned with the plans Google has for the new UI service and Wayland support. We have also started to upstream our patches to the Chromium project.

We have also been updating our meta-browser fork to make the browser available for the AGL system (https://www.automotivelinux.org/). Recently, we have synced our fork and updated the recipe in such a way that it can support several recipe targets, and shared a common part of the recipe between different "subrecipes". You can check it here - https://github.com/Igalia/meta-browser/tree/recipe-with-downstream-sources. The recipe has been tested on a Renesas Rcar m3 board and compiled for the rpi3 target, but not tested yet (we have our internal recipe tested on rpi2 and rpi3, which is very similar to the one we have in GitHub now).

Our idea is to upstream the recipe we have and keep stock Chromium support and Chromium with Wayland support at the same time making it available for meta-browser users.

What do you think about the recipe we have and how do you asses the idea to upstream it?

Best regards,
Maksim Sisov
Igalia S.L.

otavio, 2017-11-27:
Hello Maksim,

On Mon, Nov 27, 2017 at 7:26 AM, Maksim Sisov <notifications@github.com>
wrote:
>
> What do you think about the recipe we have and how do you asses the idea
> to upstream it?
>
Absolutely.

Please prepare a pull request for review. I did a brief look at your
commits and it does not to follow the OE commit guidelines, so please pay
attention to it as well :-)

Thanks for all this work :-)

-- 
Otavio Salvador                             O.S. Systems
http://www.ossystems.com.br        http://code.ossystems.com.br
Mobile: +55 (53) 9981-7854            Mobile: +1 (347) 903-9750

embexus, 2017-11-29:
This PR is breaking the build !!!  

ERROR: chromium-x11-62.0.3202.94-r0 do_package: QA Issue: chromium-x11: Files/directories were installed but not shipped in any package:
  /usr/lib/chromium/natives_blob.bin
  /usr/lib/chromium/resources.pak
  /usr/lib/chromium/chrome_100_percent.pak
  /usr/lib/chromium/chromium-bin
  /usr/lib/chromium/chrome-sandbox
  /usr/lib/chromium/chrome_200_percent.pak
  /usr/lib/chromium/snapshot_blob.bin
  /usr/lib/chromium/icudtl.dat
  /usr/lib/chromium/chromium-wrapper
  /usr/lib/chromium/locales
  /usr/lib/chromium/locales/tr.pak
  /usr/lib/chromium/locales/hi.pak
  /usr/lib/chromium/locales/de.pak
  /usr/lib/chromium/locales/et.pak
  /usr/lib/chromium/locales/ru.pak
  /usr/lib/chromium/locales/lt.pak
  /usr/lib/chromium/locales/am.pak
  /usr/lib/chromium/locales/fr.pak
  /usr/lib/chromium/locales/zh-CN.pak
  /usr/lib/chromium/locales/lv.pak
  /usr/lib/chromium/locales/sr.pak
  /usr/lib/chromium/locales/ko.pak
  /usr/lib/chromium/locales/es-419.pak
  /usr/lib/chromium/locales/ms.pak
  /usr/lib/chromium/locales/ml.pak
  /usr/lib/chromium/locales/hu.pak
  /usr/lib/chromium/locales/el.pak
  /usr/lib/chromium/locales/vi.pak
  /usr/lib/chromium/locales/fa.pak
  /usr/lib/chromium/locales/gu.pak
  /usr/lib/chromium/locales/pt-BR.pak
  /usr/lib/chromium/locales/mr.pak
  /usr/lib/chromium/locales/te.pak
  /usr/lib/chromium/locales/cs.pak
  /usr/lib/chromium/locales/ta.pak
  /usr/lib/chromium/locales/sw.pak
  /usr/lib/chromium/locales/kn.pak
  /usr/lib/chromium/locales/he.pak
  /usr/lib/chromium/locales/bn.pak
  /usr/lib/chromium/locales/sl.pak
  /usr/lib/chromium/locales/id.pak
  /usr/lib/chromium/locales/fil.pak
  /usr/lib/chromium/locales/en-US.pak
  /usr/lib/chromium/locales/sv.pak
  /usr/lib/chromium/locales/th.pak
  /usr/lib/chromium/locales/bg.pak
  /usr/lib/chromium/locales/ro.pak
  /usr/lib/chromium/locales/da.pak
  /usr/lib/chromium/locales/ar.pak
  /usr/lib/chromium/locales/pt-PT.pak
  /usr/lib/chromium/locales/ja.pak
  /usr/lib/chromium/locales/es.pak
  /usr/lib/chromium/locales/nl.pak
  /usr/lib/chromium/locales/pl.pak
  /usr/lib/chromium/locales/ca.pak
  /usr/lib/chromium/locales/hr.pak
  /usr/lib/chromium/locales/sk.pak
  /usr/lib/chromium/locales/it.pak
  /usr/lib/chromium/locales/fi.pak
  /usr/lib/chromium/locales/en-GB.pak
  /usr/lib/chromium/locales/nb.pak
  /usr/lib/chromium/locales/zh-TW.pak
  /usr/lib/chromium/locales/uk.pak
  /usr/bin/chromium
  /usr/share/applications
  /usr/share/applications/chromium.desktop
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
chromium-x11: 66 installed and not shipped files. [installed-vs-shipped]
ERROR: chromium-x11-62.0.3202.94-r0 do_package: Fatal QA errors found, failing task.
ERROR: chromium-x11-62.0.3202.94-r0 do_package: Function failed: do_package
ERROR: Logfile of failure stored in: /var/lib/jenkins/workspace/beld-os/beld/build/tmp/work/armv7ahf-neon-imx-beld-linux-gnueabi/chromium-x11/62.0.3202.94-r0/temp/log.do_package.19365
NOTE: recipe chromium-x11-62.0.3202.94-r0: task do_package: Failed
ERROR: Task (/var/lib/jenkins/workspace/beld-os/beld/meta-browser/recipes-browser/chromium/chromium-x11_62.0.3202.94.bb:do_package) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6651 tasks of which 6633 didn't need to be rerun and 1 failed.

otavio, 2017-11-29:
I think the packaging will need adjustment. @msisov  can you prepare the fix for it?

msisov, 2017-11-30:
@embexus, @otavio, I'm sorry, this patch was on the top of my local branch patches and I forgot to include it to the previous PR.

msisov, 2017-11-30:
Thanks for reviewing the recipes. We will be updating the recipe regularly.

closed 2017-11-30 by msisov

otavio, 2017-11-30:
Thanks for your work; please let's know if there is anything pending :-)