OSSystems/meta-browser #80: Missing Rocko branch

mellowcandle, 2017-12-18:
Release is out, I think it's time to create a branch here.

otavio, 2017-12-18:
Until we have an incompatible change, which makes it fail to build on rocko, we are not intending to branch as it will increase the maintenance work.

closed 2017-12-18 by otavio

mellowcandle, 2018-01-15:
I think we've reached that point:
`| DEBUG:root:Running: /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/build/write_buildflag_header.py --output base/allocator/features.h --gen-dir /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/out_bootstrap/gen --definitions /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/out_bootstrap/gen/base/allocator/features.h.tmp
| DEBUG:root:Running: /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/build/write_buildflag_header.py --output base/debug/debugging_flags.h --gen-dir /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/out_bootstrap/gen --definitions /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/out_bootstrap/gen/base/debug/debugging_flags.h.tmp
| DEBUG:root:Running: /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/build/write_build_date_header.py /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/out_bootstrap/gen/base/generated_build_date.h default
| DEBUG:root:Running: ninja -C /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/out_bootstrap -w dupbuild=err -v gn
| ninja: Entering directory `/local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/out_bootstrap'
| [1/363] g++  -MMD -MF base/at_exit.o.d  -I/local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/out_bootstrap/gen -I/local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132 -isystem/local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/recipe-sysroot-native/usr/include -O2 -pipe -DNO_TCMALLOC -D__STDC_FORMAT_MACROS -O2 -g0 -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -isystem/local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/recipe-sysroot-native/usr/include -O2 -pipe -std=c++14 -Wno-c++11-narrowing -c /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/base/at_exit.cc -o base/at_exit.o
| FAILED: base/at_exit.o
| g++  -MMD -MF base/at_exit.o.d  -I/local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/out_bootstrap/gen -I/local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132 -isystem/local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/recipe-sysroot-native/usr/include -O2 -pipe -DNO_TCMALLOC -D__STDC_FORMAT_MACROS -O2 -g0 -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -isystem/local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/recipe-sysroot-native/usr/include -O2 -pipe -std=c++14 -Wno-c++11-narrowing -c /local2/workspace/rocko2/build/tmp-glibc/work/x86_64-linux/gn-native/63.0.3239.132-r0/chromium-63.0.3239.132/base/at_exit.cc -o base/at_exit.o
| g++: error: unrecognized command line option ‘-std=c++14’
`

otavio, 2018-01-15:
This is the `gn-native` package that uses the host `gcc` for the build. What distro are you using?

mellowcandle, 2018-01-15:
Ubuntu 14.04

rakuco, 2018-01-15:
That's similar to the problem @twoerner had with openSUSE. You basically need to install a newer compiler; having separate branches won't help here as @otavio pointed out.