OSSystems/meta-browser #81: chromium fails for

schnitzeltony, 2017-12-21:
Have tested chromium-x11 with quite recent master layers. It was build with

PACKAGECONFIG = "use-egl impl-side-painting proprietary-codecs"

for Raspi3.

On a fresh image chromium starts but as soon as I open e.g https://github.com it crashes. All further starts crash immediately. If I delete $HOME/.config/chromium I can start chromium again but moving to https://github.com crashes.

So I started chromium by console and get:

> [2263:2263:1221/015335.950665:ERROR:gl_implementation.cc(241)] Failed to load /usr/lib/chromium/libGLESv2.so: /usr/lib/chromium/libGLESv2.so: cannot open shared object file: No such file or directory
> [2263:2263:1221/015335.957888:ERROR:gpu_child_thread.cc(254)] Exiting GPU process due to errors during initialization
> [2213:2247:1221/015335.971247:ERROR:browser_gpu_channel_host_factory.cc(108)] Failed to launch GPU process.
> Bus error (core dumped)

I did some investigations:
* On Target: Creating a link /usr/lib/chromium/libGLESv2.so to /usr/lib//libGLESv2.so.2 did not change anything
* In OE-Workdir: Have no idea if this is something of importance but I found many entries like 'solibs = ./libEGL.so ./libGLESv2.so' in the created ninja files. The error message path is libexec / /usr/lib/chromium so this might cause chromium look there.

twoerner, 2017-12-21:
When you build chromium-x11 for the rpi3, when done you'll find the two files libGLESv2.so and libEGL.so left over in the build directory. If you install them to ${D}${libdir}/chromium you'll find that chromium now uses gpu acceleration when run. Not sure why this isn't the default. I.e.

do_install_append() {
    install libGLESv2.so ${D}${libdir}/chromium
    install libEGL.so ${D}${libdir}/chromium
}

I don't think I'm using anything other than the default PACKAGECONFIGs when I build.

msisov, 2017-12-21:
I guess PACKAGECONFIG must be looking the following
PACKAGECONFIG_pn-chromium-x11 = "use-egl impl-side-painting proprietary-codecs"

and then it will work. At least, I always specify.

schnitzeltony, 2017-12-21:
@msisov / @twoerner: Thanks for quick response

@msisov: I do this in an chromium%.bbappend file in my pseudo distro layer meta-misc. I think there the syntax should be ok.

@twoerner: I will give this a try - building will take some time...
Thinking of this some questions come up: 
* does that mean chromium builds it's own EGL/GLES2?
* Is that Raspi specific or is it necessary for other machines too? If so: does it make sense to extend recipe to install these files by default?

twoerner, 2017-12-21:
It must be building its own EGL/GLES2 since those files show up after a build in the out/Release folder. Plus, if I build EGL/GLES2 as part of my image and try to copy those files into /usr/lib/chromium, chromium complains about some symbols etc and doesn't show acceleration at chromium://gpu.

My only focus with chromium is on rpi3 (32, so far). I haven't built for any other target lately. Before sending a patch I wanted to prove that chromium is actually GPU-accelerated by running some benchmark just to make sure the chromium://gpu screen was actually telling the truth.

otavio, 2017-12-21:
@schnitzeltony can you open a pull request with the fix?

schnitzeltony, 2017-12-21:
@otavio: Will do - or @twoerner do you want to?

But before sending out I would like to get it to work:
Situation got better slightly: The error messages are gone / and 2nd start works now. As soon as I open https://github.com I still get 'Bus error' crash. Will try a remote debug session tonight (hope it still works with oe-core recipe specific sysroot...) and see what causes this crash.

Will come back then...

schnitzeltony, 2017-12-21:
Aarg - chromium strips before packing - yet another challenge...

twoerner, 2017-12-21:
@schnitzeltony go ahead, you're already there

schnitzeltony, 2017-12-21:
@otavio / @twoerner: Pull request is out - should do no harm as is

msisov, 2017-12-21:
including vc4graphics into your image may resolve the issue. At least, we don't face it with ToT chromium build for rpi.

schnitzeltony, 2017-12-22:
> Aarg - chromium strips before packing - yet another challenge...

It's not exactly strip. Hmm - the more I think of this extra build of EGL/GLES I get the feeling that chromium builds way too much. I've seen this before [1] and it bites me again now: Current solution is removing -g in https://github.com/OSSystems/meta-browser/blob/ab2c3ad13a69a438851b9f06e01e6bb4d5f2a8e3/recipes-browser/chromium/chromium-gn.inc#L120 which makes remote debugging useless - no symbols... Don't remember exactly but in thread [1] down @kraj suggested something helpful at that time.

@msisov : vc4graphics only - I'm really tired of blobs and all their special interfaces causing 'nice' issues all over the place.

[1] http://lists.openembedded.org/pipermail/openembedded-devel/2017-August/114122.html

twoerner, 2018-02-24:
Should we re-open this thread? Or perhaps https://github.com/OSSystems/meta-browser/pull/82 ? Or maybe start a new one?

Building the latest chromium-x11 (64.0.3282.167) with openembedded-core + meta-openembedded + meta-raspberrypi for MACHINE=raspberrypi3 on master of all layers, I am able to get GPU acceleration working (or at least apparently working as per **`chrome://gpu`**) with the following:
1. when the build is finished there are the files `libEGL.so` and `libGLESv2.so` in the `out/Release` folder
2. install these files to `/usr/lib/chromium`
3. `MACHINE_FEATURES_append = " vc4graphics"`
4. `DISTRO_FEATURES_append = " opengl x11"`
5. `DISTRO_FEATURES_remove = "wayland"`
6. goto `/usr/lib/chromium` and create the following symlink:
    `# ln -s libGLESv2.so libGLESv2.so.2`
7. gcc7 (this didn't work before with gcc6 and an earlier version of chromium)
8. don't specify any `--use-gl` flag, just `# chromium --no-sandbox` as root

```
Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "opensuse-42.3"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta-raspberrypi     = "master:72bc798ff548ba610e8dea53c80e39a37de8e043"
meta                 = "master:6e5dbb53a7cbd05baf07de62b1abc88dccfd4cb9"
meta-oe              
meta-gnome           
meta-python          
meta-networking      = "master:2f7de931885c1b9e63c4e4238f0f7ad1388e8b6d"
meta-browser         = "master:a413b4f6608eac46b72a3a79dcada0767fe4cf58"
```

rakuco, 2018-03-01:
[These](https://chromium.googlesource.com/chromium/src/+/64.0.3282.167/ui/gl/init/gl_factory_x11.cc#29) are the GL backends that Chromium will try to use, in order. [libGLESv2.so.2](https://chromium.googlesource.com/chromium/src/+/64.0.3282.167/ui/gl/init/gl_initializer_x11.cc#38) is the name of the library that `--use-gl=egl` will try to load, and [libGLESv2.so](https://chromium.googlesource.com/chromium/src/+/64.0.3282.167/ui/gl/init/gl_initializer_x11.cc#41) is the ANGLE implementation of the GL libraries.

Looking at the steps you've described, I'm assuming there's no [libGL.so.1](https://chromium.googlesource.com/chromium/src/+/64.0.3282.167/ui/gl/init/gl_initializer_x11.cc#35) in your RPi3 image, which means the desktop GL backend doesn't work and Chromium (possibly) tries to use the EGL backend as a fallback. You're then making Chromium actually use the ANGLE implementation instead of an actual EGL one.

Can you try a few things:
* Does the RPi really not have `libGL.so.1`?
* What happens if you don't do step 6 (the symlink bit), do follow steps 1 and 2 (install the libraries with their original names) and pass `--use-gl=angle` to Chromium?

rakuco, 2018-03-01:
(no idea why GCC was having an effect on your tests, or what vc4graphics does though)

schnitzeltony, 2018-04-05:
I think we can finally close this. It was not related to chromium but VC4 dri2/dri3 improper configuration [1].

* @rakuco: Thanks a lot for taking care and giving helpful background information - now it makes sense to me
* Just to mention: This is written on Raspi / have checked my companies Outlook emails / watched some OK Go videos: Perfect / no issues

[1] https://lists.yoctoproject.org/pipermail/yocto/2018-April/040613.html

closed 2018-04-05 by schnitzeltony

twoerner, 2018-04-05:
To close out this issue...

**vc4graphics**
To get accelerated graphics working on the Pi, there are two paths that can be taken:
1. userland (https://github.com/raspberrypi/userland)
1. vc4 (https://github.com/anholt/mesa/wiki/VC4, https://github.com/anholt/mesa/wiki/VC4-OpenGL-support)

Myself and @schnitzeltony always use vc4graphics, so @msisov 's comment about trying vc4 to resolve the issue is irrelevant.

**gcc6/7**
I know that at one point I wasn't able to get chromium to display anything on the screen if gcc7 was used. This has been resolved. At some point some change was made somewhere so now builds of chromium with gcc7 are able to render things on the screen.

**libGL.so**
The RPi does have a libGL.so.1; it is provided by mesa.

**symlink, accelerated graphics, system vs chromium-built libEGL & libGLESv2**
In my builds mesa/vc4/etnaviv/freedreno are providing libGL, libEGL, libGLES etc as system libraries, found in /usr/lib. When I build chromium-x11, in the `out/Release` folder are found chromium's own chromium-built libEGL and libGLES. These files are created for rpi3-32 (vc4), rpi3-64 (vc4), dragonboard-410c (freedreno), and wandboard (etnaviv); therefore it's a good assumption that chromium builds its own libEGL/libGLES for arm32 and aarch64 MACHINEs (and probably all others too?).

_On the RPi_: If I run this chromium, it would run, but (according to chrome://gpu) there was no graphics acceleration. If I copied the chromium-built libEGL and libGLES from the build machine to /usr/lib/chromium on the RPi and created the necessary symlink, chromium would run and be accelerated. If any of these were missing, chromium would run, but not be accelerated. Any attempt to get chromium to be accelerated by using the system-provided libEGL/libGLES would not succeed.

However, @schnitzeltony very recently stumbled upon the "modified vs tiling" problem that I had been seeing both with chromium and non-chromium graphical applications on both the rpi3-32 and rpi3-64. He and @fooishbar traced it down to being a problem with mesa and the fact that OE was only enabling dri2 and not dri3. @schnitzeltony  submitted a patch, and it was accepted, to always enable dri3 when building mesa for x11.

As a result, chromium is now able to use the system-provided libEGL and libGLES and be accelerated! There is no need to copy the chromium-built libEGL/libGLES to /usr/lib/chromium, there's no need for any symlinks, and there's no need for any patches to chromium. chromium-x11 is now able to use the system libEGL/libGLES that come from mesa/vc4/etnaviv/freedreno, builds and runs with gcc7, the chromium app comes up on the display, and graphics are accelerated on the 4 builds I've tested.

_NOTE_: neither the dragonboard-410c (freedreno) nor the wandboard (etnaviv) were affected by this whatsoever. This issue appears to be only an issue on the RPi. In other words, I never had to copy the chromium-built libEGL/libGLES libraries to the target to get acceleration. Also with these two boards the graphics always works and chromium is always accelerated regardless of whether or not dri3 is enabled in mesa.

rakuco, 2018-04-11:
Thanks a lot for all the investigation and the detailed explanation of what's going on! I'm glad this is finally solved, and even more glad the chromium recipe wasn't at fault :-)

> When I build chromium-x11, in the out/Release folder are found chromium's own chromium-built libEGL and libGLES. These files are created for rpi3-32 (vc4), rpi3-64 (vc4), dragonboard-410c [...]

Yes, those libraries come from ANGLE and are built for all platforms as far as I can see. They're the ones only shipped by the non-stable official Chrome .deb and .rpm packages.

gil0109, 2018-07-10:
@twoerner Any chance you can share your local.conf and bblayer?

twoerner, 2018-07-10:
@gil0109 there isn't anything interesting going on in my bblayers.conf, other than to add meta-browser

my local.conf only has the following additions that affect chromium-x11:

```
IMAGE_FEATURES += "x11"
MACHINE_FEATURES_append = " vc4graphics"
DISTRO_FEATURES_append = " opengl x11"
DISTRO_FEATURES_remove = "wayland

CORE_IMAGE_EXTRA_INSTALL += "chromium-x11"
```

then I build core-image-full-cmdline

gil0109, 2018-07-14:
@twoerner thanks.  Did you manually add the libEGL and libGLES via .bbappend then?

gil0109, 2018-07-14:
@twoerner I can't seem to get chromium to be GPU accelerated.  

From this thread, it seems like I don't need to copy any files any longer.  But even if I do, it doesn't seem to work.  Any suggestions?

twoerner, 2018-07-15:
@gil0109 yes, that's true: the acceleration stuff has stopped working. I tend to work on master.

Originally I was copying the gles and egl files from the chromium build to the target filesystem in a *.bbappend and it was working. Then @schnitzeltony discovered that by enabling dri3 in mesa, the system-built gles/egl files would work fine, so I dropped my *.bbappend and simply used the ones built from oe-core. That all worked too.

That worked for about a week or two; then it stopped working. (I do daily builds off master, so lots of things change in a typical image from day to day)

I tried going back to copying the chromium-built files, but that wasn't working anymore either. At the time I worked on trying to figure out what happened to cause it to stop working, but never found the culprit. I had eliminated the chromium version bumps. When I tried building an older version of chromium (one that I knew had been working) with the latest of everything else, it still wouldn't claim to be accelerated. At that point I had to spend time on other things, but my next target would have been mesa itself. Around that time, oe-core switched from mesa-17 to mesa-18. It's a total guess on my part, but there's a chance this might have been the change that messed chromium up.

Around that time I was also working on trying to get hard numbers on the acceleration. I have a funny feeling that, although chromium _says_ it is accelerated, it's really not (or at least not accelerated very much). Ideally I would like to find some test suite that I can run in chromium to provide actual numbers for the alleged speed-up. If anyone knows of one, that'd be great.

Romaxx, 2018-10-01:
> @gil0109 there isn't anything interesting going on in my bblayers.conf, other than to add meta-browser
>
> my local.conf only has the following additions that affect chromium-x11:
>
> ```
> IMAGE_FEATURES [...]

Have do you tried with MACHINE="raspberrypi3-64" with this configuration ?