OSSystems/meta-browser #89: Is it possible to use this layer to create WebRTC binaries?

circuitry2, 2017-12-27:
I am trying to generate a valid WebRTC binary for ARM V7. Will running this recipe on my Yocto host generate WebRTC binaries, or will it just generate the Chromium package?

rakuco, 2017-12-27:
Not being very familiar with the WebRTC layer, I can only say the recipe ultimately runs `ninja chrome`. Whether that produces the binaries you need is unclear though. Can you provide some additional information about what you're trying to do?

circuitry2, 2017-12-27:
Sure, thanks for the quick response @rakuco.

I am trying to generate WebRTC binaries (generally named libwebrtc.a) that I can use to hook up to a golang library (https://github.com/keroserene/go-webrtc).

I specifically need to get WebRTC running on an ARMV7 Yocto machine.

The process of generating these binaries is kind of a nightmare and I have not yet been able to get through it. To build WebRTC binaries you need to download the Chromium developer tools and about 10GB of source code to ultimately make a 20MB binary (https://webrtc.org/native-code/development/).

I am hoping I can use the Chromium Yocto recipe with my existing build infrastructure (bitbake recipes) to generate the binaries I need, because I'm not really having much luck manually building the binaries.

rakuco, 2017-12-27:
Oh, you mean a static library (`libwebrtc.a`), not a binary in the sense of an executable.

`libwebrtc.a` is indeed produced during the build, but only for the target architecture since it's used as part of the final `chrome` binary.

Those instructions in the WebRTC website essentially show how to check out the parts of Chromium the code depends on and then build it, which is pretty much the same thing that's done as part of what this recipe builds.

circuitry2, 2017-12-27:
Awesome! I will give it a shot, thanks!!

rakuco, 2017-12-29:
Closing for now; there isn't much for us to do on our side here.

closed 2017-12-29 by rakuco