OSSystems/meta-browser #94: Ycoto/chromium-x11: can't resolve `protected_memory'

rkchrome, 2018-01-30:
When I build the chromium-x11 for M64, there is a error for building, do you have a patch to fix it? 

a7hf-neon-vfpv4-rk-linux-gnueabi/chromium-x11/64.0.3282.119-r0/recipe-sysroot= -g1 -c ../../base/memory/protected_memory_posix.cc -o obj/base/base/protected_memory_posix.o
| {standard input}: Assembler messages:
| {standard input}:316: Warning: ignoring changed section attributes for protected_memory
| {standard input}:644: Error: can't resolve `protected_memory' {protected_memory section} - `.Ltext0' {.text section}
| {standard input}:645: Error: can't resolve `protected_memory' {protected_memory section} - `.Ltext0' {.text section}
| {standard input}:646: Error: can't resolve `protected_memory' {protected_memory section} - `.Ltext0' {.text section}
| {standard input}:647: Error: can't resolve `protected_memory' {protected_memory section} - `.Ltext0' {.text section}
| {standard input}:650: Error: can't resolve `protected_memory' {protected_memory section} - `.Ltext0' {.text section}
| {standard input}:651: Error: can't resolve `protected_memory' {protected_memory section} - `.Ltext0' {.text section}
| {standard input}:652: Error: can't resolve `protected_memory' {protected_memory section} - `.Ltext0' {.text section}
| {standard input}:653: Error: can't resolve `protected_memory' {protected_memory section} - `.Ltext0' {.text section}

Perhaps, we should need add the Linking with lld is required to workaround crbug.com/792777.

kraj, 2018-01-30:
while lld is a upcoming option, I dont think its good to make it a requirement. We should see why this error happens, probably replacing -g1 with -g would get you through this. Please add information about compiler version etc too. it probably is a compiler issue, this is not related to chromium bug you are citing here. Its an assembler error.

rakuco, 2018-01-30:
True, I'd rather not have to depend on lld and clang for as long as possible. And like @kraj said, can you please tell us which Yocto or OE branch you're using, your compiler, your `MACHINE` and whether you have `ld-is-gold` in your `DISTRO_FEATURES`?

rkchrome, 2018-01-30:
Used the gcc 7.3 version.  I have built the chromium-x11 with M63 on Ycoto, but have the above error with updating the ycoto. (chromium os update to M64)

Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-rk-linux-gnueabi"
MACHINE              = "kylin-rk3036"
DISTRO               = "rk-x11"
DISTRO_VERSION       = "2.4+snapshot-20180130"
TUNE_FEATURES        = "arm armv7ve vfp neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "HEAD:0c537554fc2cf58a9cbfeaf39ed918c607483dce"

rkchrome, 2018-02-01:
Hmmm, I'm changing something for debug. is it correct? 

--- a/recipes-browser/chromium/chromium-gn.inc
+++ b/recipes-browser/chromium/chromium-gn.inc
@@ -99,7 +99,7 @@ GN_ARGS = " \
 # (debug, release, official) but for historical reasons there are two
 # separate flags.
 # See also: https://groups.google.com/a/chromium.org/d/msg/chromium-dev/hkcb6AOX5gE/PPT1ukWoBwAJ
-GN_ARGS += "is_debug=false is_official_build=true"
+GN_ARGS += "is_debug=true is_official_build=false"
 
 # Starting with M61, Chromium defaults to building with its own copy of libc++
 # instead of the system's libstdc++. Explicitly disable this behavior.
@@ -120,11 +120,11 @@ GN_ARGS += "use_custom_libcxx=false"
 #    instead, so that if DEBUG_BUILD is not set GN will not create a huge debug
 #    binary anyway. Since our compiler flags are passed after GN's, -g0 does
 #    not cause any issues if DEBUG_BUILD is set, as -g1 will be passed later.
-DEBUG_FLAGS_remove_arm = "-g"
-DEBUG_FLAGS_append_arm = "-g1"
+DEBUG_FLAGS_remove_arm = "-g1"
+DEBUG_FLAGS_append_arm = "-g"
 DEBUG_FLAGS_remove_x86 = "-g"
 DEBUG_FLAGS_append_x86 = "-g1"
-GN_ARGS += "symbol_level=0"
+GN_ARGS += "symbol_level=2"

rkchrome, 2018-02-01:
--- a/recipes-browser/chromium/chromium-gn.inc
+++ b/recipes-browser/chromium/chromium-gn.inc
....
-GN_ARGS += "is_debug=false is_official_build=true"
+GN_ARGS += "is_debug=true is_official_build=false"
 ...
-DEBUG_FLAGS_remove_arm = "-g"
-DEBUG_FLAGS_append_arm = "-g1"
+DEBUG_FLAGS_remove_arm = "-g1"
+DEBUG_FLAGS_append_arm = "-g"
 DEBUG_FLAGS_remove_x86 = "-g"
 DEBUG_FLAGS_append_x86 = "-g1"
-GN_ARGS += "symbol_level=0"
+GN_ARGS += "symbol_level=2"
......

rkchrome, 2018-02-01:
Sirs, the below change will repro the above issue, but the default build that's fine. Is't correct?
 
--- a/recipes-browser/chromium/chromium-gn.inc
+++ b/recipes-browser/chromium/chromium-gn.inc
....
-GN_ARGS += "is_debug=false is_official_build=true"
+GN_ARGS += "is_debug=true is_official_build=false"
...
-DEBUG_FLAGS_remove_arm = "-g"
-DEBUG_FLAGS_append_arm = "-g1"
+DEBUG_FLAGS_remove_arm = "-g1"
+DEBUG_FLAGS_append_arm = "-g"
DEBUG_FLAGS_remove_x86 = "-g"
DEBUG_FLAGS_append_x86 = "-g1"
-GN_ARGS += "symbol_level=0"
+GN_ARGS += "symbol_level=2"
......

rakuco, 2018-02-01:
Oh, so you're actually trying to do a full debug build. There are comments in `chromium-gn.inc` indicating why we pass those flags, and it basically boils down to the generated binaries being too big for the toolchain to handle (especially 32-bit ones).

How familiar are you with upstream Chromium development? I think we'd need upstream support to solve this kind of issue, so the best bet would be trying to build Chromium's master branch for desktop Linux with an ARM target and see how it fares.

otavio, 2018-06-04:
If no one says otherwise, I will close this issue.

closed 2018-06-12 by otavio