OSSystems/meta-browser #98: PACKAGECONFIG not working ?

tchapi, 2018-02-28:
Hi,

First of all thanks for this layer — super practical and well maintained.

In my custom layer, I have a `recipes-browser/chromium/chromium-x11_%.bbappend` file with the following content :

    FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"
    PACKAGECONFIG = "use-egl kiosk-mode proprietary-codecs"

... but chromium doesn't start in **kiosk-mode** — it seems these knobs are not taken into account.
Am I missing something ?

Thanks a lot !

otavio, 2018-02-28:
There are few things I'd confirm:

Check if PACKAGECONFIG is being applied:

```
$: bitbake chromium-x11 | grep 'PACKAGECONFIG='
```

If it does, likely something is broken on the extra arguments being given to it.

Check:

https://github.com/OSSystems/meta-browser/blob/master/recipes-browser/chromium/chromium-gn.inc#L248

and 

https://github.com/OSSystems/meta-browser/blob/master/recipes-browser/chromium/chromium-gn.inc#L373

tchapi, 2018-02-28:
**Thanks a lot**, I'll have a look at that ASAP. 
Just to be sure, I'm launching Chromium with `/usr/bin/chromium` with no specific arguments — that's the right way to do it, right ?

Best

otavio, 2018-02-28:
Yes, it should be fine.

tchapi, 2018-03-01:
Indeed it does apply the arguments : 

    $ bitbake -e chromium-x11 | grep 'PACKAGECONFIG='
    PACKAGECONFIG="use-egl kiosk-mode proprietary-codecs"

... so something is broken with the extra arguments maybe (or these parameters are incompatible with each other maybe?). Can you point me in the right direction ?

thx

rakuco, 2018-03-01:
Do the right command-line arguments exist in the wrapper script that gets installed (`/usr/lib/chromium/chromium-wrapper`)? I added `kiosk-mode` to `PACKAGECONFIG_pn-chromium-x11` in my `local.conf` and the right options were added there:

```
CHROME_EXTRA_ARGS="                           --start-fullscreen --kiosk --no-first-run --incognito "

# Note: exec -a below is a bashism.
exec -a "$0" "$HERE/chromium-bin" ${CHROME_EXTRA_ARGS} "$@"
```

Can you also check what `CHROME_EXTRA_ARGS` is set to via `bitbake -e`?

tchapi, 2018-03-01:
Ok, so.

The `/usr/lib/chromium/chromium-wrapper` script is fine, with all the relevant options. I think I found the culprit, and it's actually not a problem in this layer : the Chromium `--start-fullscreen` command line parameter doesn't really work, so it's actually not starting fullscreen, but to a bizarre size that is somewhat 3/4 of the screen.

Doing this works (credits to https://unix.stackexchange.com/a/361009) : 

    --window-size=7000,7000 --start-fullscreen

I wonder if it's specific to the hardware I test on (I'm on Intel boards, but the flag does not work on a RPi either, I just tested), or if it's a global behaviour of Chromium. Would it be worth it to put it in this layer too ? I don't know. Maybe just in the README (_I can make a PR if you wish_).

If somehow stumbles on this error at least there is a working workaround.
Thank you again Raphael

closed 2018-03-01 by tchapi

tchapi, 2018-03-01:
> If it does, I wonder if it may be related to the window manager you are running on your target hardware that's failing to report the right geometry to Chromium.

It may be yes. I'm using `matchbox-wm` and `mini-session`. I'm not really proficient in window managers to investigate further though, but I'll keep an eye open.

rakuco, 2018-03-01:
That combination might be the one to blame. FWIW, I've tried out core-image-sato on an Intel Atom NUC with the rocko branch and it worked fine in kiosk mode. I could see Chromium taking roughly half the screen for a second and then expanding and showing its UI. What's curious is that it worked without `--start-fullscreen` as well: just passing the other kiosk-mode arguments also showed the same kind of UI.

twoerner, 2018-03-01:
> What's curious is that it worked without --start-fullscreen as well

When I added "kiosk" capabilities I was trying to find the magical incantation that would do what I wanted. As I was researching, one of the first things I came across was the "--start-fullscreen" option. I tried it but noticed it didn't work, so I kept digging and found the other arguments (which seemed to work). It was just too strange to me that chromium had a cmdline arg called "--start-fullscreen" that didn't appear to contribute to making the app start full screen, so I kept it as part of the lineup anyway because... it has to be doing _something_, yes? ;-)

rakuco, 2018-03-01:
As far as I can see in https://bugs.chromium.org/p/chromium/issues/detail?id=167430 and [in the code](https://chromium.googlesource.com/chromium/src/+/64.0.3282.186/chrome/browser/ui/startup/startup_browser_creator_impl.cc#364), "--start-fullscreen" is a subset of "--kiosk". In both cases, the browser will start maximized, but "--start-fullscreen" without "--kiosk" allows one to exit full screen mode. I'm now considering removing it from the command-line options in a pull request.

twoerner, 2018-03-01:
That sounds reasonable.

rakuco, 2018-03-01:
> That sounds reasonable.

#99