OSSystems/meta-browser #109: chromium: Drop GCC 5 support?

rakuco, 2018-04-13:
As I work on the next upcoming Chromium milestone, once again I find myself spending most of my time fighting GCC 5 and trying to keep the build working with it. Sometimes it's a bug in the compiler itself, most of the time it's a bug in libstdc++. `chromium-gcc5-workarounds.patch` keeps growing, and I think things will only get worse with each new milestone.

Right now, the M66 build is broken with GCC 5 due to https://gcc.gnu.org/bugzilla/show_bug.cgi?id=77334. The bug says GCC 5.5.0 contains a fix, but I don't think anyone's willing to update GCC in Yocto's pyro branch at this point.

I can either spend more time trying to find a workaround (which will most likely also affect the build with other compilers too), or decide we only support GCC 6.4.0 and above (essentially, the default compiler versions for pyro and later branches).

I remember we originally started supporting GCC 5 to accommodate the needs of the Igalians, but it's not clear if they still need it or not.

rakuco, 2018-04-13:
@msisov @jdapena @tonikitoo @otavio @twoerner

otavio, 2018-04-13:
It does not seem to be worth the effort of maintaining the GCC 5 support alive. We can use the default compilers and reduce the maintenance work for the future updates in my option.

If we all agree in dropping it, I'd suggest to do that in following steps:

1. drop the `chromium-gcc5-workarounds.patch` of current version
1. do the upgrade in another patch

twoerner, 2018-04-13:
I don't need gcc5, everything I'm doing works fine with gcc7.

msisov, 2018-04-13:
I don't need gcc5 either. I will push an update to ozone wayland recipe soon. In a meanwhile, we have a fork of this repo with the latest m67 chromium.

PR #110 (2018-04-16): chromium: Remove support for GCC 5.
GCC 5 is old by Chromium's standards, and its support for C++14 is not
completely functional.

Keeping the recipe building with GCC 5 requires a lot of effort and
patching (which also affects the build with other compilers), and based on
the discussion in #109 it does not seem like the compiler is being widely
used -- and it helps unblock the upcoming update to Chromium 66, as it was
the last pending item to get everything building with the new milestone.

The recipe is building fine with GCC 6 (the default compiler on Yocto's pyro
branch) and GCC 7 (the def [... cut]
commit 1bfc47f2 (2018-04-16): chromium: Remove support for GCC 5.

GCC 5 is old by Chromium's standards, and its support for C++14 is not
completely functional.

Keeping the recipe building with GCC 5 requires a lot of effort and
patching (which also affects the build with other compilers), and based on
the discussion in #109 it does not seem like the compiler is being widely
used -- and it helps unblock the upcoming update to Chromium 66, as it was
the last pending item to get everything building with the new milestone.

The recipe is building fine with GCC 6 (the default compiler on Yocto's pyro
branch) and GCC 7 (the de [... cut]
closed 2018-04-16 by otavio