OSSystems/meta-browser #129: chromium-x11 do_compile: ninja: build stopped: subcommand failed.

g0hl1n, 2018-06-29:
Hi,
using the latest master (1cd38d701a49eade80a04140f70d3383117b9745) fails the build of chromium-x11_67.0.3396.87 for armhf (i.MX6).

Following error occurs repeatedly (please find more details in the attached log):
```
error: cannot convert 'const mojo::ScopedInterfaceEndpointHandle' to 'bool' in return
```

I'm using following branches and configuration:
```
Build Configuration:
BB_VERSION           = "1.39.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-myos-linux-gnueabi"
MACHINE              = "sdimx6"
DISTRO               = "myos"
DISTRO_VERSION       = "0.0.1"
TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "master:c7c64baa411fd96886918c235b851ae3e5a6a61f"
meta-oe
meta-multimedia      = "master:4517d1c695efd0b175f4e7242e1a7bb4361527d9"
meta-java            = "master-next:c8f7e412aaee4e41ec8ff980f463536c757538d0"
meta-freescale       = "master:aa3c33cdcd9c336ec39765d5a73dc62f87393f2c"
meta-browser         = "master:1cd38d701a49eade80a04140f70d3383117b9745"
```

The complete do_compile log is here: [log.do_compile.log](https://github.com/OSSystems/meta-browser/files/2148580/log.do_compile.log)

Any help is much appreciated!

regards;Richard.L

rakuco, 2018-06-29:
It looks like you've already found a fix :) M67 does need a few patches to build fine with GCC 8, let me know if you intend to send a pull request, otherwise I can look at this next week.

g0hl1n, 2018-07-03:
Thanks. Any help is greatly appreciated!
Sadly that fix alone didn't solved the issue, I'm getting `cc1plus: internal compiler error: Segmentation fault` errors now... I'll take another look at it later that day.

otavio, 2018-07-03:
@g0hl1n this is likely due to your build host being without enough memory to build it.

PR #130 (2018-07-03): chromium-x11: Import patch to fix the build with GCC 8
Fixes #129.
commit 8e159ff9 (2018-07-03): chromium-x11: Import patch to fix the build with GCC 8

Fixes #129.
closed 2018-07-03 by otavio