OSSystems/meta-browser #131: Build content_shell

joone, 2018-07-03:
Hi,

I'm trying to build content_shell, but I get the below error:

/home/joone/otc/yocto/poky/build/tmp/work/corei7-64-poky-linux/chromium-x11/67.0.3396.87-r0/chromium-67.0.3396.87/out/Release
| + do_compile
| + ninja -v -j 40 chrome chrome_sandbox chromedriver content_shell
| 
| + cd /home/joone/otc/yocto/poky/build/tmp/work/corei7-64-poky-linux/chromium-x11/67.0.3396.87-r0/chromium-67.0.3396.87/out/Release
| + do_compile
| + ninja -v -j 40 chrome chrome_sandbox chromedriver content_shell
| ninja: error: '../../third_party/test_fonts/test_fonts/Arimo-Bold.ttf', needed by 'test_fonts/Arimo-Bold.ttf', missing and no known rule to make it
| 
| ninja: error: '../../third_party/test_fonts/test_fonts/Arimo-Bold.ttf', needed by 'test_fonts/Arimo-Bold.ttf', missing and no known rule to make it
| + bb_exit_handler
| + ret=1

How can we fix this problem?

rakuco, 2018-07-04:
The fonts are downloaded as part of the `gclient runhooks` process, which is why they're not being found in the tarball when you run your build.

I see a few different options depending on what exactly you want to do with content_shell:
1. Add "check out depot_tools" and "run `gclient runhooks`" as steps to your recipe. The downside is that you will also need to create a .gclient file and a lot more hooks will be run (e.g. the one that downloads clang).
1. Check out depot_tools, but run the hook that downloads fonts manually (e.g. you invoke `download_from_google_storage` in the recipe).
1. Provide the fonts yourself.
1. Patch the required GN files so that those fonts aren't required. You might need to change content_shell as well since I think it looks for those fonts in order to run (or at least when running in layout test mode).

Is there anything in specific that you need from content_shell that the `chrome` binary can't provide?

joone, 2018-07-07:
I tried the first and second options, but it doesn't work well under proxy environment. I could set proxy, but checking out depot_tools and extra works make the script complicated. The third option works fine, but it's a temporary solution. I think that the fourth option is best.

We could only run content_shell with ozoen-gbm when toolkit_views = false, but chromium is not built with the build flag.  Can we add ozone-gbm recipe as part of Chromium recipe even if it only runs content_shell?

rakuco, 2018-07-09:
> We could only run content_shell with ozoen-gbm when toolkit_views = false, but chromium is not built with the build flag. Can we add ozone-gbm recipe as part of Chromium recipe even if it only runs content_shell?

At least for now, this looks like downstream work and I don't see the need to add this to master just yet (and keeping a `.bbappend` around should be fairly easy). We can revisit this once the work has matured and stabilized though.

joone, 2018-07-09:
Okay, thanks!

closed 2018-07-09 by joone