OSSystems/meta-browser #135: chromium-x11: Consider dropping support for GCC 6 (and Yocto pyro)

rakuco, 2018-07-16:
I'm currently working on Chromium 68 (currently in the beta channel), and both ARM and ARM64 builds (I've tried `MACHINE="qemuarm64"` and `MACHINE="raspberrypi3"`) currently fail with pyro's GCC 6 due to an internal compiler error when building some bits of Skia that use NEON and other optimizations:

```
arch64-poky-linux-gcc  --sysroot=/data/src/yocto/poky/build-pyro-qemuarm64/tmp/work/aarch64-poky-linux/chromium-x11/68.0.3440.59-r0/recipe-sysroot -MMD -MF obj/skia/skcm
s/Transform.o.d -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DNO_TCMALLOC -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE
_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_F
ORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -I../../third_party/skia/third_party/skcms -w -std=c11 -fno-strict-aliasing --param=ssp
-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -Wall -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-n
ull-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvi
sibility=hidden -std=gnu11  -O2 -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/data/src/yocto/poky/build-pyro-qemuarm64/tmp/work/aarch64-poky-linux/chromium-x11
/68.0.3440.59-r0=/usr/src/debug/chromium-x11/68.0.3440.59-r0 -fdebug-prefix-map=/data/src/yocto/poky/build-pyro-qemuarm64/tmp/work/aarch64-poky-linux/chromium-x11/68.0.34
40.59-r0/recipe-sysroot-native= -fdebug-prefix-map=/data/src/yocto/poky/build-pyro-qemuarm64/tmp/work/aarch64-poky-linux/chromium-x11/68.0.3440.59-r0/recipe-sysroot= -g1
-c ../../third_party/skia/third_party/skcms/src/Transform.c -o obj/skia/skcms/Transform.o
In file included from ../../third_party/skia/third_party/skcms/src/Transform.c:176:0:
../../third_party/skia/third_party/skcms/src/Transform_inl.h: In function 'exec_ops':
../../third_party/skia/third_party/skcms/src/Transform_inl.h:459:32: internal compiler error: in convert_move, at expr.c:299
     v = (U16)( ((v<<8)|(v>>8)) & 0xffff );
              ~~~~~~~~~~~~~~~~~~^~~~~~~~~~
```

The code builds fine on rocko and later (which have GCC 7). I see a few different options here:

1. Explicitly drop support for GCC 6 and pyro and stop testing it altogether.
1. Just stop building and testing pyro and GCC 6 on ARM/ARM64, but make sure an x86/x86-64 build still works.
1. Bisect GCC and try to get the internal compiler error fix into pyro's GCC recipe. That requires building multiple versions of GCC as a cross-compiler, which, let's say, is definitely not a pleasant experience.

rakuco, 2018-07-16:
@jdapena, @msisov too

rakuco, 2018-07-16:
> Just stop building and testing pyro and GCC 6 on ARM/ARM64, but make sure an x86/x86-64 build still works.

Actually, I think the code section causing an issue is machine-independent, so I guess other architectures are affected as well.

otavio, 2018-07-16:
I am fine dropping it. We can make a pyro branch with latest working and move on.

rakuco, 2018-07-17:
For completeness and my own peace of mind, this is the actual bug that was causing GCC 6.4.0 to ICE: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=83623

I'll try to send a backport to the Yocto list tomorrow.

kraj, 2018-07-17:
That seems to be the case. I forgot to update this issue last night. Please send a patch for relevant branches for oe-core

rakuco, 2018-07-17:
OTOH, I launched a pyro + qemuarm64 build here with the upcoming M68 work and there was another build error that seems to have been fixed in GCC 7. The code in question is also present in M67, which likely means nobody's been testing pyro + aarch64 lately and makes me question if it's worth continuing to support this branch.

rakuco, 2018-07-18:
> Please send a patch for relevant branches for oe-core

[Done](http://lists.openembedded.org/pipermail/openembedded-core/2018-July/153052.html).

rakuco, 2018-08-09:
There's been no response to my patch and the upstream pyro branch remains broken. Can we consider there's enough lack of interest and stop supporting pyro here?

kraj, 2018-08-09:
Please ping Armin who is stable maintainer.

rakuco, 2018-09-24:
I [pinged](http://lists.openembedded.org/pipermail/openembedded-core/2018-August/153977.html) Armin, and you [also did](http://lists.openembedded.org/pipermail/openembedded-core/2018-August/154449.html), but my patch hasn't been merged yet.

Given the lack of upstream interest, I've sent #175 to explicitly mention we don't support pyro and earlier.

kraj, 2018-09-24:
OK.

PR #175 (2018-09-24): chromium-x11: Make it clear that we need GCC >= 7.3.1
GCC 6 (i.e. pyro and earlier) builds are currently broken on ARM, and a
patch to fix the issue has not been merged in months. Nobody has complained
so far, so it should be safe to say supporting older GCC releases is not
worth the effort.

Fixes #135.
commit 50b56b08 (2018-09-24): chromium-x11: Make it clear that we need GCC >= 7.3.1

GCC 6 (i.e. pyro and earlier) builds are currently broken on ARM, and a
patch to fix the issue has not been merged in months. Nobody has complained
so far, so it should be safe to say supporting older GCC releases is not
worth the effort.

Fixes #135.
closed 2018-09-24 by rakuco