OSSystems/meta-browser #136: did the chromium-ozone-wayland been compiled with musl libc library?

13824125580, 2018-07-17:
now i am porting the meta-browser on my our own platform,  which only supports musl libc toolchain for reduce code size. i meet so many compile errors,  basically related to the difference of glibc and musl libc, 
for example, directly call "__libc_malloc" symble on chromium code and some library interface can`t be supported by musl.
so, i lost my confidence for porting rightly to the platform with muscle,  has someone ever been done this?

msisov, 2018-07-17:
hi. No, we haven't tried. sorry

rakuco, 2018-07-17:
As you've noticed, using musl is not supported either here or upstream, and making it work takes considerable effort and only makes sense if someone commits to maintaining it upstream as well. So much so that we actually dropped all musl patches we used to carry in #106

closed 2018-07-17 by rakuco