OSSystems/meta-browser #137: crash issue when launch chromium

13824125580, 2018-07-17:
i am porting the chromium-ozone-wayland to my target platform, but it immediately crashed when launch "chromium" command, so, has someone meet this before?
thanks for your help.
root@tulip-mozart:/# chromium 
[89124.065882] chromium-bin[1542]: unhandled level 3 translation fault (11) at 0x7fd4339fd0, esr 0x92000047
[89124.076458] pgd = ffffffc029b2d000
[89124.080287] [7fd4339fd0] *pgd=0000000063feb003, *pud=0000000063feb003, *pmd=000000006a199003, *pte=0000000000000000
[89124.091897] 
[89124.093551] CPU: 0 PID: 1542 Comm: chromium-bin Tainted: G           O    4.4.89 #27
[89124.102169] Hardware name: sun50iw1 (DT)
[89124.106521] task: ffffffc029424980 ti: ffffffc029c9c000 task.ti: ffffffc029c9c000
[89124.114849] PC is at 0x5593569a10
[89124.118523] LR is at 0x5593569a68
[89124.122218] pc : [<0000005593569a10>] lr : [<0000005593569a68>] pstate: 80000000
[89124.130436] sp : 0000007fd433a000
[89124.134109] x29: 0000007fd433a000 x28: 0000007f9d2c7000 
[89124.140024] x27: 0000007f9c7a8320 x26: 0000007f9d2c53d0 
[89124.145923] x25: 0000007f9d2c7000 x24: 0000007fd4b38b30 
[89124.151841] x23: 0000000000000001 x22: 0000000000011c00 
[89124.157739] x21: 000000559a44603c x20: 0000005599f51f30 
[89124.163660] x19: 0000007f9c7b3d10 x18: 0000000000000800 
[89124.169558] x17: 0000005593569a10 x16: 0000007f9c7b3a40 
[89124.175474] x15: 0000000000000000 x14: 0000007f9d217d00 
[89124.181396] x13: 0000005590c45ac8 x12: 0000000000000018 
[89124.187295] x11: 0000000000000010 x10: 0000007f9d2c7f30 
[89124.193218] x9 : 0000000000000010 x8 : 0000005599eea688 
[89124.199121] x7 : 00000000ffffffff x6 : ffffffffffffffff 
[89124.205037] x5 : 00000000000007f8 x4 : 0000007f9d276be8 
[89124.210959] x3 : 0000005593569e08 x2 : 0000000000000000 
[89124.216859] x1 : 0000000000011c00 x0 : 0000000000011c00 
[89124.222777] 
Segmentation fault

rakuco, 2018-07-17:
@msisov

msisov, 2018-07-17:
just a quick guess - try to pass --no-sandbox

13824125580, 2018-07-18:
sandly :-(
root@tulip-mozart:/# chromium --no-sandbox
[190202.580786] chromium-bin[1662]: unhandled level 3 translation fault (11) at 0x7fc5724fe0, esr 0x92000047
[190202.591435] pgd = ffffffc029eea000
[190202.595302] [7fc5724fe0] *pgd=0000000063fe1003, *pud=0000000063fe1003, *pmd=0000000064924003, *pte=0000000000000000
[190202.607050] 
[190202.608802] CPU: 2 PID: 1662 Comm: chromium-bin Tainted: G           O    4.4.89 #27
[190202.617509] Hardware name: sun50iw1 (DT)
[190202.621973] task: ffffffc029424980 ti: ffffffc029e90000 task.ti: ffffffc029e90000
[190202.630388] PC is at 0x5569d97a10
[190202.634157] LR is at 0x5569d97a68
[190202.637927] pc : [<0000005569d97a10>] lr : [<0000005569d97a68>] pstate: 80000000
[190202.646246] sp : 0000007fc5725010
[190202.650025] x29: 0000007fc5725010 x28: 0000007f7de1e000 
[190202.656018] x27: 0000007f7d2ff320 x26: 0000007f7de1c3d0 
[190202.662037] x25: 0000007f7de1e000 x24: 0000007fc5f24350 
[190202.668029] x23: 0000000000000001 x22: 0000000000011c00 
[190202.674043] x21: 0000005570c7403c x20: 000000557077ff30 
[190202.680053] x19: 0000007f7d30ad10 x18: 0000000000000800 
[190202.686053] x17: 0000005569d97a10 x16: 0000007f7d30aa40 
[190202.692061] x15: 0000000000000000 x14: 0000007f7dd6ed00 
[190202.698057] x13: 0000005567473ac8 x12: 0000000000000018 
[190202.704066] x11: 0000000000000010 x10: 0000007f7de1ef30 
[190202.710076] x9 : 0000000000000010 x8 : 0000005570718688 
[190202.716074] x7 : 00000000ffffffff x6 : ffffffffffffffff 
[190202.722084] x5 : 00000000000007f8 x4 : 0000007f7ddcdbe8 
[190202.728076] x3 : 0000005569d97e08 x2 : 0000000000000000 
[190202.734084] x1 : 0000000000011c00 x0 : 0000000000011c00 
[190202.740093] 
Segmentation fault

i am using musl libc cross-toolchain, did this matter for this?

msisov, 2018-07-19:
maybe. Can you use gdb server and attach a remote debugger to get a meaningful backtrace?

twoerner, 2018-07-21:
meta-browser (i.e. this github repository) doesn't support musl.

kraj, 2018-07-21:
in this case the sources are not being prepared  from meta-browser recipe it seems. So musl-cross toolchain is a separate cross SDK then yocto/OE, its possible that target system is running OE,  so as such I dont think this question is relevant to OE directly. However, at one point we had chromium-x11 and firefox both functional with musl, but since community wasn't testing it as much it kind of has regressed, it should be resurrected so I am interested in all the patches that you are carrying for musl support and you might go back in meta-browser history to see all the patches that we once had for these browsers to work with musl.

13824125580, 2018-07-25:
thanks you all.

closed 2018-07-25 by 13824125580