OSSystems/meta-browser #138: how to lunch the browser on terminal?

13824125580, 2018-07-18:
i meed crash issue after launch browser, i am not sure if the launch command is right, 
what am i done is on serial terminal, just launch the browser with chromium command like that:
#chromium
is that right? any parameter need to added?

rakuco, 2018-07-18:
Assuming you're using chromium-x11, just make sure you are not running as root and that should be enough. For chromium-ozone-wayland, you might need additional parameters (cc @msisov).

13824125580, 2018-07-18:
what parameters? i am using "chromium-ozone-wayland", has this been tested before?

msisov, 2018-07-19:
If you're running chromium as a root, you have to pass --no-sandbox parameter. Other parameters (--mus and --ozone-platform=wayland) are handled automatically.

cchilumbu, 2018-07-20:
While running as root, as most embedded targets do, we get the " --no-sandbox is not supported...." message appear on the browser after launching it.

What is the best way to disable this?

Perhaps commenting out this message in the source code via a .bbappend file?

rakuco, 2018-07-20:
For a few milestones now, the chromium-x11 recipe (which is tracking a more recent release than chromium-ozone-wayland) has given up supporting launching Chromium as root because of this; it's a use case completely not supported by upstream, you should really consider launching Chromium as an unprivileged user.

cchilumbu, 2018-07-20:
Thank you for Raphael, I will do just that.

cchilumbu, 2018-07-20:
I have opened a new issue here regarding a build error I am getting, could you please take a look?

Thank you.

13824125580, 2018-07-20:
hi cchilumbu, have you been running chromium-ozone-wayland? from your description above, seems you have done this before

cchilumbu, 2018-07-20:
I have been running chromium-x11 successfully.

I would like to move away from X11 and use only weston & wayland.

I have posted the build errors I am getting when i try to build M64.

Please check the issue I have raised.

Your help is much appreciated.

13824125580, 2018-07-25:
thanks you all

closed 2018-07-25 by 13824125580