OSSystems/meta-browser #139: is there any version demands for "pokey" and "oe" layer to compile the chromium-ozone-wayland pass?

13824125580, 2018-07-19:
is there any version demands for "pokey" and "oe" layer to compile the chromium-ozone-wayland pass?

msisov, 2018-07-19:
there is no hard dependency.

cchilumbu, 2018-07-20:
I am using the yocto pyro branch, trying to build "chromium-ozone-wayland" for x86 target.

Do you have a meta-browser branch compatible with pyro?

I am using branch "ozone/wayland/67.0.3375.0.r543955" and chromium-x11 builds fine, but when I try to build chromium-ozone-wayland , I am getting the errors shown below:

Any ideas please how I can resolve this?

 [16658/35221] g++  -MMD -MF yocto_native/obj/third_party/flatbuffers/compiler_files/idl_gen_cpp.o.d  -isystem/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/recipe-sysroot-native/usr/include -isystem/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/recipe-sysroot-native/usr/include -O2 -pipe -D_GLIBCXX_USE_CXX11_ABI=0 -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../../third_party/flatbuffers/src/grpc -I../.. -Iyocto_native/gen -I../../third_party/flatbuffers/src/include -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -std=gnu++14 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wno-narrowing -c ../../third_party/flatbuffers/src/src/idl_gen_cpp.cpp -o yocto_native/obj/third_party/flatbuffers/compiler_files/idl_gen_cpp.o
| [16659/35221] g++  -MMD -MF yocto_native/obj/third_party/flatbuffers/compiler_files/idl_parser.o.d  -isystem/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/recipe-sysroot-native/usr/include -isystem/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/recipe-sysroot-native/usr/include -O2 -pipe -D_GLIBCXX_USE_CXX11_ABI=0 -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../../third_party/flatbuffers/src/grpc -I../.. -Iyocto_native/gen -I../../third_party/flatbuffers/src/include -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -std=gnu++14 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wno-narrowing -c ../../third_party/flatbuffers/src/src/idl_parser.cpp -o yocto_native/obj/third_party/flatbuffers/compiler_files/idl_parser.o
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/temp/log.do_compile.1367)
ERROR: Task (/home/cchilumbu/projects/yocto-dr0081/sources/meta-browser/recipes-browser/chromium/chromium-ozone-wayland_67.0.3375.0.r543955.igalia.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6005 tasks of which 5990 didn't need to be rerun and 1 failed.

cchilumbu, 2018-07-20:
I will raise a new issue for this.

13824125580, 2018-07-25:
thanks you all

closed 2018-07-25 by 13824125580