OSSystems/meta-browser #140: Build error for "chromium-ozone-wayland" - yocto pyro

cchilumbu, 2018-07-20:
I am using the yocto pyro branch, trying to build "chromium-ozone-wayland" for x86 target.

Do you have a meta-browser branch compatible with pyro?

I am using branch "ozone/wayland/67.0.3375.0.r543955" and chromium-x11 builds fine, but when I try to build chromium-ozone-wayland , I am getting the errors shown below:

Any ideas please how I can resolve this?

[16658/35221] g++ -MMD -MF yocto_native/obj/third_party/flatbuffers/compiler_files/idl_gen_cpp.o.d -isystem/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/recipe-sysroot-native/usr/include -isystem/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/recipe-sysroot-native/usr/include -O2 -pipe -D_GLIBCXX_USE_CXX11_ABI=0 -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../../third_party/flatbuffers/src/grpc -I../.. -Iyocto_native/gen -I../../third_party/flatbuffers/src/include -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -std=gnu++14 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wno-narrowing -c ../../third_party/flatbuffers/src/src/idl_gen_cpp.cpp -o yocto_native/obj/third_party/flatbuffers/compiler_files/idl_gen_cpp.o
| [16659/35221] g++ -MMD -MF yocto_native/obj/third_party/flatbuffers/compiler_files/idl_parser.o.d -isystem/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/recipe-sysroot-native/usr/include -isystem/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/recipe-sysroot-native/usr/include -O2 -pipe -D_GLIBCXX_USE_CXX11_ABI=0 -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../../third_party/flatbuffers/src/grpc -I../.. -Iyocto_native/gen -I../../third_party/flatbuffers/src/include -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -std=gnu++14 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wno-narrowing -c ../../third_party/flatbuffers/src/src/idl_parser.cpp -o yocto_native/obj/third_party/flatbuffers/compiler_files/idl_parser.o
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/67.0.3375.0.r543955.igalia.1-r0/temp/log.do_compile.1367)
ERROR: Task (/home/cchilumbu/projects/yocto-dr0081/sources/meta-browser/recipes-browser/chromium/chromium-ozone-wayland_67.0.3375.0.r543955.igalia.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6005 tasks of which 5990 didn't need to be rerun and 1 failed.

rakuco, 2018-07-20:
That branch and that recipe version do not exist in this repository, it looks like you're using https://github.com/igalia/meta-browser.

cchilumbu, 2018-07-20:
Is there a commit here that you can recommend that is compatible with pyro?

I would like to build chromium-ozone-wayland.

Thanks

rakuco, 2018-07-20:
I can only speak for the version that is in the repository; if you're trying to build M67 or anything more recent than M64 you should raise the issues in Igalia/meta-browser instead.

cchilumbu, 2018-07-20:
Ok let me try to build the version here M64.

Thanks.

cchilumbu, 2018-07-20:
I now get this error:

| [473/31164] python ../../tools/json_schema_compiler/compiler.py --root=../../ --destdir=gen --namespace=extensions::api::\%\(namespace\)s --bundle-name=Chrome --generator=cpp-bundle-schema --include-rules=extensions/common/api:extensions::api::\%\(namespace\)s ../../chrome/common/extensions/api/accessibility_features.json ../../chrome/common/extensions/api/accessibility_private.json ../../chrome/common/extensions/api/activity_log_private.json ../../chrome/common/extensions/api/autofill_private.idl ../../chrome/common/extensions/api/automation.idl ../../chrome/common/extensions/api/automation_internal.idl ../../chrome/common/extensions/api/autotest_private.idl ../../chrome/common/extensions/api/bookmark_manager_private.json ../../chrome/common/extensions/api/bookmarks.json ../../chrome/common/extensions/api/braille_display_private.idl ../../chrome/common/extensions/api/browser.idl ../../chrome/common/extensions/api/chrome_web_view_internal.json ../../chrome/common/extensions/api/cloud_print_private.json ../../chrome/common/extensions/api/command_line_private.json ../../chrome/common/extensions/api/content_settings.json ../../chrome/common/extensions/api/context_menus_internal.json ../../chrome/common/extensions/api/context_menus.json ../../chrome/common/extensions/api/cookies.json ../../chrome/common/extensions/api/cryptotoken_private.idl ../../chrome/common/extensions/api/dashboard_private.json ../../chrome/common/extensions/api/data_reduction_proxy.json ../../chrome/common/extensions/api/debugger.json ../../chrome/common/extensions/api/desktop_capture.json ../../chrome/common/extensions/api/developer_private.idl ../../chrome/common/extensions/api/dial.idl ../../chrome/common/extensions/api/downloads.idl ../../chrome/common/extensions/api/downloads_internal.idl ../../chrome/common/extensions/api/easy_unlock_private.idl ../../chrome/common/extensions/api/experience_sampling_private.json ../../chrome/common/extensions/api/font_settings.json ../../chrome/common/extensions/api/gcm.json ../../chrome/common/extensions/api/history.json ../../chrome/common/extensions/api/hotword_private.idl ../../chrome/common/extensions/api/i18n.json ../../chrome/common/extensions/api/identity.idl ../../chrome/common/extensions/api/identity_private.idl ../../chrome/common/extensions/api/image_writer_private.idl ../../chrome/common/extensions/api/inline_install_private.idl ../../chrome/common/extensions/api/instance_id.json ../../chrome/common/extensions/api/language_settings_private.idl ../../chrome/common/extensions/api/manifest_types.json ../../chrome/common/extensions/api/media_galleries.idl ../../chrome/common/extensions/api/notifications.idl ../../chrome/common/extensions/api/omnibox.json ../../chrome/common/extensions/api/page_capture.json ../../chrome/common/extensions/api/passwords_private.idl ../../chrome/common/extensions/api/permissions.json ../../chrome/common/extensions/api/preferences_private.json ../../chrome/common/extensions/api/resources_private.idl ../../chrome/common/extensions/api/screenlock_private.idl ../../chrome/common/extensions/api/sessions.json ../../chrome/common/extensions/api/settings_private.idl ../../chrome/common/extensions/api/signed_in_devices.idl ../../chrome/common/extensions/api/streams_private.idl ../../chrome/common/extensions/api/sync_file_system.idl ../../chrome/common/extensions/api/system_indicator.idl ../../chrome/common/extensions/api/system_private.json ../../chrome/common/extensions/api/tab_capture.idl ../../chrome/common/extensions/api/tabs.json ../../chrome/common/extensions/api/types.json ../../chrome/common/extensions/api/web_navigation.json ../../chrome/common/extensions/api/webrtc_audio_private.idl ../../chrome/common/extensions/api/webrtc_desktop_capture_private.idl ../../chrome/common/extensions/api/webrtc_logging_private.idl ../../chrome/common/extensions/api/webstore_private.json ../../chrome/common/extensions/api/webstore_widget_private.idl ../../chrome/common/extensions/api/windows.json ../../chrome/common/extensions/api/processes.idl ../../chrome/common/extensions/api/input_ime.json ../../chrome/common/extensions/api/mdns.idl ../../chrome/common/extensions/api/cast_streaming_receiver_session.idl ../../chrome/common/extensions/api/cast_streaming_rtp_stream.idl ../../chrome/common/extensions/api/cast_streaming_session.idl ../../chrome/common/extensions/api/cast_streaming_udp_transport.idl ../../chrome/common/extensions/api/browsing_data.json ../../chrome/common/extensions/api/extension.json ../../chrome/common/extensions/api/idltest.idl ../../chrome/common/extensions/api/music_manager_private.idl ../../chrome/common/extensions/api/top_sites.json ../../chrome/common/extensions/api/app.json ../../chrome/common/extensions/api/browser_action.json ../../chrome/common/extensions/api/commands.json ../../chrome/common/extensions/api/declarative_content.json ../../chrome/common/extensions/api/page_action.json ../../chrome/common/extensions/api/privacy.json ../../chrome/common/extensions/api/proxy.json ../../chrome/common/extensions/api/tts_engine.json ../../chrome/common/extensions/api/tts.json ../../chrome/common/extensions/api/webstore.json ../../chrome/common/extensions/api/input_method_private.json
| [474/31164] python ../../tools/metrics/histograms/generate_expired_histograms_array.py -nchrome_metrics -ogen -Hchrome/browser/metrics/expired_histograms_array.h -d../../chrome/MAJOR_BRANCH_DATE ../../tools/metrics/histograms/histograms.xml ../.. [... cut ...]
ERROR: Task (/home/cchilumbu/projects/yocto-dr0081/sources/meta-browser/recipes-browser/chromium/chromium-ozone-wayland_64.0.3274.0.r517731.igalia.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6005 tasks of which 5994 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/cchilumbu/projects/yocto-dr0081/sources/meta-browser/recipes-browser/chromium/chromium-ozone-wayland_64.0.3274.0.r517731.igalia.1.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

cchilumbu, 2018-07-20:
Does the below help?

Thanks

[439/31164] x86_64-poky-linux-g++  -m64 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2 --sysroot=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0/recipe-sysroot -MMD -MF obj/chrome/common/non_code_constants/env_vars.o.d   -O2 -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0=/usr/src/debug/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0 -fdebug-prefix-map=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0/recipe-sysroot= -g1 -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -Wall -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -std=gnu++14 -Wno-narrowing -fno-rtti -fno-exceptions -fvisibility-inlines-hidden -c ../../chrome/common/env_vars.cc -o obj/chrome/common/non_code_constants/env_vars.o
[440/31164] x86_64-poky-linux-g++  -m64 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2 --sysroot=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0/recipe-sysroot -MMD -MF obj/base/base/field_trial.o.d   -O2 -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0=/usr/src/debug/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0 -fdebug-prefix-map=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0/recipe-sysroot= -g1 -DUSE_SYMBOLIZE -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DBASE_IMPLEMENTATION -I../.. -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -Wall -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unused-parameter -fno-omit-frame-pointer -g0 -fvisibility=hidden -O2 -fno-ident -fdata-sections -ffunction-sections -std=gnu++14 -Wno-narrowing -fno-rtti -fno-exceptions -fvisibility-inlines-hidden -c ../../base/metrics/field_trial.cc -o obj/base/base/field_trial.o
[441/31164] python ../../tools/json_schema_compiler/feature_compiler.py . APIFeature APIFeatureProvider gen/chrome/common/extensions/api api_features ../../extensions/common/api/_api_features.json ../../chrome/common/extensions/api/_api_features.json
[442/31164] touch obj/chrome/common/extensions/api/api_features_json_features.stamp
[443/31164] python ../../chrome/browser/resources/optimize_webui.py --host downloads --input ../../chrome/browser/resources/md_downloads --out_folder gen/chrome/browser/resources/md_downloads --depfile gen/chrome/browser/resources/md_downloads/build.d --html_in_files downloads.html --html_out_files vulcanized.html --js_out_files crisper.js --insert_in_head \<base\ href=chrome://downloads\>
FAILED: gen/chrome/browser/resources/md_downloads/vulcanized.html gen/chrome/browser/resources/md_downloads/crisper.js
python ../../chrome/browser/resources/optimize_webui.py --host downloads --input ../../chrome/browser/resources/md_downloads --out_folder gen/chrome/browser/resources/md_downloads --depfile gen/chrome/browser/resources/md_downloads/build.d --html_in_files downloads.html --html_out_files vulcanized.html --js_out_files crisper.js --insert_in_head \<base\ href=chrome://downloads\>
../../third_party/node/linux/node-linux-x64/bin/node ../../third_party/node/node_modules/polymer-bundler/lib/bin/polymer-bundler --exclude chrome://resources/html/polymer.html --exclude chrome://resources/polymer/v1_0/polymer/polymer.html --exclude chrome://resources/polymer/v1_0/polymer/polymer-micro.html --exclude chrome://resources/polymer/v1_0/polymer/polymer-mini.html --exclude chrome://resources/polymer/v1_0/web-animations-js/web-animations-next-lite.min.js --exclude chrome://resources/css/roboto.css --exclude chrome://resources/css/text_defaults.css --e [... cut ...] r517731.igalia.1/chrome/browser/resources/md_history" --out-dir gen/chrome/browser/resources/md_history/bundled --shell app.html --in-html app.html --in-html lazy_load.html failed: /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0/chromium-ozone-wayland-dev-64.0.3274.0.r517731.igalia.1/third_party/node/node_modules/polymer-bundler/lib/bin/polymer-bundler.js:178
        const [prefix, path] = redirect.split('|');

rakuco, 2018-07-20:
Apparently the node.js version you're using is failing.

cchilumbu, 2018-07-20:
Do you know how I can fix it please?

rakuco, 2018-07-20:
I use https://github.com/ostroproject/meta-iot-web for node here. I think meta-oe also has a version, but it may be too old in its pyro branch.

twoerner, 2018-07-21:
nodejs in meta-openembedded pyro branch: 4.8.2
nodejs in meta-iot-web master: 6.11.0
nodejs in meta-openembedded master: 8.11.3

msisov, 2018-07-23:
We are using nodejs 8.6 in our recipe for hardware we use - https://github.com/Igalia/meta-browser/tree/ozone/wayland/67.0.3375.0.r543955/recipes-devtools/nodejs

cchilumbu, 2018-07-24:
Thanks for the responses.

I am now using the nodejs 8.4 recipe, and have added the meta-iot-web layer. The error reported earlier seems to have been fixed.

cchilumbu, 2018-07-24:
I can see that meta-crosswalk does provide a recipe that installs snapshot_blob.bin.

https://github.com/crosswalk-project/meta-crosswalk/blob/master/recipes-crosswalk/crosswalk/crosswalk_20.50.533.12.bb

Is anyone using this? My concern is that there have not been updates to this git repository in years.

rakuco, 2018-07-24:
Over to @msisov, as this is an error in the chromium-ozone-wayland recipe.

cchilumbu, 2018-07-25:
Hello Maksim,

If you have suggestions for me to try out, please let me know.

I would really like to get this build completed error free.

msisov, 2018-07-25:
Sorry, we haven't tried to built yocto x86 before. I have no idea what can be broken now. As long as it still has been referring to js, you can try to use the latest nodejs recipe and see if the issue is resolved.

cchilumbu, 2018-07-25:
Thank you Maksim, I will give that a try.

otavio, 2018-10-19:
No feedback since July 25th, closing due to no activity.

closed 2018-10-19 by otavio

caroldesjardins, 2018-12-07:
I got the same issue last week with this recipe. 
bitbake chromium-ozone-wayland doesn't compile successfully for iMX8MQEVK. If someone can find a solution to repair this recipe soon. we will apreciate. :)

kraj, 2018-12-07:
can you try it with master branches ?

caroldesjardins, 2018-12-07:
It's done yet, and i got the same fail. 
chromium-ozone-wayland_71.0.3545.0
I'm not a Python specialist, but the script
FAILED: gen/chrome/browser/resources/md_downloads/vulcanized.html gen/chrome/browser/resources/md_downloads/vulcanized.p2.html gen/chrome/browser/resources/md_downloads/crisper.js 
python ../../chrome/browser/resources/optimize_webui.py --host downloads --input ../../chrome/browser/resources/md_downloads --out_folder gen/chrome/browser/resources/md_downloads --depfile gen/chrome/browser/resources/md_downloads/build.d --html_in_files downloads.html --html_out_files vulcanized.html --html_out_files_polymer2 vulcanized.p2.html --js_out_files crisper.js --insert_in_head \<base\ href=chrome://downloads\>
Traceback (most recent call last):
  File "../../chrome/browser/resources/optimize_webui.py", line 266, in <module>
    main(sys.argv[1:])
  File "../../chrome/browser/resources/optimize_webui.py", line 249, in main
    manifest_out_path = _optimize(args.input, args)
  File "../../chrome/browser/resources/optimize_webui.py", line 206, in _optimize
    _polymer_css_build(out_path, crisper_html_out_paths, args.html_out_files, 1)
  File "../../chrome/browser/resources/optimize_webui.py", line 225, in _polymer_css_build
    html_in_paths + ['-o'] + html_out_paths)
  File "../../third_party/node/node.py", line 27, in RunNode
    raise RuntimeError('%s failed: %s' % (cmd, stderr))
RuntimeError: ../../third_party/node/linux/node-linux-x64/bin/node ../../third_party/node/node_modules/polymer-css-build/bin/polymer-css-build --polymer-version 1 --no-inline-includes -f /mnt/NewHDD/iMX8M-EVK/imx-yocto-bsp/build-wayland/tmp/work/aarch64-mx8m-poky-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_downloads/bundled/vulcanized.html -o /mnt/NewHDD/iMX8M-EVK/imx-yocto-bsp/build-wayland/tmp/work/aarch64-mx8m-poky-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_downloads/vulcanized.html failed: Inspector is not available

[28/31902] python ../../chrome/browser/resources/optimize_webui.py --host bookmarks --input ../../chrome/browser/resources/md_bookmarks --out_folder gen/chrome/browser/resources/md_bookmarks --depfile gen/chrome/browser/resources/md_bookmarks/build.d --html_in_files bookmarks.html --html_out_files vulcanized.html --html_out_files_polymer2 vulcanized.p2.html --js_out_files crisper.js
FAILED: gen/chrome/browser/resources/md_bookmarks/vulcanized.html gen/chrome/browser/resources/md_bookmarks/vulcanized.p2.html gen/chrome/browser/resources/md_bookmarks/crisper.js 
python ../../chrome/browser/resources/optimize_webui.py --host bookmarks --input ../../chrome/browser/resources/md_bookmarks --out_folder gen/chrome/browser/resources/md_bookmarks --depfile gen/chrome/browser/resources/md_bookmarks/build.d --html_in_files bookmarks.html --html_out_files vulcanized.html --html_out_files_polymer2 vulcanized.p2.html --js_out_files crisper.js
Traceback (most recent call last):
  File "../../chrome/browser/resources/optimize_webui.py", line 266, in <module>
    main(sys.argv[1:])
  File "../../chrome/browser/resources/optimize_webui.py", line 249, in main
    manifest_out_path = _optimize(args.input, args)
  File "../../chrome/browser/resources/optimize_webui.py", line 206, in _optimize
    _polymer_css_build(out_path, crisper_html_out_paths, args.html_out_files, 1)
  File "../../chrome/browser/resources/optimize_webui.py", line 225, in _polymer_css_build
    html_in_paths + ['-o'] + html_out_paths)
  File "../../third_party/node/node.py", line 27, in RunNode
    raise RuntimeError('%s failed: %s' % (cmd, stderr))
RuntimeError: ../../third_party/node/linux/node-linux-x64/bin/node ../../third_party/node/node_modules/polymer-css-build/bin/polymer-css-build --polymer-version 1 --no-inline-includes -f /mnt/NewHDD/iMX8M-EVK/imx-yocto-bsp/build-wayland/tmp/work/aarch64-mx8m-poky-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_bookmarks/bundled/vulcanized.html -o /mnt/NewHDD/iMX8M-EVK/imx-yocto-bsp/build-wayland/tmp/work/aarch64-mx8m-poky-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_bookmarks/vulcanized.html failed: Inspector is not available

[29/31902] aarch64-poky-linux-g++  --sysroot=/mnt/NewHDD/iMX8M-EVK/imx-yocto-bsp/build-wayland/tmp/work/aarch64-mx8m-poky-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot -MMD -MF obj/ppapi/cpp/objects/pass_file_handle.o.d -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DNO_TCMALLOC -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -Wall -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O2  [... cut ...] x8m-poky-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot-native= -g1 -c ../../ppapi/cpp/private/net_address_private.cc -o obj/ppapi/cpp/objects/net_address_private.o
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile (log file is located at /mnt/NewHDD/iMX8M-EVK/imx-yocto-bsp/build-wayland/tmp/work/aarch64-mx8m-poky-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/temp/log.do_compile.5897)

rakuco, 2018-12-07:
> RuntimeError: ../../third_party/node/linux/node-linux-x64/bin/node ../../third_party/node/node_modules/polymer-css-build/bin/polymer-css-build --polymer-version 1 --no-inline-includes -f [...]

See https://github.com/OSSystems/meta-browser/issues/195#issuecomment-440590439

caroldesjardins, 2018-12-07:
On my side, the recipe chromium-x11 compiled perfectly, but the wayland version didn't.