OSSystems/meta-browser #143: compile issue for dependence.

13824125580, 2018-07-24:
i meet a compile issue about "libwayland-egl.so" dependency with the default bbfile of this repo.
alghough it can be skipped by add "INSANE_SKIP_${PN}+= "file_rdeps", but i dont want to do that, 
why arise this error with the default  chromium-ozone bbfile?  how to dealt with this problem?

thanks for your support.

Initialising tasks: 100% |###########################################################################################################################################################################| Time: 0:00:13
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: chromium-ozone-wayland-64.0.3274.0.r517731.igalia.1-r0 do_package_qa: QA Issue: /usr/lib/chromium/chromium-bin contained in package chromium-ozone-wayland requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_chromium-ozone-wayland? [file-rdeps]
ERROR: chromium-ozone-wayland-64.0.3274.0.r517731.igalia.1-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: chromium-ozone-wayland-64.0.3274.0.r517731.igalia.1-r0 do_package_qa: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /home/zilong/WorkSpace/disk1/glibc/build/tmp/work/aarch64-poky-linux/chromium-ozone-wayland/64.0.3274.0.r517731.igalia.1-r0/temp/log.do_package_qa.3728
ERROR: Task (/home/zilong/WorkSpace/disk1/glibc/sources/meta-tina/recipes-web/recipes-browser/chromium/chromium-ozone-wayland_64.0.3274.0.r517731.igalia.1.bb:do_package_qa) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4607 tasks of which 4584 didn't need to be rerun and 1 failed.

rakuco, 2018-07-24:
over to @msisov

kraj, 2018-07-24:
two things to check

1. libwayland-egl.so is provided by graphics drivers usually and make sure that the SONAME inside the file is correct. you can use readelf -d to dump this file and see the header

2. See if .so is packaged in -dev package accidentally.

13824125580, 2018-07-24:
thanks @kraj 
i am sure the so-name is correct, and  i believe it is very like the 2nd reason you said.
so, did this mean that the .so should not be packaged in -dev package? which package should it reside in?
thanks for your kindly help.

kraj, 2018-07-24:
if the shared object is unversioned which is case in many of prebuilt objects then you should force it into main package probably PN, default OE packaging rules will push it into -dev package since thats what libtool generated .so are just symlinks to real versioned .so.X.Y.Z files

you might change the packaging of graphics recipe which is providing it.

```
SOLIBS = ".so"
FILES_SOLIBSDEV = ""
```

13824125580, 2018-07-25:
thanks you all

closed 2018-07-25 by 13824125580