OSSystems/meta-browser #149: does the error output after launch matter?

13824125580, 2018-08-01:
[   62.994655] sunxi-bt bt: check bluetooth io_regulator voltage: 1800000

[1952:1973:0101/000223.543868:ERROR:bus.cc(395)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory
[1952:1952:0101/000223.797757:ERROR:browser_process_platform_part_ozone.cc(20)] Not implemented reached in virtual void BrowserProcessPlatformPart::RegisterInProcessServices(content::Content)
[1952:2006:0101/000224.044825:ERROR:xkb_keyboard_layout_engine.cc(694)] Not implemented reached in virtual bool ui::XkbKeyboardLayoutEngine::SetCurrentLayoutByName(const string&)
[1952:2007:0101/000224.061897:ERROR:gl_surface_egl.cc(314)] eglChooseConfig failed with error EGL_BAD_ATTRIBUTE
[1952:2028:0101/000224.438268:ERROR:bus.cc(395)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1952:2006:0101/000224.547272:ERROR:wayland_keyboard.cc(140)] Not implemented reached in static void ui::WaylandKeyboard::RepeatInfo(void*, wl_keyboard*, int32_t, int32_t)
[1952:2007:0101/000224.764758:ERROR:gl_surface_egl.cc(314)] eglChooseConfig failed with error EGL_BAD_ATTRIBUTE
[1952:1952:0101/000224.774233:ERROR:layer_tree_host_impl.cc(2508)] Forcing zero-copy tile initialization as worker context is missing
[1952:2031:0101/000225.007757:ERROR:cert_verify_proc_nss.cc(922)] CERT_PKIXVerifyCert for www.gstatic.com failed err=-8181
[1952:2031:0101/000225.145284:ERROR:cert_verify_proc_nss.cc(922)] CERT_PKIXVerifyCert for translate.googleapis.com failed err=-8181
[2035:2061:0101/000225.376187:ERROR:layer_tree_host_impl.cc(2508)] Forcing zero-copy tile initialization as worker context is missing
[1952:2031:0101/000230.349124:ERROR:cert_verify_proc_nss.cc(922)] CERT_PKIXVerifyCert for www.google.com failed err=-8181

rakuco, 2018-08-01:
The bus.cc errors are related to d-bus. It's not clear if you just need to have it running or your setup is using different paths from what Chromium expects.

13824125580, 2018-08-02:
is this matter?  and did the following error related the d-bus error？ such as xkb_keyboard_layout_engine.cc and wayland errors

closed 2018-08-06 by 13824125580