OSSystems/meta-browser #163: chromium-x11 do_compile: ninja: build stopped: subcommand failed on rocko

nishantpoorswani, 2018-08-23:
Hi, I'm using the latest meta-browser(master) . It fails during the build of chromium-x11_68.0.3440.106 for aarch64 (hikey960). I'm using the following configurations. 

Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "aarch64-linaro-linux"
MACHINE              = "hikey960"
DISTRO               = "rpb"
DISTRO_VERSION       = "2.4+linaro"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta-rpb             = "HEAD:96d618f2597d18da1fc868a0f801854f0f2161e6"
meta-oe              
meta-gnome           
meta-xfce            
meta-initramfs       
meta-multimedia      
meta-networking      
meta-webserver       
meta-filesystems     
meta-perl            
meta-python          = "HEAD:352531015014d1957d6444d114f4451e241c4d23"
meta-browser-master  = "<unknown>:<unknown>"
meta-qt5             = "HEAD:682ad61c071a9710e9f9d8a32ab1b5f3c14953d1"
meta-virtualization  = "HEAD:b704c689b67639214b9568a3d62e82df27e9434f"
meta-clang           = "HEAD:533f882e77714a231a1cef985fd9bb7f56e44145"
meta-96boards        = "HEAD:7f473a9d0028911a04b6e797d5dc9a0c638bfac1"
meta-intel           = "HEAD:9baa0974b397a8520d548a5f7b768db1acc34177"
meta-qcom            = "HEAD:432e3b1d881bf91a5f5537387a68d3e2cb88bccd"
meta-st-cannes2      = "HEAD:88fadd917c3f5b7e0feba318aa2d7bbb596eb95f"
meta-ti              = "HEAD:1f781998aa2d3f3526aa6e3c7a8384bcf54b800d"
meta-linaro          
meta-linaro-toolchain 
meta-optee           = "HEAD:75dfb67bbb14a70cd47afda9726e2e1c76731885"
meta-backports       = "HEAD:4b51a7a0e950ff2d4bd62e045471a44dc940ad62"
meta                 = "HEAD:bdf13518e79ab949c4320226a399ee4a3913ee30"

The gcc version is 7.1
I get the following error | ninja: build stopped: subcommand failed. 
The complete log is attached here: 
[log.do_compile.log](https://github.com/OSSystems/meta-browser/files/2313790/log.do_compile.log)

Thanking you in advance.
Regards:Nishant Poorswani

kraj, 2018-08-23:
```text
../../third_party/skia/third_party/skcms/src/Transform_inl.h:459:32: internal compiler error: in convert_move, at expr.c:259
     v = (U16)( ((v<<8)|(v>>8)) & 0xffff );
              ~~~~~~~~~~~~~~~~~~^~~~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
```

there is a compiler crash and you seem to be using Linaro gcc. So please work with Linaro folks
to get this sorted.

kraj, 2018-08-27:
can you remove meta-linaro* from your bblayers.conf and see if that helps.

nishantpoorswani, 2018-08-27:
I removed all the meta-linaro layers and compiled again. It still fails.  Here is my bblayers.conf: 
# LAYER_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
LCONF_VERSION = "7"
OEROOT := "${@os.path.abspath(os.path.dirname(d.getVar('FILE', True)))}/../.."

BBPATH = "${TOPDIR}"

BBFILES = ""

# These layers hold recipe metadata not found in OE-core, but lack any machine or distro content
BASELAYERS ?= " \
  ${OEROOT}/layers/meta-openembedded/meta-oe \
  ${OEROOT}/layers/meta-openembedded/meta-gnome \
  ${OEROOT}/layers/meta-lxde \
  ${OEROOT}/layers/meta-openembedded/meta-xfce \
  ${OEROOT}/layers/meta-openembedded/meta-initramfs \
  ${OEROOT}/layers/meta-openembedded/meta-multimedia \
  ${OEROOT}/layers/meta-openembedded/meta-networking \
  ${OEROOT}/layers/meta-openembedded/meta-webserver \
  ${OEROOT}/layers/meta-openembedded/meta-filesystems \
  ${OEROOT}/layers/meta-openembedded/meta-perl \
  ${OEROOT}/layers/meta-openembedded/meta-python \
  ${OEROOT}/layers/meta-browser \
  ${OEROOT}/layers/meta-qt5 \
  ${OEROOT}/layers/meta-virtualization \
  ${OEROOT}/layers/meta-clang \
  ${OEROOT}/layers/meta-qt5-extra \
"

# These layers hold machine specific content, aka Board Support Packages
BSPLAYERS ?= " \
  ${OEROOT}/layers/meta-96boards \
  ${OEROOT}/layers/meta-intel \
  ${OEROOT}/layers/meta-qcom \
  ${OEROOT}/layers/meta-st-cannes2 \
  ${OEROOT}/layers/meta-ti \
"

# Add your overlay location to EXTRALAYERS
# Make sure to have a conf/layers.conf in there
EXTRALAYERS ?= " \
  ${OEROOT}/layers/meta-backports \
"

BBLAYERS = " \
  ${OEROOT}/layers/meta-rpb \
  ${BASELAYERS} \
  ${BSPLAYERS} \
  ${EXTRALAYERS} \
  ${OEROOT}/layers/openembedded-core/meta \
"

Here is my logs: 
[log.do_compile.tar.gz](https://github.com/OSSystems/meta-browser/files/2323378/log.do_compile.tar.gz)

Thank you for the prompt replies.

kraj, 2018-08-27:
```
aarch64-oe-linux-g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
```
gcc is crashing.

how much RAM do you have on your build system

nishantpoorswani, 2018-08-27:
I have 8Gb RAM. I can allocate more and try.

kraj, 2018-08-27:
Try using 16G minimum when building Chromium. See

[system requirements](https://chromium.googlesource.com/chromium/src/+/lkcr/docs/linux_build_instructions.md#system-requirements)

nishantpoorswani, 2018-09-06:
Hi I increased the ram to 16 Gb and it compiled successfully. Thank you so much for your help.

closed 2018-09-06 by nishantpoorswani