OSSystems/meta-browser #166: build failed on poky master with proprietary-codecs enabled

fthiery, 2018-09-04:
Lately meta-browser master fails to build against poky master:
[log.do_compile.9181.log](https://github.com/OSSystems/meta-browser/files/2348545/log.do_compile.9181.log)

Note that proprietary-codecs option is enabled:
```
PACKAGECONFIG_append = " proprietary-codecs"
```

[log.do_compile.9181.log](https://github.com/OSSystems/meta-browser/files/2348555/log.do_compile.9181.log)
Log excerpt:
```
[7577/33157] x86_64-poky-linux-g++  -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0/recipe-sysroot -MMD -MF obj/third_party/openh264/common/WelsThreadPool.o.d -DX86_ASM -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../../third_party/openh264/src/codec/api/svc -I../../third_party/openh264/src/codec/common/arm -I../../third_party/openh264/src/codec/common/inc -I../../third_party/openh264/src/codec/common/src -I../.. -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -Wno-format -Wno-header-hygiene -Wno-unused-function -Wno-unused-value -std=gnu++14 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wno-narrowing  -O2 -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0=/usr/src/debug/chromium-x11/68.0.3440.106-r0 -fdebug-prefix-map=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0/recipe-sysroot= -fdebug-prefix-map=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0/recipe-sysroot-native= -g1 -c ../../third_party/openh264/src/codec/common/src/WelsThreadPool.cpp -o obj/third_party/openh264/common/WelsThreadPool.o
FAILED: obj/third_party/openh264/common/WelsThreadPool.o 
x86_64-poky-linux-g++  -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0/recipe-sysroot -MMD -MF obj/third_party/openh264/common/WelsThreadPool.o.d -DX86_ASM -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../../third_party/openh264/src/codec/api/svc -I../../third_party/openh264/src/codec/common/arm -I../../third_party/openh264/src/codec/common/inc -I../../third_party/openh264/src/codec/common/src -I../.. -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -Wno-format -Wno-header-hygiene -Wno-unused-function -Wno-unused-value -std=gnu++14 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wno-narrowing  -O2 -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0=/usr/src/debug/chromium-x11/68.0.3440.106-r0 -fdebug-prefix-map=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0/recipe-sysroot= -fdebug-prefix-map=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0/recipe-sysroot-native= -g1 -c ../../third_party/openh264/src/codec/common/src/WelsThreadPool.cpp -o obj/third_party/openh264/common/WelsThreadPool.o
cc1plus: error: -Wformat-security ignored without -Wformat [-Werror=format-security]
cc1plus: warning: unrecognized command line option '-Wno-header-hygiene'
cc1plus: some warnings being treated as errors
[7578/33157] python ../../third_party/yasm/run_yasm.py yasm -DPIC -felf64 -m amd64 -I../../third_party/libvpx/source/config -I../../third_party/libvpx/source/config/linux/x64 -I../../third_party/libvpx/source/libvpx -I. -I../.. -Igen -DCHROMIUM -o obj/third_party/libvpx/libvpx_yasm/highbd_sad4d_sse2.o ../../third_party/libvpx/source/libvpx/vpx_dsp/x86/highbd_sad4d_sse2.asm
[7579/33157] x86_64-poky-linux-gcc  -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0/recipe-sysroot -MMD -MF obj/third_party/libvpx/libvpx_intrinsics_sse2/fwd_txfm_sse2.o.d -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -I../../third_party/libvpx/source/config -I../../third_party/libvpx/source/config/linux/x64 -I../../third_party/libvpx/source/libvpx -msse2 -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-function -Wno-sign-compare -std=gnu11  -O2 -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0=/usr/src/debug/chromium-x11/68.0.3440.106-r0 -fdebug-prefix-map=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0/recipe-sysroot= -fdebug-prefix-map=/home/gitlab-runner/yocto/mediacoder-yocto/build/tmp/work/corei7-64-poky-linux/chromium-x11/68.0.3440.106-r0/recipe-sysroot-native= -g1 -c ../../third_party/libvpx/source/libvpx/vpx_dsp/x86/fwd_txfm_sse2.c -o obj/third_party/libvpx/libvpx_intrinsics_sse2/fwd_txfm_sse2.o
ninja: build stopped: subcommand failed.

```

kraj, 2018-09-04:
a workaround is
```
SECURITY_STRINGFORMAT = ""
```

real fix it to add ```-Wformat``` into the chromium build system

fthiery, 2018-09-05:
Thanks. 

I can't seem to pin down what exactly changed recently in poky/oe that triggers these failures, because the SECURITY_STRINGFORMAT in https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/conf/distro/include/security_flags.inc#n16 did not change since 2016 (30f9a5072d9); i know for sure this was building okay a few weeks back.

kraj, 2018-09-05:
poky defaults now to use hardening flags since few weeks

fthiery, 2018-09-05:
Alright, thanks. Should we create an upstream ticket ? I don't feel overly confident doing that.

Is that the actual involved commit ? https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=491082c56ce34f3fd644f8d4457ccd52af951087

kraj, 2018-09-05:
yes thats the commit which changed the defaults. Filing ticket upstream should be fine. However I think patches for these fixes might already be in upstream openh264 so check it there and then may be propose the backports to chromium

fthiery, 2018-09-07:
The problem seems to be caused by the -Wno-header-hygiene flag. I think i managed to reproduce the issue with openh264 1.8 (master):

```
$ git diff
diff --git a/build/platform-linux.mk b/build/platform-linux.mk
index 52230a2a..90a08dce 100644
--- a/build/platform-linux.mk
+++ b/build/platform-linux.mk
@@ -3,7 +3,7 @@ SHAREDLIBSUFFIX = so
 SHAREDLIBSUFFIXFULLVER=$(SHAREDLIBSUFFIX).$(FULL_VERSION)
 SHAREDLIBSUFFIXMAJORVER=$(SHAREDLIBSUFFIX).$(SHAREDLIB_MAJORVERSION)
 SHLDFLAGS = -Wl,-soname,$(LIBPREFIX)$(PROJECT_NAME).$(SHAREDLIBSUFFIXMAJORVER)
-CFLAGS += -Wall -fno-strict-aliasing -fPIC -MMD -MP
+CFLAGS += -Wall -fno-strict-aliasing -fPIC -MMD -MP -Wno-header-hygiene
 LDFLAGS += -lpthread
 STATIC_LDFLAGS += -lpthread -lm
 AR_OPTS = crD $@
$ make ARCH=x86_64
...
g++ -O3 -DNDEBUG -m64 -DX86_ASM -DHAVE_AVX2 -Wall -fno-strict-aliasing -fPIC -MMD -MP -Wno-header-hygiene -DGENERATED_VERSION_HEADER -DHAVE_AVX2 -I./codec/api/svc -I./codec/common/inc -Icodec/common/inc  -I./codec/encoder/core/inc -I./codec/encoder/plus/inc -I./codec/processing/interface -c -o codec/encoder/core/src/encoder_ext.o codec/encoder/core/src/encoder_ext.cpp
codec/encoder/core/src/encoder_ext.cpp: In function 'int32_t WelsEnc::WelsInitEncoderExt(WelsEnc::sWelsEncCtx**, WelsEnc::SWelsSvcCodingParam*, SLogContext*, WelsEnc::SExistingParasetList*)':
codec/encoder/core/src/encoder_ext.cpp:2332:70: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of non-trivially copyable type 'WelsEnc::SWelsSvcCodingParam' {aka 'struct WelsEnc::TagWelsSvcCodingParam'}; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
   memcpy (pCtx->pSvcParam, pCodingParam, sizeof (SWelsSvcCodingParam)); // confirmed_safe_unsafe_usage
                                                                      ^
In file included from ./codec/encoder/core/inc/encoder_context.h:45,
                 from ./codec/encoder/core/inc/encoder.h:43,
                 from codec/encoder/core/src/encoder_ext.cpp:41:
./codec/encoder/core/inc/param_svc.h:106:16: note: 'WelsEnc::SWelsSvcCodingParam' {aka 'struct WelsEnc::TagWelsSvcCodingParam'} declared here
 typedef struct TagWelsSvcCodingParam: SEncParamExt {
                ^~~~~~~~~~~~~~~~~~~~~
codec/encoder/core/src/encoder_ext.cpp: In function 'int32_t WelsEnc::WelsEncoderApplyLTR(SLogContext*, WelsEnc::sWelsEncCtx**, SLTRConfig*)':
codec/encoder/core/src/encoder_ext.cpp:4457:70: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of non-trivially copyable type 'WelsEnc::SWelsSvcCodingParam' {aka 'struct WelsEnc::TagWelsSvcCodingParam'}; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
   memcpy (&sConfig, (*ppCtx)->pSvcParam, sizeof (SWelsSvcCodingParam));
                                                                      ^
In file included from ./codec/encoder/core/inc/encoder_context.h:45,
                 from ./codec/encoder/core/inc/encoder.h:43,
                 from codec/encoder/core/src/encoder_ext.cpp:41:
./codec/encoder/core/inc/param_svc.h:106:16: note: 'WelsEnc::SWelsSvcCodingParam' {aka 'struct WelsEnc::TagWelsSvcCodingParam'} declared here
 typedef struct TagWelsSvcCodingParam: SEncParamExt {
                ^~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-header-hygiene'
```

The meson build included in openh264 1.8 seems to just drop this option which seems to be unsupported by the C++ compiler alltogether (according to my g++ man page at least): https://github.com/cisco/openh264/blob/master/meson.build#L46

```$ git diff meson.build 
diff --git a/meson.build b/meson.build
index c5793dbc..59421d8c 100644
--- a/meson.build
+++ b/meson.build
@@ -46,7 +46,9 @@ cpu_family = host_machine.cpu_family()
 supported_arguments = cpp.get_supported_arguments([
   '-Wno-non-virtual-dtor',
   '-Wno-class-memaccess',
-  '-Wno-strict-aliasing'])
+  '-Wno-strict-aliasing',
+  '-Wno-header-hygiene',
+  ])
 
 add_project_arguments(supported_arguments, language: 'cpp')
```
Looking at ninja -v shows that the -Wno-header-hygiene is not passed at all.

I will open an upstream ticket to drop unsupported compiler flags in the non-meson build system.

otavio, 2018-09-07:
Are you going to send a PR fixing it? Please do :-)

fthiery, 2018-09-20:
I continued digging, and found out the issue is because of the following combination (and not Wno-header-hygiene, sorry about that)

```
-Wformat -Wformat-security -Werror=format-security -Wno-format
```

I guess that the combined -Wformat and -Wno-format are contradicting each other. I'm not sure where this is being added (in the yocto or chromium build system). @kraj what do you think ?

You can replicate on the openh264 root by doing:
```
LANG=C g++ -O3 -DNDEBUG -m64 -DX86_ASM -DHAVE_AVX2 -Wall -fno-strict-aliasing -fPIC -MMD -MP -DGENERATED_VERSION_HEADER -DHAVE_AVX2 -I./codec/api/svc -I./codec/common/inc -Icodec/common/inc  -I./codec/encoder/core/inc -I./codec/encoder/plus/inc -I./codec/processing/interface -c -o codec/encoder/core/src/encoder_ext.o codec/encoder/core/src/encoder_ext.cpp -Wformat -Wformat-security -Werror=format-security -Wno-format
cc1plus: error: -Wformat-security ignored without -Wformat [-Werror=format-security]
cc1plus: some warnings being treated as errors
```

kraj, 2018-09-20:
Apply the fix I suggested in earlier comment to disable security warnings

fthiery, 2018-09-21:
Yes, that's what i did and it solves it but i was trying to help finding the proper fix :) Having -Wformat and -Wno-format sounds strange to me.

According to https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html

"Option -Wformat is equivalent to -Wformat=1, and -Wno-format is equivalent to -Wformat=0. "

So something here is adding both options at the same time.

fthiery, 2018-09-21:
Okay so it seems that chromium is manually forcing -Wno-format on the openh264 compilation flags:

```
./out/Release/obj/third_party/openh264/processing.ninja:cflags = -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -Wno-format -Wno-header-hygiene -Wno-unused-function -Wno-unused-value
./out/Release/obj/third_party/openh264/common.ninja:cflags = -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -Wno-format -Wno-header-hygiene -Wno-unused-function -Wno-unused-value
./out/Release/obj/third_party/openh264/encoder.ninja:cflags = -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -Wno-format -Wno-header-hygiene -Wno-unused-function -Wno-unused-value
```

What do you think should be done to fix this properly ?

kraj, 2018-09-21:
1. stop chromium to force -Wno-format
2. create fixes in source to overcome the warnings

rakuco, 2018-09-24:
> The `-Wno-header-hygiene` bit is just a warning that shouldn't cause the build to fail.

(and the warning is only shown if there's a previous diagnostic message, which should've been fixed by the CL above)

fthiery, 2018-09-24:
> The `-Wno-header-hygiene` bit is just a warning that shouldn't cause the build to fail.

Yes, as i indicated earlier, the issue is not with "-Wno-header-hygiene" as i suspected initially, but with the fact that yocto adds "-Wformat" but chromium unconditionally adds "-Wno-format". These two options are basically contradicting each other.

I'll try again with M69.

fthiery, 2018-09-25:
Looks like it compiled, let me double check.

fthiery, 2018-09-27:
Yay, i triple checked and it's now working. Thanks @rakuco

closed 2018-09-27 by fthiery

rakuco, 2018-09-27:
Awesome, thanks for checking!