OSSystems/meta-browser #167: did the chromium port supports running more than one instances simultantously？

13824125580, 2018-09-11:
i try to run the chromium browser more than one inistances, but got the error output from the second one, the error ouput as follows.
so, did this port support more than one instance running simultantously like chromium on the desktop ?
thanks for your support.!
root@tulip-mozart:/# chromium --no-sandbox
[1554:1554:0101/051007.512147:ERROR:browser_process_platform_part_ozone.cc(20)] Not implemented reached in virtual void BrowserProcessPlatformPart::RegisterInProcessServices(content::ContentBrowserClient::StaticServiceMap*)
[1554:1584:0101/051007.642023:ERROR:xkb_keyboard_layout_engine.cc(694)] Not implemented reached in virtual bool ui::XkbKeyboardLayoutEngine::SetCurrentLayoutByName(const string&)
[1554:1585:0101/051007.647838:ERROR:gl_surface_egl.cc(314)] eglChooseConfig failed with error EGL_BAD_ATTRIBUTE
Created new window in existing browser session.
Segmentation fault
root@tulip-mozart:/# chromium --no-sandbox
[1612:1612:0101/051026.699049:ERROR:browser_process_platform_part_ozone.cc(20)] Not implemented reached in virtual void BrowserProcessPlatformPart::RegisterInProcessServices(content::ContentBrowserClient::StaticServiceMap*)
[1612:1642:0101/051026.827107:ERROR:xkb_keyboard_layout_engine.cc(694)] Not implemented reached in virtual bool ui::XkbKeyboardLayoutEngine::SetCurrentLayoutByName(const string&)
[1612:1643:0101/051026.835510:ERROR:gl_surface_egl.cc(314)] eglChooseConfig failed with error EGL_BAD_ATTRIBUTE
Created new window in existing browser session.
Segmentation fault

msisov, 2018-09-11:
For of all, if you want to have a second browser instance, you have to provide a different --usr-data-dir=/here/comes/the/path

Secondly, seg faults are not related to that.
What meta-browser are you using? Is it Intel's chromium ozone port? We haven't got browser_process_platform_part_ozone.cc file for ages...

13824125580, 2018-09-11:
@msisov  i use the chromium-ozone-wayland port of this repo, 64 version .
what do you mean have not got browser_process_platform_part_ozone.cc?  i use the exactly 'chromium-ozone-wayland" port of this repo.

msisov, 2018-09-11:
sorry. I forgot it's that old :)) I have almost completed a M71 recipe suitable for this repo.

msisov, 2018-09-11:
it's our https://github.com/igalia/chromium branch, which has ~85% of patches upstreamed. 70-80% of the time we work in the upstream now. But the recipe is always based on our downstream repo, which is rebased on a weekly basis.

I'm build the recipe against sumo branch for rpi3 now. Will send it for review tomorrow.

13824125580, 2018-09-12:
which chromium version it based on?  i want to apply your " https://github.com/igalia/chromium“　on my platform allwinner A64 soc, how to start? is any guide to integrate this code to yocto and compile it?

thanks for your kindly support!

msisov, 2018-09-12:
`which is rebased on a weekly basis.`

closed 2018-09-15 by 13824125580