OSSystems/meta-browser #169: be stucked in "kiosk" mode

13824125580, 2018-09-12:
hi foks:
  i am using ozone-wayland port in my project, for some reasons, i need running the chromium in full screen "kiosk mode”, i run the command like that:
"chromium --no-sandbox --kiosk"
but nothing happens on the screen , it seems the lauch process has been stucked for some reason.
so, did any precondition needed for the chrome running in "kiosk" mode?  please give me some hints for this, thank you!

twoerner, 2018-09-12:
when building chromium, there's a PACKAGECONFIG for that: kiosk-mode
https://github.com/OSSystems/meta-browser/blob/master/recipes-browser/chromium/chromium-gn.inc#L76

13824125580, 2018-09-12:
@twoerner , i open the "kiosk-mode" and recompile the chromium.
the screen turns out nothing change still like before, and the terminal stucked, from the "ps -aux" command , it shows the chromium has been running already,  i dont know why the gui not showup.
need this operations need waiting too much?

[1260:1271:0101/014212.017164:ERROR:bus.cc(395)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory
[1260:1260:0101/014212.273513:ERROR:browser_process_platform_part_ozone.cc(20)] Not implemented reached in virtual void BrowserProcessPlatformPart::RegisterInProcessServices(content::ContentBrowserClient::StaticS)
[1260:1291:0101/014212.510357:ERROR:xkb_keyboard_layout_engine.cc(694)] Not implemented reached in virtual bool ui::XkbKeyboardLayoutEngine::SetCurrentLayoutByName(const string&)
[1260:1292:0101/014212.526338:ERROR:gl_surface_egl.cc(314)] eglChooseConfig failed with error EGL_BAD_ATTRIBUTE
[1260:1302:0101/014212.901096:ERROR:bus.cc(395)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1260:1291:0101/014213.018942:ERROR:wayland_keyboard.cc(140)] Not implemented reached in static void ui::WaylandKeyboard::RepeatInfo(void*, wl_keyboard*, int32_t, int32_t)
[1260:1260:0101/014213.273089:ERROR:fullscreen_ozone.cc(12)] Not implemented reached in bool IsFullScreenMode()
[1260:1292:0101/014213.308524:ERROR:gl_surface_egl.cc(314)] eglChooseConfig failed with error EGL_BAD_ATTRIBUTE
[1260:1260:0101/014213.316651:ERROR:layer_tree_host_impl.cc(2508)] Forcing zero-copy tile initialization as worker context is missing

13824125580, 2018-09-12:
did the line 
[1260:1260:0101/014213.273089:ERROR:fullscreen_ozone.cc(12)] Not implemented reached in bool IsFullScreenMode()

mean something ?

msisov, 2018-09-12:
[1260:1260:0101/014213.273089:ERROR:fullscreen_ozone.cc(12)] Not implemented reached in bool IsFullScreenMode() nope. Ignore it.

The problem here is in
[1260:1292:0101/014212.526338:ERROR:gl_surface_egl.cc(314)] eglChooseConfig failed with error EGL_BAD_ATTRIBUTE

13824125580, 2018-09-12:
@msisov  work in normal mode with command "chromium --no-sandbox" still has this error ouput
[7735:7777:0912/100518.566886:ERROR:gl_surface_egl.cc(314)] eglChooseConfig failed with error EGL_BAD_ATTRIBUTE
 
is this matter?

13824125580, 2018-09-14:
fixed with your 70 chromium, thank you.

closed 2018-09-14 by 13824125580