OSSystems/meta-browser #176: chromium 71 would not get into the kiosk mode

13824125580, 2018-09-27:
hi folks:
i use the latest chromium ozone wayland port 71 in your igilia repo .
when i launch it with command"
chromium --no-sandbox --kiosk
and found that the address bar and tools menu still hung it and did not work in full screen mode.

that is different from the previous version of 70, so, is this expected?

msisov, 2018-09-27:
Please file a bug in the correct repo.

13824125580, 2018-09-27:
@msisov 
could you tell me which place shall i file the bug?

msisov, 2018-09-27:
I’d suggest to file a bug in the upstream bug tracker as long as the work is currently happens mostly there.

https://bugs.chromium.org/p/chromium/issues/list 

File a bug and add me (msisov@igalia.com) to CC.

otavio, 2018-10-19:
This is clearly not related here, closing it.

closed 2018-10-19 by otavio

msisov, 2018-10-19:
And it’s fixed in the upstream, btw. The fix will be a part of m72 recipe