OSSystems/meta-browser #180: Occur error when execute `bitbake chromium-ozone-wayland`

ayunran4, 2018-10-05:
```
Log data follows:    
| DEBUG: Executing shell function do_configure    
| ERROR at //ui/events/BUILD.gn:80:12: Assignment had no effect.  
|   friend = [ ":*" ]  
|            ^------  
| You set the variable "friend" here and it was unused before it went  
| out of scope.  
| See //BUILD.gn:211:7: which caused the file to be included.  
|       "//ui/events:events_unittests",  
|       ^-----------------------------  
| WARNING: exit code 1 from a shell command.  
| ERROR: Function failed: do_configure 
```
I was followed https://github.com/Igalia/meta-browser/wiki.
When I execute `bitbake chromium-ozone-wayland`, above error was occured.
How can I solved this error?

rakuco, 2018-10-05:
Can you confirm you're using the chromium-ozone-wayland from this repository and not the one from Igalia's meta-browser? And what branch are you using?

msisov, 2018-10-05:
make sure the gn used is up-to-date.

ayunran4, 2018-10-05:
@rakuco I'm using this repository, not Igalia's. And branch is master.
@msisov How can I check the gn used is up-to-date?

ayunran4, 2018-10-05:
I checked I'm using repository pushed latest commit.
My yocto verson is pyro. I runed `bitbake -c configure chromium-ozone-wayland`, but it doesn't work.

rakuco, 2018-10-05:
Hmm, even if you do get it to configure, I'm pretty sure you will run into build issues due to pyro's GCC version -- we explicitly started requiring GCC 7.1 and rocko a while ago due to pyro's GCC 6 being too outdated.

rakuco, 2018-10-05:
(you could try using clang instead though)

ayunran4, 2018-10-05:
Sorry, I never heard about clang. Could you tell me what is clang?

rakuco, 2018-10-05:
It's a different C++ compiler.

ayunran4, 2018-10-05:
Got it! Thanks for very fast respond:)

kraj, 2018-10-05:
FWIW, clang is the default compiler user for developing chromium now a days.

kraj, 2018-10-05:
@rakuco do we still support pyro for latest meta-browser master ?

rakuco, 2018-10-08:
> @rakuco do we still support pyro for latest meta-browser master ?

Only if meta-clang works with it, otherwise #175 should've started requiring a GCC version not present in pyro.

kraj, 2018-10-08:
i have not tested pyro/meta-clang, if someone does and runs into issues I will support

ayunran4, 2018-10-11:
I tested on rocko, but I got a build error.
```
[27/32999] arm-agl-linux-gnueabi-g++  -march=armv7ve -marm -mfpu=neon-vfpv4 -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/media/rubin/ssd/agl6/build/tmp/work/armv7vehf-neon-vfpv4-agl-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot -MMD -MF obj/components/blacklist/opt_out_blacklist/opt_out_blacklist/opt_out_blacklist.o.d -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -march=armv7ve -mfloat-abi=hard -mtune=generic-armv7-a -mfpu=neon-vfpv4 -mthumb -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -std=gnu++14 -Wno-narrowing -fno-exceptions -fno-rtti -fvisibility-inlines-hidden  -O2 -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/media/rubin/ssd/agl6/build/tmp/work/armv7vehf-neon-vfpv4-agl-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0=/usr/src/debug/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0 -fdebug-prefix-map=/media/rubin/ssd/agl6/build/tmp/work/armv7vehf-neon-vfpv4-agl-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot-native= -fdebug-prefix-map=/media/rubin/ssd/agl6/build/tmp/work/armv7vehf-neon-vfpv4-agl-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot= -g1 -c ../../components/blacklist/opt_out_blacklist/opt_out_blacklist.cc -o obj/components/blacklist/opt_out_blacklist/opt_out_blacklist/opt_out_blacklist.o
In file included from ../../base/bind.h:10:0,
                 from ../../components/blacklist/opt_out_blacklist/opt_out_blacklist.cc:7:
../../base/bind_internal.h: In function 'bool base::internal::QueryCancellationTraitsImpl(base::internal::BindStateBase::CancellationQueryMode, const Functor&, const BoundArgsTuple&, std::index_sequence<indices ...>) [with Functor = void (blacklist::OptOutBlacklist::*)(std::unique_ptr<blacklist::BlacklistData>); BoundArgsTuple = std::tuple<base::WeakPtr<blacklist::OptOutBlacklist> >; unsigned int ...indices = {0}]':
../../base/bind_internal.h:752:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
```
I need a help! How can I solve it?

rakuco, 2018-10-11:
That's not an error, it's just a warning.

ayunran4, 2018-10-11:
Oh! I saw log again, error is RuntimeError.
```
RuntimeError: ../../third_party/node/linux/node-linux-x64/bin/node ../../third_party/node/node_modules/polymer-css-build/bin/polymer-css-build --polyme    r-version 1 --no-inline-includes -f /media/rubin/ssd/agl6/build/tmp/work/armv7vehf-neon-vfpv4-agl-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589    108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_downloads/bundled/vulcanized.html -    o /media/rubin/ssd/agl6/build/tmp/work/armv7vehf-neon-vfpv4-agl-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-way    land-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_downloads/vulcanized.html failed: Inspector is not available
```
@rakuco Can you help me?

rakuco, 2018-10-11:
It looks like your nodejs was built without ICU support; I started a discussion upstream about this in http://lists.openembedded.org/pipermail/openembedded-devel/2018-September/120658.html but didn't have time to propose a fix yet.

In the meantime, try using @msisov's nodejs recipe from https://github.com/Igalia/meta-browser

otavio, 2019-03-27:
The nodejs has been fixed on recent releases of Yocto Project, so I am closing this now.

closed 2019-03-27 by otavio