OSSystems/meta-browser #183: Compilation errors with chromium-ozone-wayland

NeenaReddy, 2018-10-17:
I am getting errors when building chromium-ozone-wayland using master branch.

In file included from ../../third_party/skia/src/core/SkOpts.cpp:43:0:
../../third_party/skia/src/opts/SkRasterPipeline_opts.h: In function 'neon::F neon::from_half(neon::U16)':
../../third_party/skia/src/opts/SkRasterPipeline_opts.h:662:26: error: cannot convert 'neon::U16 {aka short unsigned int}' to 'float16x4_t {aka __vector(4) __fp16}' for argument '1' to 'float32x4_t vcvt_f32_f16(float16x4_t)'
     return vcvt_f32_f16(h);
                          ^
../../third_party/skia/src/opts/SkRasterPipeline_opts.h: In function 'neon::U16 neon::to_half(neon::F)':
../../third_party/skia/src/opts/SkRasterPipeline_opts.h:682:26: error: cannot convert 'neon::F {aka float}' to 'float32x4_t {aka __vector(4) float}' for argument '1' to 'float16x4_t vcvt_f16_f32(float32x4_t)'
     return vcvt_f16_f32(f);

Looks like there are patches in igalias meta-browser to fix compilation errors. Are there any plans to merge those patches here?

kraj, 2018-10-17:
always post ```Build Configuration``` that is printed by bitbake at top of the build

NeenaReddy, 2018-10-17:
Here is my Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "imx8qmmek"
DISTRO               = "fsl-imx-xwayland"
DISTRO_VERSION       = "4.14-sumo"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta
meta-poky            = "HEAD:36d5cee56bd389ecbe66dbfad9e5d57c13b56b95"
meta-oe
meta-multimedia      = "HEAD:2bb21ef27c4b0c9d52d30b3b2c5a0160fd02b966"
meta-freescale       = "HEAD:407c6cf408969445031a492e2d25e0e2749582ea"
meta-freescale-3rdparty = "HEAD:beed0f08b50be497e3587e33c628fc4f59d8a177"
meta-freescale-distro = "HEAD:f7e2216e93aff14ac32728a13637a48df436b7f4"
meta-browser         = "HEAD:4f39525c52c72e2c6711e0ea0ede79d82fd52daa"
meta-gnome
meta-networking
meta-python
meta-filesystems     = "HEAD:2bb21ef27c4b0c9d52d30b3b2c5a0160fd02b966"
meta-qt5             = "HEAD:d0ef3c0819e03af4982ba6066bcda8967bedb44a"

kraj, 2018-10-17:
@NeenaReddy your meta-browser does not include the fixes that got merged today which fixes issues with compiling chromium/wayland on aarch64. Please update meta-browser to latest and report back

NeenaReddy, 2018-10-17:
@kraj Thank you! i did not check commits today. I will rebuild my chromium-ozone-wayland.

otavio, 2018-10-19:
@NeenaReddy did the commit fix your issue?

NeenaReddy, 2018-10-19:
@otavio yes, it did fix my issue. I am able to build chromium-ozone-wayland now.

closed 2018-10-19 by NeenaReddy