OSSystems/meta-browser #186: chromium-x11: Update to M70

rakuco, 2018-10-19:
Tracking bug to describe progress with the M70 update. I've got things building locally, but I still need to finish cleaning up some patches.

Additionally, a nodejs recipe built with support for internationalization is required now (as the chromium-ozone-wayland recipe already does). I [emailed oe-devel a while ago about this](http://lists.openembedded.org/pipermail/openembedded-devel/2018-September/120658.html), but haven't had time to write a proper patch yet.

I'll be at TPAC next week, and probably won't have time to make much progress until I come back.

shr-project, 2018-10-19:
Wasn't nodejs already resolved by Khem in:
http://git.openembedded.org/meta-openembedded/commit/?id=d7d0cc5227d0dc7d3ff91ded9da841d65c3f3632
?

rakuco, 2018-10-22:
Ah, nice, I didn't know what had landed. It looks like this is only in master at the moment, which constrains the branches we can actually support in meta-browser, but this is exactly what we need. @kraj what's the process for getting that into e.g. sumo and rocko?

closed 2018-11-16 by otavio

rakuco, 2018-11-16:
So this has been fixed, but the caveats remain: that nodejs patch still needs to be backported to the stable branches.