OSSystems/meta-browser #187: chromium-ozone-wayland fails when gbm is not availiable

NeenaReddy, 2018-11-05:
chromium-ozone-wayland fails on machines which do not support gbm.  There is a patch in igalia meta-browser github to address this issue. 

https://github.com/Igalia/meta-browser/commit/658899881ccdee6a31bddd0b91cae5afe3b184e6#diff-745a16067c85f904b61734cb398cc502

Are there any plans to include the patch here?

kraj, 2018-11-05:
I think you are the one finding this for first time here. So please prepare a pull request and send it our way

rakuco, 2018-11-05:
@msisov @jdapena

otavio, 2018-11-05:
@msisov would you mind to prepare a new snapshot for integration when looking at it?

msisov, 2018-11-05:
This patch is an incorrect way to handle this. I didn’t have time to fix it in the upstream and started my parental leave. I’m back on 26th of November and will able to handle after that date.

otavio, 2019-03-27:
As we had multiple releases merged, please @NeenaReddy give it a new go. I am closing it for now.

closed 2019-03-27 by otavio