OSSystems/meta-browser #188: compile chromium-ozone-wayland error

MuYusen, 2018-11-13:
use the master branch to compile chromium,I meet to error :

the first one 

```bash
| FAILED: gen/chrome/browser/resources/md_bookmarks/vulcanized.html gen/chrome/browser/resources/md_bookmarks/vulcanized.p2.html gen/chrome/browser/resources/md_bookmarks/crisper.js
| python ../../chrome/browser/resources/optimize_webui.py --host bookmarks --input ../../chrome/browser/resources/md_bookmarks --out_folder gen/chrome/browser/resources/md_bookmarks --depfile gen/chrome/browser/resources/md_bookmarks/build.d --html_in_files bookmarks.html --html_out_files vulcanized.html --html_out_files_polymer2 vulcanized.p2.html --js_out_files crisper.js
| Traceback (most recent call last):
|   File "../../chrome/browser/resources/optimize_webui.py", line 266, in <module>
|     main(sys.argv[1:])
|   File "../../chrome/browser/resources/optimize_webui.py", line 249, in main
|     manifest_out_path = _optimize(args.input, args)
|   File "../../chrome/browser/resources/optimize_webui.py", line 206, in _optimize
|     _polymer_css_build(out_path, crisper_html_out_paths, args.html_out_files, 1)
|   File "../../chrome/browser/resources/optimize_webui.py", line 225, in _polymer_css_build
|     html_in_paths + ['-o'] + html_out_paths)
|   File "../../third_party/node/node.py", line 27, in RunNode
|     raise RuntimeError('%s failed: %s' % (cmd, stderr))
| RuntimeError: ../../third_party/node/linux/node-linux-x64/bin/node ../../third_party/node/node_modules/polymer-css-build/bin/polymer-css-build --polymer-version 1 --no-inline-includes -f /home/ts/88/build_bsp/tmp/work/aarch64-poky-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_bookmarks/bundled/vulcanized.html -o /home/ts/88/build_bsp/tmp/work/aarch64-poky-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_bookmarks/vulcanized.html failed: Inspector is not available

```

the second 

```bash
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/printbackends/libprintbackend-lpr.so contained in package gtk3-printbackend-lpr requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-printbackend-lpr? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/printbackends/libprintbackend-file.so contained in package gtk3-printbackend-file requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-printbackend-file? [file-rdeps]
NOTE: recipe gstreamer1.0-libav-1.8.3-r0: task do_install: Started
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-cyrillic-translit.so contained in package gtk3-immodule-cyrillic-translit requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-cyrillic-translit? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-cedilla.so contained in package gtk3-immodule-cedilla requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-cedilla? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-am-et.so contained in package gtk3-immodule-am-et requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-am-et? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-multipress.so contained in package gtk3-immodule-multipress requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-multipress? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-thai.so contained in package gtk3-immodule-thai requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-thai? [file-rdeps]
NOTE: recipe gtk+3-3.22.17-r0: task do_package_write_ipk: Succeeded
NOTE: recipe gstreamer1.0-libav-1.8.3-r0: task do_install: Succeeded
NOTE: Running task 5145 of 5191 (/home/ts/88/build_bsp/../meta-bsp/recipes-multimedia/gstreamer/gstreamer1.0-libav_1.8.3.bb:do_package)
NOTE: Running task 5146 of 5191 (/home/ts/88/build_bsp/../meta-bsp/recipes-multimedia/gstreamer/gstreamer1.0-libav_1.8.3.bb:do_populate_sysroot)
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-ti-er.so contained in package gtk3-immodule-ti-er requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-ti-er? [file-rdeps]
NOTE: recipe gstreamer1.0-libav-1.8.3-r0: task do_package: Started
NOTE: recipe gstreamer1.0-libav-1.8.3-r0: task do_populate_sysroot: Started
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/libgdk-3.so.0.2200.17 contained in package gtk+3 requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk+3? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-viqr.so contained in package gtk3-immodule-viqr requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-viqr? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-inuktitut.so contained in package gtk3-immodule-inuktitut requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-inuktitut? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-ti-et.so contained in package gtk3-immodule-ti-et requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-ti-et? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA Issue: /usr/lib/gtk-3.0/3.0.0/immodules/im-ipa.so contained in package gtk3-immodule-ipa requires libwayland-egl.so()(64bit), but no providers found in RDEPENDS_gtk3-immodule-ipa? [file-rdeps]
ERROR: gtk+3-3.22.17-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: gtk+3-3.22.17-r0 do_package_qa: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /home/ts/88/build_bsp/tmp/work/aarch64-poky-linux/gtk+3/3.22.17-r0/temp/log.do_package_qa.129953
```

MuYusen, 2018-11-13:
i user `openembedded-core` at rocko branch. and add the layer at bblayers.conf as follow

```bash
    ${TOPDIR}/../meta-browser \
    ${TOPDIR}/../meta-rust \
    ${TOPDIR}/../meta-openembedded/meta-gnome \
    ${TOPDIR}/../meta-openembedded/meta-python \
    ${TOPDIR}/../meta-openembedded/meta-networking \
    ${TOPDIR}/../openembedded-core/meta \
    ${TOPDIR}/../openembedded-core/meta-selftest \
    ${TOPDIR}/../openembedded-core/meta-skeleton \
```

add chromium at `local.conf`:

```bash
# Add Chromium
IMAGE_INSTALL_append = \
    "${@bb.utils.contains('DISTRO_FEATURES', 'wayland', ' chromium-ozone-wayland libexif', \
        bb.utils.contains('DISTRO_FEATURES',     'x11', ' chromium-x11 libexif', \
                                                        '', d), d)}"
LICENSE_FLAGS_WHITELIST = \
    "${@bb.utils.contains('DISTRO_FEATURES', 'wayland', 'commercial', \
        bb.utils.contains('DISTRO_FEATURES',     'x11', 'commercial', \
                                                        '', d), d)}"

```

rakuco, 2018-11-14:
> the first one

I'm guessing the actual error looks like this:

> RuntimeError: ../../third_party/node/linux/node-linux-x64/bin/node ../../third_party/node/node_modules/polymer-css-build/bin/polymer-css-build --polymer-version 1 --no-inline-includes -f /data/src/yocto/poky/build-sumo-beaglebone-wayland/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_history/bundled/app.vulcanized.html /data/src/yocto/poky/build-sumo-beaglebone-wayland/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_history/bundled/lazy_load.vulcanized.html -o /data/src/yocto/poky/build-sumo-beaglebone-wayland/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_history/app.vulcanized.html /data/src/yocto/poky/build-sumo-beaglebone-wayland/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/chromium-ozone-wayland-dev-71.0.3545.0.r589108.igalia.1/out/Release/gen/chrome/browser/resources/md_history/lazy_load.vulcanized.html failed: Inspector is not available

in which case you need to build your nodejs-native recipe with ICU support (see the discussion in #186).

closed 2018-11-20 by MuYusen