OSSystems/meta-browser #189: Cannot login gmail account with recent chromium-x11

schnitzeltony, 2018-11-14:
It is a bit of a luxury problem but I use my Raspi to check emails from time to time (e.g when my PC is blocked by OE-Builds :)

Trying to log in, gmail asks for my account/password but after entering password I get something like 'an error occured' (did not find much helpful in the message) and the suggestion to retry later. I followed that suggestion but retries resulted in same error.

Have no idea with which update of chromium the issue came up.

otavio, 2018-11-14:
It might be due to the lack of ca-certificate. Is it successfully connecting with HTTPS?

otavio, 2018-11-24:
Did you figure what was causing this?

schnitzeltony, 2018-11-24:
Not really - all https pages seem to work properly / e.g github login works fine.

By random I found a short article in german 'c't-magazine' writing that google login now uses/requires javascript. Did not find much about it yet - maybe it is related...

schnitzeltony, 2018-11-24:
The article references [1]

[1] https://security.googleblog.com/2018/10/announcing-some-security-treats-to.html

kraj, 2018-11-24:
I doubt that, since that should result in a different error as article suggests.

schnitzeltony, 2018-11-24:
I have a different error: 

> Something went wrong
> Sorry, something went wrong there. Try again

Try again does not change nothing...

msisov, 2018-11-24:
Reason - https://groups.google.com/a/chromium.org/forum/m/#!topic/chromium-dev/XnS6bSfqaK4

Possible fix:
make sure that chrome://flags/#account-consistency is set to enabled dice (fix auth errors).

schnitzeltony, 2018-11-24:
> Possible fix:
> make sure that chrome://flags/#account-consistency is set to enabled dice (fix auth errors).

Tested - same error :(

otavio, 2019-03-27:
@schnitzeltony is it possible to try the new releases? I'd like to know if this still happens...

schnitzeltony, 2019-03-27:
You caught me in release-prepare-panic. After 14k build tasks I'll report.

FWIW: I tried gmail login from time to time but no success.

schnitzeltony, 2019-04-14:
Have chromium 73.0.3683.75 on Raspi currently: Still **cannot** login (with or without `chrome://flags/#account-consistency set to enabled dice` suggested here)

Question: Can anybody login gmail account with chromium build from this repo? Maybe my image is missing something mandatory.

otavio, 2019-04-14:
Do you have `ca-certificates` installed?

schnitzeltony, 2019-04-14:
Yes - asked opkg: it is 20190110

msisov, 2019-08-06:
To resolve the issue, you have to recompile chromium with api keys set (gmail and etc).

Here is how-to: https://www.chromium.org/developers/how-tos/api-keys

closed 2019-08-06 by msisov