OSSystems/meta-browser #190: firefox_60.1.0esr.bb:do_configure failed

jiahongxujia, 2018-11-16:
I am afraid firefox in meta-browser not workable since 2018-07-29 after
upgraded from 45.9 to 52.8 in which it uses python virtualenv to build.
http://lists.openembedded.org/pipermail/openembedded-devel/2018-November/197352.html

Unfortunately Yocto does not support python virtualenv very well.
I am curious how  the build test while the serial upgradings since 52.8

jiahongxujia, 2018-11-16:
Here is the failure log of do_configure

DEBUG: Executing shell function do_configure
 0:00.95 Clobber not needed.
 0:00.95 Adding make options from /buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/mozconfig
    CONFIG_GUESS=x86_64-wrs-linux
    MOZ_OBJDIR=/buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/firefox-build-dir
    OBJDIR=/buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/firefox-build-dir
    FOUND_MOZCONFIG=/buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/mozconfig
    export FOUND_MOZCONFIG
 0:00.96 /buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/hosttools/make -f client.mk -s configure
 0:01.01 Generating /buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/configure
 0:01.02 Generating /buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/js/src/configure
 0:01.02 cd /buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/firefox-build-dir
 0:01.02 /buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/configure
 0:01.24 Creating Python environment
 0:04.18 New python executable in /buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/firefox-build-dir/_virtualenv/bin/python2.7
 0:04.18 Also creating executable in /buildarea1/hjia/wrlinux-1018/build_nic_18lts_2018111322/build/tmp-glibc/work/core2-64-wrs-linux/firefox/60.1.0esr-r0/firefox-60.1.0/firefox-build-dir/_virtualenv/bin/python
 0:04.18 Installing setuptools, pip, wheel...done.
 0:04.54 running build_ext
 0:04.54 building 'psutil._psutil_linux' extension
 0:04.54 x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DPSUTIL_POSIX=1 -DPSUTIL_VERSION=543 -DPSUTIL_LINUX=1 -DPSUTIL_ETHTOOL_MISSING_TYPES=1 -I/usr/include/python2.7 -c psutil/_psutil_common.c -o build/temp.linux-x86_64-2.7/psutil/_psutil_common.o
 0:04.54 unable to execute 'x86_64-linux-gnu-gcc': No such file or directory
 0:04.54 error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
 0:04.54 
 0:04.54 Error processing command. Ignoring because optional. (optional:setup.py:third_party/python/psutil:build_ext:--inplace)
 0:04.54 Error processing command. Ignoring because optional. (optional:packages.txt:comm/build/virtualenv_packages.txt)
 0:04.54 Reexecuting in the virtualenv

rakuco, 2018-11-16:
@ashie

ashie, 2018-11-19:
> I am afraid firefox in meta-browser not workable since 2018-07-29 after
> upgraded from 45.9 to 52.8 in which it uses python virtualenv to build.
> http://lists.openembedded.org/pipermail/openembedded-devel/2018-November/197352.html

Please don't mix various versions & issues, ensure to ditinguish them.
Probably the issue for 52.8 you mean is the following report:

  * http://lists.openembedded.org/pipermail/openembedded-devel/2018-July/195706.html

It's already resolved by #162.

> Here is the failure log of do_configure

> 0:04.54 unable to execute 'x86_64-linux-gnu-gcc': No such file or directory

Probably your problem is different from the above issue. Sorry, I don't know the cause of your issue, I've not seen such error.

For 60.1.0 in #160, I tested on qemu-x86_64, arm and aarch64 and succesfully completed building firefox with the forked version of met-rust (meta-rust/meta-rust#191). Note that my PR in #160 requires forked version of meta-rust, I'm not sure whether the current meta-browser is buildable against the master branch of meta-rust. But I think your issue is differenrt from meta-rust's issue.

ashie, 2018-11-19:
> Do your host install virtualenv, python-pip and python-setuptools? If yes, please uninstall them and do the build again.

For arm & aarch64, I don't install them.
Here is my vagrant box to build them:

  * https://app.vagrantup.com/ashie/boxes/renesas-gecko-dev
    * Source: https://github.com/clear-code/renesas-gecko-devenv
      * Ansible playbook: https://github.com/clear-code/renesas-gecko-devenv/blob/master/provisioning/playbook.yml

ashie, 2018-11-19:
> Do your host install virtualenv, python-pip and python-setuptools?

For x86_64, yes. I install host's python-pip and build against poky:master.

> If yes, please uninstall them and do the build again.

I can't test it soon, I have no time do in the near future.

kraj, 2018-11-19:
We should have arm/arm64/x86/x86_64 builds working usually if not mips/ppc that can be optional

ashie, 2018-11-23:
>> Do your host install virtualenv, python-pip and python-setuptools?
> 
> For x86_64, yes. I install host's python-pip and build against poky:master.

I tested it for generic-x86-64 & qemux86-64 without host's python-pip, but I can't reproduce it.

closed 2019-06-19 by otavio