OSSystems/meta-browser #192: Chromium 70 build broken for 32bit x86

mdavis777, 2018-11-19:
When I try to build x86 based Chromium under Thud / GCC8 the build fails with:
| In file included from ../../mojo/public/c/system/types.h:15,
|                  from ../../mojo/public/c/system/platform_handle.h:16,
|                  from ../../mojo/public/cpp/platform/platform_handle.h:12,
|                  from ../../mojo/public/cpp/platform/platform_channel_endpoint.h:10,
|                  from ../../mojo/core/connection_params.h:11,
|                  from ../../mojo/core/channel.h:15,
|                  from ../../mojo/core/node_channel.h:19,
|                  from ../../mojo/core/node_controller.h:25,
|                  from ../../mojo/core/node_controller.cc:5:
| ../../mojo/public/c/system/buffer.h:33:42: error: static assertion failed: int64_t has weird alignment
|  MOJO_STATIC_ASSERT(MOJO_ALIGNOF(int64_t) == 8, "int64_t has weird alignment");
| ../../mojo/public/c/system/macros.h:15:53: note: in definition of macro 'MOJO_STATIC_ASSERT'
|  #define MOJO_STATIC_ASSERT(expr, msg) static_assert(expr, msg)
|                                                      ^~~~
| ../../mojo/public/c/system/data_pipe.h:43:42: error: static assertion failed: int64_t has weird alignment
|  MOJO_STATIC_ASSERT(MOJO_ALIGNOF(int64_t) == 8, "int64_t has weird alignment");
| ../../mojo/public/c/system/macros.h:15:53: note: in definition of macro 'MOJO_STATIC_ASSERT'
|  #define MOJO_STATIC_ASSERT(expr, msg) static_assert(expr, msg)
|                                                      ^~~~
| ../../mojo/public/c/system/message_pipe.h:38:42: error: static assertion failed: int64_t has weird alignment
|  MOJO_STATIC_ASSERT(MOJO_ALIGNOF(int64_t) == 8, "int64_t has weird alignment");
| ../../mojo/public/c/system/macros.h:15:53: note: in definition of macro 'MOJO_STATIC_ASSERT'
|  #define MOJO_STATIC_ASSERT(expr, msg) static_assert(expr, msg)

Discussed [here](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=51628#c0):

Fedora already has a patch that should fix this.
https://src.fedoraproject.org/cgit/rpms/chromium.git/tree/chromium-66.0.3359.170-gcc8-alignof.patch

kraj, 2018-11-19:
yes gcc8 has a bit different behavior w.r.t aligof, fedora patch you point should fix this and should be backported here as well.

rakuco, 2018-11-20:
I'll try to look into it this week.

closed 2018-11-21 by rakuco