OSSystems/meta-browser #194: Unable to Link Chrome 70

mdavis777, 2018-11-20:
Every time I try to build Chromium 70 I get the following linker error.
> | python "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- x86_64-poky-linux-g++  -m64 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2 --sysroot=/build/mdavis/test/tmp/work/corei7-64-poky-linux/chromium-x11/70.0.3538.102-r0/recipe-sysroot -pie -Wl,--version-script=../../build/linux/chrome.map -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,now -Wl,-z,relro -Wl,-z,defs -Wl,--as-needed -m64 -Wl,-O2 -Wl,--gc-sections -rdynamic -Wl,-rpath-link=. -Wl,--disable-new-dtags -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group   -latomic -ldl -lpthread -lrt -lX11 -lX11-xcb -lxcb -lXcomposite -lXcursor -lXdamage -lXext -lXfixes -lXi -lXrender -lXtst -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -lxml2 -lexpat -ldbus-1 -lXss -lwebpdemux -lwebpmux -lwebp -lfreetype -ljpeg -luuid -lXrandr -lresolv -lgio-2.0 -lm -lasound -lz -lpangocairo-1.0 -lpango-1.0 -lcairo -lpci -latk-1.0 -latk-bridge-2.0 -lFLAC -lgtk-3 -lgdk-3 -lcairo-gobject -lgdk_pixbuf-2.0 -lxslt
| obj/third_party/pdfium/third_party/libfx_agg.a: error adding symbols: Malformed archive

This is with Sumo GCC 7 intel-corei7-64 or core2-32 on a Fedora 26 64bit host.

Chromium 69 was building fine on the same system.

mdavis777, 2018-11-20:
Trying with clang now to see if is the same

mdavis777, 2018-11-20:
Just finished up clang passed fine.

kraj, 2018-11-20:
OK while you are here, can you see if this works as well as your normal testing worked when built with gcc

mdavis777, 2018-11-20:
I am confused on the "this" you are referring too.  If you are talking about testing the chrome from the clang build then I have an image building but it will take a bit of time.  
I can say that GCC 8 with Thud is getting the same linker error on my system.

kraj, 2018-11-20:
yes here 
this = "build with clang"

mdavis777, 2018-11-20:
My image failed on an unrelated error ( a different layer not ready for thud).  I have an image of sumo + clang (for chromium only) building.  Chromium was able to build with clang so I expect it will run property.

kraj, 2018-11-20:
OK. I am interested to know if someone switched to use clang in OE to do day to day chromium based images then what differences do they see, building-wise or runtime-wise.

mdavis777, 2018-11-20:
Ahhh I am not even close to doing that.  I just use clang for some static analysis functions so I already had the layer to try and build chromium with.  I didn't enable it as my global default compiler.
I did notice the build time for chromium was reduced over what it tends to be for GCC.

kraj, 2018-11-20:
> Ahhh I am not even close to doing that. I just use clang for some static analysis functions so I already had the layer to try and build chromium with. I didn't enable it as my global default compiler.
> I did notice the build time for chromium was reduced over what it tends to be for GCC.

keep going ...

rakuco, 2018-11-20:
> > | obj/third_party/pdfium/third_party/libfx_agg.a: error adding symbols: Malformed archive

In my testings, this is actually a "too many open files" error underneath. Running `ulimit -n 4096` before running `bitbake` made the problem go away for me.

mdavis777, 2018-11-20:
Yep I just ran it with an increased ulimit and it pass fine.

Thanks!

kraj, 2018-11-20:
> > > | obj/third_party/pdfium/third_party/libfx_agg.a: error adding symbols: Malformed archive
> 
> In my testings, this is actually a "too many open files" error underneath. Running `ulimit -n 4096` before running `bitbake` made the problem go away for me.

pretty hacky.. where is ```-n``` documented

denis-shienkov, 2019-07-15:
I got same error on my Yocto 'thud' layer but with the chromium-x11 v75:

> | python "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- arm-poky-linux-gnueabi-g++  -march=armv7-a  -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/mnt/data/Yocto-miatech/yocto-miatech/build-apalis-imx6/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/75.0.3770.80-r0/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -Wl,-O2 -Wl,--gc-sections -rdynamic -pie -Wl,--disable-new-dtags -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group   -latomic -ldl -lpthread -lrt -lX11 -lX11-xcb -lxcb -lXcomposite -lXcursor -lXdamage -lXext -lXfixes -lXi -lXrender -lXtst -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -lxml2 -ldbus-1 -lXss -lwebpdemux -lwebpmux -lwebp -lfreetype -ljpeg -lexpat -luuid -lXrandr -lresolv -lgio-2.0 -lasound -lm -lz -lpci -latk-1.0 -latk-bridge-2.0 -lpangocairo-1.0 -lpango-1.0 -lcairo -latspi -lFLAC -lgtk-3 -lgdk-3 -lcairo-gobject -lgdk_pixbuf-2.0 -lxslt
| /mnt/data/Yocto-miatech/yocto-miatech/build-apalis-imx6/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/75.0.3770.80-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/../../libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/8.2.0/ld: obj/ppapi/proxy/libppapi_proxy.a: error adding symbols: malformed archive

What is a workaround?

rakuco, 2019-07-15:
Did you run `ulimit -n 4096` as mentioned above?

denis-shienkov, 2019-07-15:
In my case it is unlimited:

> $ ulimit
unlimited

rakuco, 2019-07-15:
Note that `ulimit` and `ulimit -n` are different things. You were running the equivalent of `ulimit -f` according to the bash man page.

denis-shienkov, 2019-07-15:
Ah, yes, sorry, in my case with -n it return 1024. ok, thanks, I will try with 4096

denis-shienkov, 2019-07-16:
Yes, it does help, many thanks.

otavio, 2019-07-16:
> Note that `ulimit` and `ulimit -n` are different things. You were running the equivalent of `ulimit -f` according to the bash man page.

@rakuco we could consider adding a check on the recipe to avoid people falling into this pitfall. What do you think?

rakuco, 2019-07-16:
I agree, I just don't really know how to do that. The best option would be to get people to use gold instead, but I guess that's not very usual, right?

otavio, 2019-07-16:
I'd check if on `do_configure` if it is not `gold`. Direct on code.

PR #286 (2019-07-16): chromium: Check maximum number of open file descriptors when using ld.bfd
The usual limit (i.e. the value returned by `ulimit -n`), 1024, is too low
for ld.bfd to be able to link the `chrome` binary.

Add a fatal check for the value that aborts when ld.bfd is being used and
the resource limit is too low.

Fixes #194.
commit 2c3c891f (2019-07-16): chromium: Check maximum number of open file descriptors when using ld.bfd

The usual limit (i.e. the value returned by `ulimit -n`), 1024, is too low
for ld.bfd to be able to link the `chrome` binary.

Add a fatal check for the value that aborts when ld.bfd is being used and
the resource limit is too low.

Fixes #194.
closed 2019-07-16 by rakuco