OSSystems/meta-browser #195: chromium-x11 / 70.0.3538.102 compile error

conorslater, 2018-11-21:
Hello,

I'm having trouble compiling chromium-x11. It stops with the error `ninja: build stopped: subcommand failed` which I have seen in previous issues.

I've tried compiling it with Ubuntu 16.04 with gcc upgraded to v8 and with Ubuntu 18.04 with GCC v7. I've also installed clang but can't get pass the error.

My build machine has 16GB RAM as per the requirements.

Can any offer some insights on how to properly set up the build environment (Recommended Linux distro, compilers, etc.)?

I've include my `Build Configuration` below for more info

```
WARNING: Host distribution "ubuntu-18.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |#############################################################################################################| Time: 0:00:00Loaded 3206 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "gmpanel"
DISTRO_VERSION       = "1.0"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta
meta-poky            = "sumo:eebbc00b252a84d2502c3f5c7acd5a622dbd6e31"
meta-oe
meta-networking
meta-python
meta-perl            = "sumo:2bb21ef27c4b0c9d52d30b3b2c5a0160fd02b966"
meta-rust            = "sumo:3efa0f069ffe4dfcb3531ea89834e69cd2257157"
meta-browser         = "master:2b48f5a9c8524a4257b94f9e650c8163fd90e456"
meta-security        = "sumo:1e4b45ca888188b4eb838cb29b501ee004237552"
meta-raspberrypi     = "sumo:2d40b000021bc8a9ef7f329ed0ad410f8d227b97"
meta-gmpanel         = "sumo-dev-chromium:09889b8c3eca047bdae3e2b445dadd7694b03f03"
```

parthitce, 2018-11-21:
@conorslater I have same error, but the actual logs shows failure with node. Copying the log below for better understanding on the problem.

`FAILED: gen/chrome/browser/resources/md_downloads/vulcanized.html gen/chrome/browser/resources/md_downloads/vulcanized.p2.html gen/chrome/browser/resources/md_downloads/crisper.js 
python ../../chrome/browser/resources/optimize_webui.py --host downloads --input ../../chrome/browser/resources/md_downloads --out_folder gen/chrome/browser/resources/md_downloads --depfile gen/chrome/browser/resources/md_downloads/build.d --html_in_files downloads.html --html_out_files vulcanized.html --html_out_files_polymer2 vulcanized.p2.html --js_out_files crisper.js --insert_in_head \<base\ href=chrome://downloads\>
Traceback (most recent call last):
  File "../../chrome/browser/resources/optimize_webui.py", line 266, in <module>
    main(sys.argv[1:])
  File "../../chrome/browser/resources/optimize_webui.py", line 249, in main
    manifest_out_path = _optimize(args.input, args)
  File "../../chrome/browser/resources/optimize_webui.py", line 206, in _optimize
    _polymer_css_build(out_path, crisper_html_out_paths, args.html_out_files, 1)
  File "../../chrome/browser/resources/optimize_webui.py", line 225, in _polymer_css_build
    html_in_paths + ['-o'] + html_out_paths)
  File "../../third_party/node/node.py", line 27, in RunNode
    raise RuntimeError('%s failed: %s' % (cmd, stderr))
RuntimeError: ../../third_party/node/linux/node-linux-x64/bin/node ../../third_party/node/node_modules/polymer-css-build/bin/polymer-css-build --polymer-version 1 --no-inline-includes -f /work/pn/projects/bosch/old/beld/build/tmp/work/armv7ahf-neon-imx-beld-linux-gnueabi/chromium-x11/70.0.3538.110-r0/chromium-70.0.3538.110/out/Release/gen/chrome/browser/resources/md_downloads/bundled/vulcanized.html -o /work/pn/projects/bosch/old/beld/build/tmp/work/armv7ahf-neon-imx-beld-linux-gnueabi/chromium-x11/70.0.3538.110-r0/chromium-70.0.3538.110/out/Release/gen/chrome/browser/resources/md_downloads/vulcanized.html failed: Inspector is not available`

rakuco, 2018-11-21:
You need a nodejs-native recipe built with ICU support. It's been done in [meta-openembedded master](http://git.openembedded.org/meta-openembedded/commit/meta-oe/recipes-devtools/nodejs?id=d7d0cc5227d0dc7d3ff91ded9da841d65c3f3632) (as well the thud branch), and it's pending a backport to [rocko](http://lists.openembedded.org/pipermail/openembedded-devel/2018-November/197502.html) and [sumo](http://lists.openembedded.org/pipermail/openembedded-devel/2018-November/197503.html).

rakuco, 2018-11-21:
(or at least that's @parthitce's issue, @conorslater would need to post an actual error line for us to be sure)

parthitce, 2018-11-21:
@rakuco Thanks, let me try patching nodejs-native.

parthitce, 2018-11-21:
Just to update, nodejs 8.9.4 version doesn't compile with icu support.

`Log data follows:
| DEBUG: Executing shell function do_configure
| creating icu_config.gypi
| Error: could not load pkg-config data for "icu-i18n".
| See above errors or the README.md.
`
Am yet to figure out the reason for this. Possible PR https://github.com/nodejs/node/pull/24255, but couldn't apply it directly.

rakuco, 2018-11-21:
What branch are you in? I've personally tested thud, rocko and sumo and all built just fine. Does `bitbake -c clean nodejs-native` and then rebuilding the recipe help in any way?

parthitce, 2018-11-21:
@rakuco I am using [sumo](https://github.com/openembedded/meta-openembedded/commit/00909ae7322dac4ddd0a28b21df29191cc53f6d9#diff-4ff5ba494531acd72fed30fd72928f6c). cleanall/cleansstate didn't help either.

rakuco, 2018-11-21:
Hmm. Can you post a more complete version of your `do_configure` logs? I wonder if there's something else mentioned there, or whether there's something wrong your icu-native build instead.

parthitce, 2018-11-21:
I have edited the path for NDA reasons, below is the complete log.

`DEBUG: Executing shell function do_configure
creating icu_config.gypi
Error: could not load pkg-config data for "icu-i18n".
See above errors or the README.md.
WARNING: <EDITED_PATH>/nodejs/8.9.4-r0/temp/run.do_configure.2929237:1 exit 1 from './configure --prefix=/usr --with-intl=system-icu --without-snapshot --shared-openssl --shared-zlib --dest-cpu="arm" --dest-os=linux --with-arm-float-abi=hard --with-arm-fpu=neon'
ERROR: Function failed: do_configure (log file is located at <EDITED_PATH>/nodejs/8.9.4-r0/temp/log.do_configure.2929237)`

rakuco, 2018-11-21:
Is it possible to share what `MACHINE` value you're using in `local.conf`? That error message from node isn't very helpful. Are you building your icu recipe in any non-standard way?

parthitce, 2018-11-21:
@rakuco No there is no non standard way and custom recipe/layers for icu. MACHINE is imx6 [santaro](https://www.garz-fricke.com/en/products/embedded-systems/single-board-computer/IA-0001R/) with custom BSP.

rakuco, 2018-11-21:
At this point there isn't much else I can do than attempt some remote debugging. Can you `clean/cleansstate` your nodejs-native recipe, call `bitbake -c configure nodejs-native` and then `devshell` into the build directory? After that, what happens once you run `pkg-config --libs-only-l --cflags-only-I --libs-only-L icu-i18n` inside devshell? Are you using pkg-config or pkgconf?

conorslater, 2018-11-22:
@rakuco I rolled back the meta-browser repository to ```e4a667deaaf9a26a3a1aeb355770d1f29da549ad``` and it compiled correctly. So the incompatibility with sumo was the problem as you described.

I think it would be good to add something to the readme about nodejs-native needing ICU support and this is not implemented with rocko and sumo.

rakuco, 2018-11-22:
I was hoping the backports would be merged quickly and a disclaimer wouldn't be necessary. @kraj is there anything I can do to speed up the process in oe-devel?

kraj, 2018-11-22:
Stable maintainer has to accept it and test it before that I can ask for scheduling it in next branch

otavio, 2018-11-24:
@kraj it would be good as multiple people are facing this very same error.

otavio, 2019-06-19:
Same as #140

closed 2019-06-19 by otavio