OSSystems/meta-browser #197: the size of the image with chromium

MuYusen, 2018-11-22:
I want to know what the normal size  of a image with chromium。

My origin image size is 280M，after add chromium ，the image become 600M。

I think the change is too large for me。Is it normal。。

rakuco, 2018-11-22:
The biggest file in a Chromium RPM here (branch "thud", `MACHINE="intel-corei7-64") is the `chromium-bin` file, which is around 160MB in size. The `chrome` binary from the official Google Chrome RPM file on Fedora is about 132MB in size, so the difference is not that big.

I'm not sure where the rest of those 320MB are coming from though; do you see anything abnormal in the RPM package built for your image? Are you building in debug mode or something?

MuYusen, 2018-11-23:
thanks for your reply。Those 320M just image size change。when I mount at my Ubuntu host，the different just 200M。I find the chromium-bin file ,the size is as you said.

MuYusen, 2018-11-23:
And how can I delete something to small chromium。

rakuco, 2018-11-23:
That's what everybody wants :) I don't know what you can do besides trying to pass some more aggressive optimization flags to the compiler (or trying to build with clang instead of GCC), but I'd say the baseline is Chrome's official binary: if that one's at 132MB I don't think you can get much lower than that.

kraj, 2018-11-24:
> And how can I delete something to small chromium。

You might want to look into other browsers, if size is a big concern.

MuYusen, 2018-11-26:
@rakuco @kraj thank you very much.I will choose another browsers to try again.

closed 2018-11-26 by MuYusen