OSSystems/meta-browser #205: Building Chromium-X11 with debug symbols

cchilumbu, 2018-12-07:
Hello,

I would like to build chromium-x11 with debug symbols enabled, so that I can debug a segmentation fault at startup with gdb.

Is this something I can enable in my chromium-x11_%.bbappend recipe?

Thank you,
Chibesa

rakuco, 2018-12-07:
I haven't tried doing that from a `bbappend` file myself so I can't say for sure.

Please take a look at [chromium-gn.inc](https://github.com/OSSystems/meta-browser/blob/master/recipes-browser/chromium/chromium-gn.inc) if you haven't already -- I think you essentially need to add back `-g` rather than `-g1` to `DEBUG_FLAGS`, and maybe pass `is_debug=true` to GN. You most likely need to use gold rather than ld.bfd due to the binary size issues, and I'm pretty sure targeting a 32-bit architecture will hit some problems due to file sizes too.

cchilumbu, 2018-12-07:
Thanks Raphael,

Yes I am tweaking chromium-gn.inc. My platform is x86_64, so hopefully I will be spared the size issues.

I do not understand the 'gold rather than ld.bfd' statement. Could you please elaborate what I must do here?

Cheers

cchilumbu, 2018-12-07:
I will give that a try.

Cheers.

kraj, 2018-12-07:
> > I do not understand the 'gold rather than ld.bfd' statement. Could you please elaborate what I must do here?
> 
> I couldn't find any reference to it in Yocto's documentation, but you basically need to link Chromium with a different linker (in this case, ld.gold rather than ld.bfd which is the default). I think it should be enough to have something like `DISTRO_FEATURES_append_pn-chromium-x11 = " ld-is-gold"` in your `local.conf`.

Gold is part of toolchain, its just not default linker. One needs to enable it via `-fuse-ld=gold` compiler option during linking a package if needed to be used explicitly.

ld-is-gold is a DISTRO_FEATURE, it will not be effective per recipe as stated here. but one can add it like
`DISTRO_FEATURES_append = " ld-is-gold"`

This will made gold as default system linker. 

However, chromium does have mechanism to use gold regardless of it being system linker,  I think you can try following in `local.conf`

GN_ARGS_append_pn-chromium-x11 = " use_gold=true"

rakuco, 2018-12-08:
> ld-is-gold is a DISTRO_FEATURE, it will not be effective per recipe as stated here. but one can add it like
> `DISTRO_FEATURES_append = " ld-is-gold"`
>
> This will made gold as default system linker. [...]

For the record, I did try `DISTRO_FEATURES_append_pn-chromium-x11 = " ld-is-gold"` and it seems to have worked: chromium-gn.inc passed `use_gold=true` to GN automatically, and binaries were linked with `-fuse-ld=gold`.

kraj, 2018-12-08:
> For the record, I did try `DISTRO_FEATURES_append_pn-chromium-x11 = " ld-is-gold"` and it seems to have worked: chromium-gn.inc passed `use_gold=true` to GN automatically, and binaries were linked [...]

yes it would work. However,  do realize that DISTRO_FEATURE is in global config data space not in recipe space. since we check for DISTRO_FEATURE to use it, it will do the needed in recipe namespace but global data presented to build this recipe will be inconsistent with rest of the build.

cchilumbu, 2018-12-11:
I am not able to get a successful compilation for a debug build. I am using 'musl' libc, which I understand that you are no longer supporting. I suspect I would have to apply more patches to get a debug build to compile successfully on 'musl'.

My target is intel x86_64, and in chromium-gn.inc, I have set "is_debug=true  and is_official=false".

I have also set the following in my local.conf:
GN_ARGS_append_pn-chromium-x11 = " use_gold=true" 
DISTRO_FEATURES_append_pn-chromium-x11 = " ld-is-gold"

The compilation log reveals the following:
[12821/32962] g++  -MMD -MF yocto_native/obj/base/base/command_line.o.d -DUSE_SYMBOLIZE -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_DEBUG -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DWTF_USE_DYNAMIC_ANNOTATIONS=1 -D_GLIBCXX_DEBUG=1 -DBASE_IMPLEMENTATION -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_32 -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_26 -I../.. -Iyocto_native/gen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -Wall -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O0 -fno-omit-frame-pointer -g0 -fvisibility=hidden -isystem../../../recipe-sysroot-native/usr/include/glib-2.0 -isystem../../../recipe-sysroot-native/usr/lib/glib-2.0/include -isystem../../../recipe-sysroot-native/usr/include -std=gnu++14 -Wno-narrowing -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -isystem/home/cchilumbu/projects/yocto-sumo/build-pico-apl3/tmp/work/corei7-64-poky-linux-musl/chromium-x11/68.0.3440.106-r0-hdl1/recipe-sysroot-native/usr/include -isystem/home/cchilumbu/projects/yocto-sumo/build-pico-apl3/tmp/work/corei7-64-poky-linux-musl/chromium-x11/68.0.3440.106-r0-hdl1/recipe-sysroot-native/usr/include -O2 -pipe -c ../../base/command_line.cc -o yocto_native/obj/base/base/command_line.o
FAILED: yocto_native/obj/base/base/command_line.o
g++  -MMD -MF yocto_native/obj/base/base/command_line.o.d -DUSE_SYMBOLIZE -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_DEBUG -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DWTF_USE_DYNAMIC_ANNOTATIONS=1 -D_GLIBCXX_DEBUG=1 -DBASE_IMPLEMENTATION -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_32 -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_26 -I../.. -Iyocto_native/gen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -Wall -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O0 -fno-omit-frame-pointer -g0 -fvisibility=hidden -isystem../../../recipe-sysroot-native/usr/include/glib-2.0 -isystem../../../recipe-sysroot-native/usr/lib/glib-2.0/include -isystem../../../recipe-sysroot-native/usr/include -std=gnu++14 -Wno-narrowing -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -isystem/home/cchilumbu/projects/yocto-sumo/build-pico-apl3/tmp/work/corei7-64-poky-linux-musl/chromium-x11/68.0.3440.106-r0-hdl1/recipe-sysroot-native/usr/include -isystem/home/cchilumbu/projects/yocto-sumo/build-pico-apl3/tmp/work/corei7-64-poky-linux-musl/chromium-x11/68.0.3440.106-r0-hdl1/recipe-sysroot-native/usr/include -O2 -pipe -c ../../base/command_line.cc -o yocto_native/obj/base/base/command_line.o
../../base/command_line.cc: In member function ‘base::CommandLine::StringType base::CommandLine::GetSwitchValueNative(const StringPiece&) const’:
../../base/command_line.cc:314:45: error: no matching function for call to ‘std::__debug::map<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char>, std::less<void> >::find(const StringPiece&) const’
   auto result = switches_.find(switch_string);

rakuco, 2018-12-11:
Curiously that's a failure to build a tool that's going to run on the host, not the target, so I'm pretty sure that part is actually using your system's libstdc++. In any case, even if you get past that building with musl will most certainly fail, and 68.0.3440.106 is a pretty ancient version by Chromium standards.

cchilumbu, 2018-12-20:
Thank you for your pointers guys. But I will park this work for now.

You may close this issue.

closed 2018-12-20 by rakuco

da-mkay, 2019-02-13:
Hi, I am getting the same error when building `chromium-ozone-wayland` (v71) with `is_debug=true`.
As suggested here I use `ld-is-gold` and `-g`. But even without I get the same error. Do you have any idea on how to solve that?
I need a debug build because my chromium is freezing.

rakuco, 2019-03-08:
By same error do you mean [the one on the host build](https://github.com/OSSystems/meta-browser/issues/205#issuecomment-446138208)? If so, I guess your system libstdc++ (and GCC) is too old and is failing to build Chromium's C++14 code.