OSSystems/meta-browser #210: compile error when migrating chromium from ozone-wayland 70 to chromium-ozone-wayland_71.0.3578.98?

13824125580, 2019-01-18:
hi guys: 
   sorry for disturb you, i meet a compile problem and cant fix it by myself when migrate the chromium-ozone-wayland versions from 70 to the latest stable chromium-ozone-wayland_71.0.3578.98, as follows:
Parsing of 2075 .bb files complete (2073 cached, 2 parsed). 2923 targets, 275 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'at-spi2-core' (but /home/caozilong/WorkSpace/tina-yocto/sources/meta-tina/recipes-web/browser/chromium/chromium-71.0.3578.98/chromium-ozone-wayland_71.0.3578.98.bb DEPENDS on or otherwise requires it)
at-spi2-core was skipped: missing required distro feature 'x11' (not in DISTRO_FEATURES)
NOTE: Runtime target 'chromium-ozone-wayland' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['chromium-ozone-wayland', 'at-spi2-core']
ERROR: Required build target 'image-tulip-mozart-chromium' has no buildable providers.
Missing or unbuildable dependency chain was: ['image-tulip-mozart-chromium', 'chromium-ozone-wayland', 'at-spi2-core']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

yes, my at-spi2-core require "x11" feature on its bb files. if i mask it, the would be many dependcey chain errors on x11 relate staff.
REQUIRED_DISTRO_FEATURES = "x11"
so may be there is a pure at-spi2-core package that based on wayland, not x11, or i cant figure out any way to explanation this,

so can any one help me on my issue? thanks very much.

rakuco, 2019-01-21:
@msisov

msisov, 2019-01-21:
Basically, that's the accessebility module, which requires atspi2. You can try to disable it with use_atk=false if you don't need it. Otherwise, you'd have to find at-spi2-core without x11 dependency if such exists.

Btw, what target and yocto distro do you use?

msisov, 2019-01-21:
```
Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5.2"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "sumo:623b77885051174d0e05198843e739110977bd18"
meta-rust            = "sumo:8b0b52d5b369532ba804dfaa9812c51e50e396a5"
meta-yocto-bsp       = "sumo:623b77885051174d0e05198843e739110977bd18"
meta-raspberrypi     = "sumo:05f21adb99f97140e874fa2a4ed66d5e3554cd33"
meta-oe              
meta-multimedia      = "sumo:b0950aeff5b630256bb5e25ca15f4d59c115e7c1"
meta-browser-release = "ozone-wayland-stable/71.0.3578.98:1cc8abef39e4201c3a6ec416caaaef27f9d81cbc"
workspace            = "sumo:623b77885051174d0e05198843e739110977bd18"

```

msisov, 2019-01-24:
@13824125580, the following patch must be able to resolve the issue you had - https://github.com/OSSystems/meta-browser/pull/211

rakuco, 2019-02-05:
Closing, as #211 should've solved this bug report.

closed 2019-02-05 by rakuco