OSSystems/meta-browser #212: chromium-x11 32bits/64bits memory usage

Romaxx, 2019-01-27:
hi,
I have built an image with chromium-x11, 32bits and 64bits version for raspberry pi.
On RPI 3B / RPI 3B+, 32bits and 64 bits version are ok, but on RPI 3A+ (512MB of RAM instead 1GB for 3B/3B+) only the 32 bit version is fast at loading, 64 bits version needs 2 or 3 minutes to launch.
I think its a memory issue with 64bits version (which use more memory). Is it possible to specify in yocto to build 32bits version for a 64bits system ? (I need 64bits system for mongodb :S).
Thank you

rakuco, 2019-02-05:
This is more of a generic problem: it sounds like you want to know whether you can build a few recipes with a different target (and maybe toolchain) given a certain `MACHINE` value. If you can do that, the Chromium recipe should pick it up automatically. I'm not entirely sure if that's possible, though @otavio and @kraj probably know better.

kraj, 2019-02-05:
Yes you can build multilibs for arm, mips and x86 in OE

e.g. for building x86 along with x86_64 you would add something like this to local.conf
```
require conf/multilib.conf
MULTILIBS = "multilib:lib32"
DEFAULTTUNE_virtclass-multilib-lib32 = "x86"
IMAGE_INSTALL_append = " lib32-chromium-x11"
```

for arm on aarch64 you might want to do

```
require conf/multilib.conf
MULTILIBS = "multilib:lib32"
DEFAULTTUNE_virtclass-multilib-lib32 = "cortexa7t2hf-neon"
IMAGE_INSTALL_append = " lib32-chromium-x11"
```

closed 2019-02-05 by Romaxx