OSSystems/meta-browser #213: compile error with wayland port of chromium 71?

13824125580, 2019-01-29:
chrome/test/chromedriver/extension/manifest.json
1165 [1045/33857] touch obj/chrome/test/chromedriver/embed_extension_in_cpp.stamp
1166 [1046/33857] touch obj/chrome/test/chromedriver/embed_js_in_cpp.stamp
1167 [1047/33857] python ../../chrome/test/chromedriver/embed_version_in_cpp.py --chromedriver-version-file ../../chrome/test/chromedriver/VERSION --chrome-version-file ../../chrome/VERSION -     -directory gen/chrome/test/chromedriver
1168 ERROR:root:Git error: rc=128, output=''                                                                                                                                                   
1169 [1048/33857] aarch64-poky-linux-g++  --sysroot=/home/caozilong/WorkSpace/tina-yocto/build/tmp/work/aarch64-poky-linux/chromium-ozone-wayland/71.0.3578.98-r0/recipe-sysroot -MMD -MF obj/c     omponents/os_crypt/os_crypt/key_storage_kwallet.o.d -DIS_OS_CRYPT_IMPL -DUSE_KWALLET -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DNO_TCMALLOC -DFUL     L_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAF

@210 issue has been fixed, thanks a alot, but i meet a compier error again, the do_compile log is like above, so how to deal with this?

13824125580, 2019-01-29:
sorry ,it should be #210

msisov, 2019-01-29:
It doesn't seem to be related to meta-browser recipe, but rather something git related.

otavio, 2019-03-27:
News @13824125580? or should we close this issue?

otavio, 2019-06-19:
Closing now.

closed 2019-06-19 by otavio