OSSystems/meta-browser #214: Failed to run chromium-ozone-wayland

da-mkay, 2019-01-30:
Hi there!

First of all: thanks for providing this repo! 👍 I were able to build chromium, but get an error when trying to run it on my Renasas R-Car M3 Board.
These were the steps to include chromium in my build:

- Added `meta-openembedded/meta-oe`, `meta-rust` and `meta-browser` to `BBLAYERS`
- Added `chromium-ozone-wayland` to `IMAGE_INSTALL_append` of my receipe
- Did the same change as [this commit from Igalia's repo](https://github.com/Igalia/meta-browser/commit/689b4134451350723f7dc034e4856bbfb03e7e28): `use_wayland_gbm=false` (Without this option build fails)
- Use default build options for chromium.

However, if I try to run chromium using `chromium --no-sandbox` I get the following error output:

> [4103:4141:0123/132849.106609:ERROR:bus.cc(396)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
> [4103:4103:0123/132849.141374:ERROR:wayland_screen.cc(95)] Not implemented reached in virtual display::Display ui::WaylandScreen::GetDisplayMatching(const gfx::Rect&) const
> [4103:4103:0123/132849.142065:ERROR:wayland_screen.cc(67)] Not implemented reached in virtual display::Display ui::WaylandScreen::GetDisplayForAcceleratedWidget(gfx::AcceleratedWidget) const
> [4103:4103:0123/132849.142251:ERROR:desktop_window_tree_host_platform.cc(432)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::InitModalType(ui::ModalType)
> [4103:4103:0123/132849.142327:ERROR:desktop_window_tree_host_platform.cc(442)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsAnimatingClosed() const
> [4103:4103:0123/132849.143910:ERROR:desktop_window_tree_host_platform.cc(455)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SizeConstraintsChanged()
> [4103:4103:0123/132849.164866:ERROR:wayland_screen.cc(72)] Not implemented reached in virtual gfx::Point ui::WaylandScreen::GetCursorScreenPoint() const
> [4103:4103:0123/132849.167299:ERROR:desktop_window_tree_host_platform.cc(426)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SetWindowIcons(const gfx::ImageSkia&, const gfx::ImageSkia&)
> [4103:4103:0123/132849.167395:ERROR:desktop_window_tree_host_platform.cc(364)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::SetWindowTitle(const string16&)
> [4103:4103:0123/132849.187629:ERROR:wayland_screen.cc(82)] Not implemented reached in virtual gfx::AcceleratedWidget ui::WaylandScreen::GetAcceleratedWidgetAtScreenPoint(const gfx::Point&) const
> [4103:4103:0123/132849.197240:ERROR:desktop_window_tree_host_platform.cc(302)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::Activate()
> [4103:4103:0123/132849.206012:ERROR:desktop_window_tree_host_platform.cc(213)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsVisible() const
> [4131:4131:0123/132849.289841:ERROR:gles2_command_buffer_stub.cc(223)] ContextResult::kSurfaceFailure: Failed to create surface.
> [4103:4103:0123/132849.295823:ERROR:gpu_process_transport_factory.cc(967)] Lost UI shared context.
> Segmentation fault (core dumped)

The segmentation fault appears after approx. 1-3 seconds.

Do you have any idea how I get chromium started?

msisov, 2019-01-30:
Pass —in-process-gpu. Renesas hasn’t pushed render node support yet and libgbm is outdated (it’s going to be updated soon).

da-mkay, 2019-01-30:
Thank you, chromium is running now 😃 👍

da-mkay, 2019-02-04:
Unfortunately chromium freezes regularly on my M3 board.
Do you have any ideas on how to investigate the problem, because not much information is logged to stdout/stderr.

msisov, 2019-02-04:
try to use gdbserver

da-mkay, 2019-02-05:
OK, so it seems, there is no known issue about that topic.

In the Wiki of Igalia/meta-browsers `meta-openembedded/meta-gnome` is included, but I use `meta-openembedded/meta-oe`. Could that be a problem?

otavio, 2019-03-27:
Any news on this @da-mkay? or should we close this issue?

cosmo0920, 2019-04-01:
> Unfortunately chromium freezes regularly on my M3 board.
> Do you have any ideas on how to investigate the problem, because not much information is logged to stdout/stderr.

Our working target is also Renesas M3 board. In our investigating, [apitrace](https://github.com/apitrace/apitrace) is very helpful to dig GPU problems.

Our apitrace recipe is [here](https://github.com/webdino/meta-gecko-embedded/blob/master/meta-gl/recipes-support/apitrace/apitrace_git.bb).

otavio, 2019-06-19:
Closing this as the author did not provide more info.

closed 2019-06-19 by otavio