OSSystems/meta-browser #227: qemu-arm segfaults

MastaG, 2019-03-04:
Hi there, I'm using meta-browser rev: https://github.com/OSSystems/meta-browser/commit/41d3e6baea266a652c2d4dbdeca767a3054e13e8
To build chromium-ozone-wayland for 32bit arm.
My openembedded-core, meta-openembedded, meta-rust and bitbake layers are all on todays master branch.

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "fedora-29"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "odroid-xu3"
DISTRO               = "odroid-os"
DISTRO_VERSION       = "release"
TUNE_FEATURES        = "arm vfp cortexa15 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"

My GN flags:
use_cups=true
ffmpeg_branding="Chrome"
proprietary_codecs=true
use_v4l2_codec=true
use_v4lplugin=true
use_linux_v4l2_only=true
is_component_build=false
use_atk=false
use_gnome_keyring=false
use_jumbo_build=false
use_kerberos=false
use_pulseaudio=true
use_system_freetype=true
use_system_libjpeg=true
closure_compile=false
host_pkg_config="pkg-config-native"
is_debug=false
is_official_build=true
use_custom_libcxx=false
use_lld=false
symbol_level=0
enable_remoting=false
enable_nacl=false
use_sysroot=false
treat_warnings_as_errors=false
is_cfi=false
fieldtrial_testing_like_official_build=true
google_api_key=""
google_default_client_id=""
google_default_client_secret=""
custom_toolchain="//build/toolchain/yocto:yocto_target"
gold_path=""
host_toolchain="//build/toolchain/yocto:yocto_native"
is_clang=false
clang_base_path="/media/data/home/mastag/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/72.0.3626.119-r0/recipe-sysroot-native/usr/bin"
clang_use_chrome_plugins=false
linux_use_bundled_binutils=false
target_cpu="arm"
use_gold=true
v8_snapshot_toolchain="//build/toolchain/yocto:yocto_target"
use_cups=true
ffmpeg_branding="Chrome"
proprietary_codecs=true
use_v4l2_codec=true
use_v4lplugin=true
use_linux_v4l2_only=true
use_ozone=true
ozone_auto_platforms=false
ozone_platform_headless=true
ozone_platform_wayland=true
ozone_platform_x11=false
use_xkbcommon=true
use_system_libwayland=true
use_system_minigbm=true
use_system_libdrm=true
enable_hangout_services_extension=true
use_wayland_gbm=true
enable_widevine=true
arm_float_abi="hard"
arm_fpu="neon-vfpv4"
arm_tune="cortex-a15"
arm_version=7

Everything works up to the point where it has to use the qemu-wrapper to build some native parts of the browser.
This seems to segfault.
See the attached log.
[log.do_compile.zip](https://github.com/OSSystems/meta-browser/files/2926444/log.do_compile.zip)

Any ideas?

kraj, 2019-03-04:
I have seen infinite hangs but not crashes. I think this is definitely something to look into qemu itself.

MastaG, 2019-03-16:
this has been fixed by upstream oe-core.
they've switched the default arm target to armv7 and now it works.

closed 2019-03-16 by MastaG