OSSystems/meta-browser #229: CEF3 Build

pl161187smi, 2019-03-06:
Hello, please help me, how I can build cef3, may be anybody has some manual about how to add cef3 to my custom linux project, I use raspberry pi 3 and succesfully biild linux image with wayland, and now I want to add cef3 to this image build, I downgrade GCCVERSION to 5.4.0 but I have many error in build...
Thank you.Regards.

rakuco, 2019-03-07:
We don't support CEF in this repository, and we do not provide a recipe for it here. #182 had a prototype recipe that we chose not to integrate, perhaps you could you use it as a starting point for your work.

pl161187smi, 2019-03-07:
Ok, thank you, I will try use provided input from the link as the prototype.

closed 2019-03-07 by pl161187smi