OSSystems/meta-browser #230: Build Error for Chromium-ozone-wayland on Freescale BSP

tstockk, 2019-03-08:
I Try to Compile this BSP (master and a defined State)  but the compile stucks at 70% with a Error. 

Can anyone help to fix this issues? 

MACHINE=imx8mm 
DISTRO= fls-imx-xwayland

I use the Yocto package from NXP 

https://source.codeaurora.org/external/imx/imx-manifest/log/?h=imx-linux-sumo
With the Integrated Commit from OCT18 and the current master its every time the same issue

Thx!

[log.do_compile.23516.txt.zip](https://github.com/OSSystems/meta-browser/files/2946157/log.do_compile.23516.txt.zip)

rakuco, 2019-03-08:
From the log:

> aarch64-poky-linux-g++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <https://gcc.gnu.org/bugs/> for instructions.

It looks like your system's running out of memory while trying to build Chromium. You could try running fewer parallel jobs (something like `PARALLEL_MAKE_pn-chromium-ozone-wayland = -jXYZ` in `local.conf` should work).

kraj, 2019-03-08:
using 16G minimum RAM is recommended. if you have 4+ CPUs on build host.

rakuco, 2019-03-11:
Closing as there doesn't seem to be anything for us to do in meta-browser.

closed 2019-03-11 by rakuco

tstockk, 2019-03-11:
I Try it today! Thx for the amazing support!