OSSystems/meta-browser #231: Chromium-ozone-wayland content_shell compilation error.

pl161187smi, 2019-03-09:
Hello all, I was try to build content shell:
`do_compile() {
	ninja -v ${PARALLEL_MAKE} chrome chrome_sandbox chromedriver content_shell
}`
But obtain the error. Firstly related to testing fonts, and I manually copy fonts but now build end with error :
`In file included from ../../chrome/browser/ui/views/overlay/overlay_window_views.cc:5:
../../chrome/browser/ui/views/overlay/overlay_window_views.h:205:47: error: 'kEndOfVideo' was not declared in this scope`
Maybe I do need to specify some GN_ARGS arguments specific to the content shell ?
For me need minimalistic one window browser, I also try run in kiosk node but it looks like full-screen mode but for me need NOT full-screen mode, I try to set window size but after start window automatic resized to full screen without window.

rakuco, 2019-03-11:
Over to @msisov

msisov, 2019-03-18:
Do you some any other additional gn args?

otavio, 2019-06-19:
Closing due inactiveness.

closed 2019-06-19 by otavio