OSSystems/meta-browser #232: chromium wayland issue

tstockk, 2019-03-12:
Hay guys, 

after sucessfull compiling i got the last issue. 

I run Chromium on the i.MX8MM with the NXP Yocto and the master of the meta-browser layer. 

I got the following Problem 

export WAYLAND_DISPLAY=wayland-0
export XDG_RUNTIME_DIR=/tmp
chromium --no-sandbox

ERROR: wayland_connection.cc(63) Failed to connect to Wayland display
ERROR: ozone_platfrom_ayland.cc (159) Failed to initialize Wayland platform

msisov, 2019-03-12:
are you sshing to the board and starting the browser from there? It looks like the display used is wrong. Try to connect the keyboard and mouse to the board and start chrome from there.

PS you shouldn't be able to run any wayland apps. It's not chromium issue

tstockk, 2019-03-12:
i try to start it on the borad directly, i have attached an HDMI monitor mouse and keyboard

tstockk, 2019-03-12:
Her is the Output

`interface: 'wl_compositor', version: 4, name: 1
interface: 'wl_subcompositor', version: 1, name: 2
interface: 'wp_viewporter', version: 1, name: 3
interface: 'wp_presentation', version: 1, name: 4
        presentation clock id: 1 (CLOCK_MONOTONIC)
interface: 'zwp_alpha_compositing_v1', version: 1, name: 5
interface: 'zwp_relative_pointer_manager_v1', version: 1, name: 6
interface: 'zwp_pointer_constraints_v1', version: 1, name: 7
interface: 'zwp_input_timestamps_manager_v1', version: 1, name: 8
interface: 'wl_data_device_manager', version: 3, name: 9
interface: 'wl_shm', version: 1, name: 10
        formats: 'YUYV'(0x56595559) 'NV12'(0x3231564e) 'YU12'(0x32315559) RGB565                                                                                                                                                              XRGB8888 ARGB8888
interface: 'weston_touch_calibration', version: 1, name: 11
interface: 'wl_viv', version: 1, name: 12
interface: 'wl_seat', version: 5, name: 13
        name: default
        capabilities: pointer keyboard
        keyboard repeat rate: 40
        keyboard repeat delay: 400
interface: 'zwp_linux_dmabuf_v1', version: 3, name: 14
        formats:
        'YU12'(0x32315559), modifier: 0x0000000000000000
        'NV12'(0x3231564e), modifier: 0x0000000000000000
        'YUYV'(0x56595559), modifier: 0x0000000000000000
        'RG16'(0x36314752), modifier: 0x0000000000000000
        'XR24'(0x34325258), modifier: 0x0000000000000000
        'AR24'(0x34325241), modifier: 0x0000000000000000
interface: 'zwp_hdr10_metadata_v1', version: 1, name: 15
interface: 'wl_output', version: 3, name: 16
        x: 0, y: 0, scale: 1,
        physical_width: 480 mm, physical_height: 270 mm,
        make: 'DEL', model: 'DELL P2217H',
        subpixel_orientation: unknown, output_transform: 90°,
        mode:
                width: 1920 px, height: 1080 px, refresh: 60.000 Hz,
                flags: current preferred
        mode:
                width: 1920 px, height: 1080 px, refresh: 59.940 Hz,
                flags:
        mode:
                width: 1920 px, height: 1080 px, refresh: 50.000 Hz,
                flags:
        mode:
                width: 1600 px, height: 900 px, refresh: 60.000 Hz,
                flags:
        mode:
                width: 1280 px, height: 1024 px, refresh: 75.025 Hz,
                flags:
        mode:
                width: 1280 px, height: 1024 px, refresh: 60.020 Hz,
                flags:
        mode:
                width: 1152 px, height: 864 px, refresh: 75.000 Hz,
                flags:
        mode:
                width: 1280 px, height: 720 px, refresh: 60.000 Hz,
                flags:
        mode:
                width: 1280 px, height: 720 px, refresh: 59.940 Hz,
                flags:
        mode:
                width: 1280 px, height: 720 px, refresh: 50.000 Hz,
                flags:
        mode:
                width: 1024 px, height: 768 px, refresh: 75.029 Hz,
                flags:
        mode:
                width: 1024 px, height: 768 px, refresh: 60.004 Hz,
                flags:
        mode:
                width: 800 px, height: 600 px, refresh: 75.000 Hz,
                flags:
        mode:
                width: 800 px, height: 600 px, refresh: 60.317 Hz,
                flags:
        mode:
                width: 720 px, height: 576 px, refresh: 50.000 Hz,
                flags:
        mode:
                width: 720 px, height: 480 px, refresh: 60.000 Hz,
                flags:
        mode:
                width: 720 px, height: 480 px, refresh: 59.940 Hz,
                flags:
        mode:
                width: 640 px, height: 480 px, refresh: 75.000 Hz,
                flags:
        mode:
                width: 640 px, height: 480 px, refresh: 60.000 Hz,
                flags:
        mode:
                width: 640 px, height: 480 px, refresh: 59.940 Hz,
                flags:
        mode:
                width: 720 px, height: 400 px, refresh: 70.082 Hz,
                flags:
interface: 'zwp_input_panel_v1', version: 1, name: 17
interface: 'zwp_input_method_v1', version: 1, name: 18
interface: 'zwp_text_input_manager_v1', version: 1, name: 19
interface: 'zxdg_shell_v6', version: 1, name: 20
interface: 'wl_shell', version: 1, name: 21
interface: 'weston_desktop_shell', version: 1, name: 22
interface: 'weston_screenshooter', version: 1, name: 23
`

tstockk, 2019-03-12:
this is the error message: 

> sh-4.4# chromium --no-sandbox --    
> EGL: Warning: No default display support on wayland
> [4377:4377:0312/141334.717889:ERROR:gl_surface_egl.cc(935)] EGL display query failed with error EGL_SUCCESS
> [4377:4377:0312/141334.717940:ERROR:gl_surface_egl.cc(949)] eglInitialize Default failed with error EGL_BAD_DISPLAY
> [4377:4377:0312/141334.717968:ERROR:gl_ozone_egl.cc(20)] GLSurfaceEGL::InitializeOneOff failed.
> [4377:4377:0312/141334.722679:ERROR:viz_main_impl.cc(184)] Exiting GPU process due to errors during initialization
> [4349:4382:0312/141334.747410:ERROR:bus.cc(396)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are 
> "tcp" and on UNIX "unix")
> [4349:4349:0312/141334.782730:ERROR:desktop_window_tree_host_platform.cc(441)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::Init
> ModalType(ui::ModalType)
> [4349:4349:0312/141334.782864:ERROR:desktop_window_tree_host_platform.cc(451)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsAn
> imatingClosed() const
> [4349:4349:0312/141334.784728:ERROR:desktop_window_tree_host_platform.cc(464)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::Size
> ConstraintsChanged()
> [4349:4349:0312/141334.813750:ERROR:desktop_window_tree_host_platform.cc(435)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SetW
> indowIcons(const gfx::ImageSkia&, const gfx::ImageSkia&)
> [4349:4349:0312/141334.813897:ERROR:desktop_window_tree_host_platform.cc(373)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::SetW
> indowTitle(const string16&)
> [4349:4349:0312/141334.856276:ERROR:desktop_window_tree_host_platform.cc(311)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::Acti
> vate()
> [4349:4349:0312/141334.859841:ERROR:desktop_window_tree_host_platform.cc(222)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsVi
> sible() const
> [4349:4349:0312/141334.916910:ERROR:wayland_connection_connector.cc(48)] Not implemented reached in virtual void ui::WaylandConnectionConnector::OnChannelDestroy
> ed(int)
> Segmentation fault (core dumped)

tstockk, 2019-03-13:
so i check the following: 

if i use the   <project remote="OSSystems" revision="75640e14e325479c076b6272b646be7a239c18aa" name="meta-browser" path="sources/meta-browser" />

Repro with Chromium 71.xx it works fine at the system. and with the current Master i see the issue

msisov, 2019-03-13:
I suspect I know how to resolve the problem. When Wayland/Gbm is used, it uses EGL_DEFAULT_DISPLAY. Instead, we must provide EGL_PLATFORM_GBM_KHR with gbm device.

tstockk, 2019-03-13:
how is it to do? I dont understand currently...

msisov, 2019-03-13:
pass EGL_PLATFORM_GBM_KHR from here https://cs.chromium.org/chromium/src/ui/ozone/platform/wayland/gpu/wayland_connection_proxy.cc?type=cs&q=wayland_connection_pr&g=0&l=156

then add new type to here https://cs.chromium.org/chromium/src/ui/gl/gl_surface_egl.h?dr&g=0&l=36

and then check for this type here https://cs.chromium.org/chromium/src/ui/gl/gl_surface_egl.cc?dr&q=EGL_DISPLA&g=0&l=262 and make correct call for the egl display here - https://cs.chromium.org/chromium/src/ui/gl/gl_surface_egl.cc?dr&q=EGL_DISPLA&g=0&l=262

morne, 2019-06-26:
Hi

Just wanted to let you know that I'm having the same issue with the i.MX6ULL using Yocto Thud and meta-browser layer.

msisov, 2019-06-27:
Hi. I was extremely busy and could not check the bug. For the next two weeks, I'll be on holidays and won't be able to tackle it again. Once I'm back, I will take it.

Cheers,
Maksim

morne, 2019-07-01:
thanks Maksim. I'll test and verify the fix once it's finished.

daescalona, 2019-07-01:
> Hi
> 
> Just wanted to let you know that I'm having the same issue with the i.MX6ULL using Yocto Thud and meta-browser layer.

I am exactly in the same situation. iMX6UL using Yocto Thud and meta-browser layer. Building for X11 it works perfectly though.

morne, 2019-07-26:
Hi Maksim - any update on this issue ?

msisov, 2019-07-29:
Hi @morne, I was busy with other tasks and had holidays.

Do you have latest version compiled? If yes, would you please try --in-process-gpu runtime flag and report if Chromium starts then?

morne, 2019-07-29:
I don't have the latest version compiled, unfortunately.

I'll compile and test the latest version with the command above, and report back (probably only tomorrow, as chromium takes forever to build).

msisov, 2019-08-06:
Do not know, how I missed this

> ERROR: wayland_connection.cc(63) Failed to connect to Wayland display
> ERROR: ozone_platfrom_wayland.cc (159) Failed to initialize Wayland platform

but this means Wayland can't find a display to connect to. It has nothing to do with anything else than wl_display_connect. Thus, all the above I wrote did not really make sense...

We pass null name to it and connect to the one specified with WAYLAND_DISPLAY[1]. Thus, there is nothing to do with chromium, but something wrong on your side. Are you able to run other applications? What if you try to run chromium

` DISPLAY=:0 chromium`

or 

` DISPLAY=:1 chromium`

[1] https://www.mankier.com/3/wl_display_connect

daescalona, 2019-09-18:
Hello,
I am still unable to make wayland display to work correctly with chromium, this means hardware acceleration is not working. This is what I get:

`# DISPLAY=:0 chromium --no-sandbox
EGL: Warning: No default display support on wayland
[3918:3918:0918/082942.775186:ERROR:gl_surface_egl.cc(951)] EGL display query failed with error EGL_SUCCESS
[3918:3918:0918/082942.775307:ERROR:gl_surface_egl.cc(965)] eglInitialize Default failed with error EGL_BAD_DISPLAY
[3918:3918:0918/082942.775361:ERROR:gl_ozone_egl.cc(20)] GLSurfaceEGL::InitializeOneOff failed.
[3918:3918:0918/082942.785537:ERROR:viz_main_impl.cc(170)] Exiting GPU process due to errors during initialization
[3896:3925:0918/082942.843851:ERROR:object_proxy.cc(619)] Failed to call method: org.freedesktop.Notifications.GetCapabilities: object_path= /org/freedesktop/Notifications: org.freedesktop.DBus.Error.ServiceUnkn
own: The name org.freedesktop.Notifications was not provided by any .service files
[3896:3957:0918/082943.167800:ERROR:object_proxy.cc(619)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org
.freedesktop.UPower was not provided by any .service files
[3896:3957:0918/082943.169583:ERROR:object_proxy.cc(619)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name
 org.freedesktop.UPower was not provided by any .service files
[3896:3957:0918/082943.171226:ERROR:object_proxy.cc(619)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name
 org.freedesktop.UPower was not provided by any .service files
[3896:3896:0918/082944.266776:ERROR:CONSOLE(1)] "Refused to execute inline event handler because it violates the following Content Security Policy directive: "script-src 'strict-dynamic' 'sha256-r1drsgoDPQZulnXw
abgollhGK3T7e1MR/3dT/kdnsHk=' 'sha256-7Ev0+xudUJf+TNOaD7bOGEsbqNx10CoivlMga0NvMM8=' 'sha256-JJcFw5vJcK0RjK1BZMPVDPOXUXIxZOVOdS1UlR8I0oc=' 'sha256-455z/8g1DIg5Sdxz0lA1l+6+NbtUDQL5F6wsSWq4awg=' 'sha256-98oC9MOukj1
vqRurBgSvb3ckqChyJKNnthWgATTmugg=' 'sha256-QtaEGY7z+CISGaCsQAzXgfpAMMmdekW9iw+F6VckWbs=' 'sha256-zGJp5mpYDsd/Mcn3LTpQznomAg4gSQ7Hv0Uwf4zf1Z4='". Either the 'unsafe-inline' keyword, a hash ('sha256-...'), or a no
nce ('nonce-...') is required to enable inline execution.
", source: chrome-search://local-ntp/local-ntp.html (1)
`

msisov, 2019-09-18:
Can you also attach the log produced by running the browser with
WAYLAND_DEBUG=1  ?

daescalona, 2019-09-18:
Sure, here it is:
`# WAYLAND_DEBUG=1 DISPLAY=:0 chromium --no-sandbox                                                                                                                                           
[3808541.210]  -> wl_display@1.get_registry(new id wl_registry@2)
[3808541.403]  -> wl_display@1.sync(new id wl_callback@3)
[3808541.697] wl_display@1.delete_id(3)
[3808541.735] wl_registry@2.global(1, "wl_compositor", 4)
[3808541.784]  -> wl_registry@2.bind(1, "wl_compositor", 4, new id [unknown]@4)
[3808541.839] wl_registry@2.global(2, "wl_subcompositor", 1)
[3808541.881]  -> wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@5)
[3808541.934] wl_registry@2.global(3, "wp_viewporter", 1)
[3808541.977] wl_registry@2.global(4, "wp_presentation", 1)
[3808542.019]  -> wl_registry@2.bind(4, "wp_presentation", 1, new id [unknown]@6)
[3808542.069] wl_registry@2.global(5, "zwp_alpha_compositing_v1", 1)
[3808542.110] wl_registry@2.global(6, "zwp_relative_pointer_manager_v1", 1)
[3808542.151] wl_registry@2.global(7, "zwp_pointer_constraints_v1", 1)
[3808542.259] wl_registry@2.global(8, "zwp_input_timestamps_manager_v1", 1)
[3808542.303] wl_registry@2.global(9, "wl_data_device_manager", 3)
[3808542.347]  -> wl_registry@2.bind(9, "wl_data_device_manager", 3, new id [unknown]@7)
[3808542.406] wl_registry@2.global(10, "wl_shm", 1)
[3808542.453]  -> wl_registry@2.bind(10, "wl_shm", 1, new id [unknown]@8)
[3808542.506] wl_registry@2.global(11, "weston_touch_calibration", 1)
[3808542.547] wl_registry@2.global(12, "wl_viv", 1)
[3808542.588] wl_registry@2.global(13, "wl_seat", 5)
[3808542.629]  -> wl_registry@2.bind(13, "wl_seat", 4, new id [unknown]@9)
[3808542.683]  -> wl_data_device_manager@7.get_data_device(new id wl_data_device@10, wl_seat@9)
[3808542.724] wl_registry@2.global(14, "zwp_linux_dmabuf_v1", 3)
[3808542.766]  -> wl_registry@2.bind(14, "zwp_linux_dmabuf_v1", 3, new id [unknown]@11)
[3808542.821]  -> wl_display@1.sync(new id wl_callback@12)
[3808542.853] wl_registry@2.global(15, "zwp_hdr10_metadata_v1", 1)
[3808542.894] wl_registry@2.global(16, "wl_output", 3)
[3808542.936]  -> wl_registry@2.bind(16, "wl_output", 3, new id [unknown]@13)
[3808543.001] wl_registry@2.global(17, "zwp_input_panel_v1", 1)
[3808543.044] wl_registry@2.global(18, "zwp_input_method_v1", 1)
[3808543.088] wl_registry@2.global(19, "zwp_text_input_manager_v1", 1)
[3808543.131]  -> wl_registry@2.bind(19, "zwp_text_input_manager_v1", 1, new id [unknown]@14)
[3808543.184] wl_registry@2.global(20, "zxdg_shell_v6", 1)
[3808543.226]  -> wl_registry@2.bind(20, "zxdg_shell_v6", 1, new id [unknown]@15)
[3808543.277] wl_registry@2.global(21, "wl_shell", 1)
[3808543.317] wl_registry@2.global(22, "weston_desktop_shell", 1)
[3808543.359] wl_registry@2.global(23, "weston_screenshooter", 1)
[3808543.400] wl_callback@3.done(55)
[3808544.123] wl_display@1.delete_id(12)
[3808544.160] wl_seat@9.capabilities(7)
[3808544.191]  -> wl_seat@9.get_pointer(new id wl_pointer@3)
[3808544.234]  -> wl_compositor@4.create_surface(new id wl_surface@16)
[3808544.271]  -> wl_seat@9.get_keyboard(new id wl_keyboard@17)
[3808544.309]  -> wl_seat@9.get_touch(new id wl_touch@18)
[3808544.346] wl_seat@9.name("default")
[3808544.373] zwp_linux_dmabuf_v1@11.modifier(538982482, 100663296, 2)
[3808544.419] zwp_linux_dmabuf_v1@11.modifier(943215175, 100663296, 2)
[3808544.461] zwp_linux_dmabuf_v1@11.modifier(842093144, 100663296, 2)
[3808544.501] zwp_linux_dmabuf_v1@11.modifier(842089048, 100663296, 2)
[3808544.541] zwp_linux_dmabuf_v1@11.modifier(842093121, 100663296, 2)
[3808544.581] zwp_linux_dmabuf_v1@11.modifier(842089025, 100663296, 2)
[3808544.621] zwp_linux_dmabuf_v1@11.modifier(892424792, 100663296, 2)
[3808544.662] zwp_linux_dmabuf_v1@11.modifier(892420696, 100663296, 2)
[3808544.702] zwp_linux_dmabuf_v1@11.modifier(892424769, 100663296, 2)
[3808544.741] zwp_linux_dmabuf_v1@11.modifier(892420673, 100663296, 2)
[3808544.782] zwp_linux_dmabuf_v1@11.modifier(909199186, 100663296, 2)
[3808544.822] zwp_linux_dmabuf_v1@11.modifier(909199170, 100663296, 2)
[3808544.862] zwp_linux_dmabuf_v1@11.modifier(875713112, 100663296, 2)
[3808544.902] zwp_linux_dmabuf_v1@11.modifier(875709016, 100663296, 2)
[3808544.944] zwp_linux_dmabuf_v1@11.modifier(875714642, 100663296, 2)
[3808544.984] zwp_linux_dmabuf_v1@11.modifier(875708738, 100663296, 2)
[3808545.020] zwp_linux_dmabuf_v1@11.modifier(875713089, 100663296, 2)
[3808545.056] zwp_linux_dmabuf_v1@11.modifier(875708993, 100663296, 2)
[3808545.091] zwp_linux_dmabuf_v1@11.modifier(875708754, 100663296, 2)
[3808545.127] zwp_linux_dmabuf_v1@11.modifier(875708738, 100663296, 2)
[3808545.161] zwp_linux_dmabuf_v1@11.modifier(1448695129, 0, 0)
[3808545.199] zwp_linux_dmabuf_v1@11.modifier(1431918169, 0, 0)
[... 668 lines cut ...]
[3817670.503] wl_display@1.delete_id(26)
[3817670.576] wp_presentation_feedback@26.sync_output(wl_output@13)
[3817670.611] wp_presentation_feedback@26.presented(0, 5747, 295739000, 16668055, 0, 334412, 7)
[3818148.987]  -> wl_surface@12.damage_buffer(146, 53, 1, 16)
[3818149.105]  -> wl_surface@12.attach(wl_buffer@24, 0, 0)
[3818149.163]  -> wl_surface@12.frame(new id wl_callback@26)
[3818149.210]  -> wp_presentation@6.feedback(wl_surface@12, new id wp_presentation_feedback@25)
[3818149.263]  -> wl_surface@12.commit()
[3818156.672] wl_display@1.delete_id(26)
[3818156.803] wl_buffer@24.release()
[3818157.026] wl_callback@26.done(5747779)
[3818170.649] wl_display@1.delete_id(25)
[... cut ...]
[3818650.147]  -> wl_surface@12.frame(new id wl_callback@25)
[3818650.283]  -> wp_presentation@6.feedback(wl_surface@12, new id wp_presentation_feedback@26)
[3818650.338]  -> wl_surface@12.commit()
[3818656.711] wl_display@1.delete_id(25)
[3818656.843] wl_buffer@24.release()
[3818657.072] wl_callback@25.done(5748279)
[3818670.647] wl_display@1.delete_id(26)
[3818670.721] wp_presentation_feedback@26.sync_output(wl_output@13)
[3818670.754] wp_presentation_feedback@26.presented(0, 5748, 295856000, 16668055, 0, 334472, 7)`

msisov, 2019-09-30:
@daescalona , you seem to have a different issue than the topic starter. Try to

export WAYLAND_DISPLAY=wayland-0 

and then run the browser again

daescalona, 2019-09-30:
Already tried that several times with the same result:

root:~# export WAYLAND_DISPLAY=wayland-0
root:~# /usr/bin/chromium 2>&1 >/var/log/chromium.log
EGL: Warning: No default display support on wayland
[3662:3662:0930/145034.254169:ERROR:gl_surface_egl.cc(951)] EGL display query failed with error EGL_SUCCESS
[3662:3662:0930/145034.254401:ERROR:gl_surface_egl.cc(965)] eglInitialize Default failed with error EGL_BAD_DISPLAY
[3662:3662:0930/145034.254477:ERROR:gl_ozone_egl.cc(20)] GLSurfaceEGL::InitializeOneOff failed.
[3662:3662:0930/145034.262940:ERROR:viz_main_impl.cc(170)] Exiting GPU process due to errors during initialization

This is working perfectly in x11, also other OpenGL apps are working nice in Wayland, so no idea why Chromium just refuses to start correctly...

agners, 2019-10-03:
@daescalona does ` --in-process-gpu` works for you?

daescalona, 2019-10-03:
@agners holy cow yes! I think I almost tried every parameter to launch Chromium but missed this one. This is the list of launch parameters I am using in case anyone needs to take a look:

CHROMIUM_EXTRA_ARGS_append = " \
        --enable-gpu \
        --enable-gpu-memory-buffer-video-frames \
        --enable-accelerated-2d-canvas \
        --enable-accelerated-vpx-decode \
        --enable-native-gpu-memory-buffers \
        --enable-zero-copy \
        --gpu-no-context-lost \
        --user-data-dir=\/home\/root\/.chromium\/ \
        --disable-infobars \
        --disable-session-crashed-bubble \
        --ignore-certificate-errors \
        --ignore-gpu-blacklist \
        --no-sandbox \
        --window-position=0,0 \
        --no-first-run \
        --in-process-gpu \
"
Display needs to be exported, use
    export WAYLAND_DISPLAY=wayland-0
for Wayland and
    export DISPLAY=:0
for X11

Thanks a lot!

msisov, 2019-10-03:
try removing this 
https://cs.chromium.org/chromium/src/ui/ozone/platform/wayland/gpu/wayland_surface_factory.cc?q=wayland_surface_factory.cc&dr&l=114

NolanZinn-bayer, 2019-10-29:
I am experiencing the same problem as @morne.  Is there a solution to this problem yet?

msisov, 2019-11-04:
> @agners holy cow yes! I think I almost tried every parameter to launch Chromium but missed this one. This is the list of launch parameters I am using in case anyone needs to take a look: [...]

You mean this helps, don't you?

> 
> Thanks a lot!

NolanZinn-bayer, 2019-11-15:
The problem for me was the XDG_RUNTIME_DIR was not set.  Adding the line XDG_RUNTIME_DIR="/run/user/0" to /etc/environment and then rebooting fixed the problem for me.  (Or you could just run "XDG_RUNTIME_DIR=/run/user/0  chromium"  to see whether or not this solves your problem)

closed 2019-12-02 by msisov