OSSystems/meta-browser #234: chromium-ozone-wayland fails to build with minigbm

MastaG, 2019-03-16:
I'm trying to build chromium-ozone-wayland with minigbm instead of the system gbm.
When I compile against the system gbm (which is mesa) it builds fine.
But I cannot run it on my target system (Odroid XU4) because the arm mali driver ships with an older libgbm which does not provide gbm_bo_map/gbm_bo_unmap.

That's why I'm trying to build with the included minigbm by setting:
use_system_minigbm=false
use_system_libdrm=false
use_exynos_minigbm=true

Now it builds but fails to link with:

> arm-oe-linux-gnueabi-clang++   -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a15 --rtlib=compiler-rt --stdlib=libc++ -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments --sysroot=/media/data/home/mastag/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/72.0.3626.121-r0/recipe-sysroot -MMD -MF obj/ui/ozone/common/linux/gbm/gbm_wrapper.o.d -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DUSE_EGL -DVK_NO_PROTOTYPES -Dtypeof\(x\)=__typeof__\(x\) -DDRV_EXYNOS -DU_USING_ICU_NAMESPACE=0 -DU_ENABLE_DYLOAD=0 -DUSE_CHROMIUM_ICU=1 -DU_STATIC_IMPLEMENTATION -DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_FILE -DUCHAR_TYPE=uint16_t -DGOOGLE_PROTOBUF_NO_RTTI -DGOOGLE_PROTOBUF_NO_STATIC_INITIALIZER -DHAVE_PTHREAD -DSK_HAS_PNG_LIBRARY -DSK_HAS_WEBP_LIBRARY -DSK_HAS_JPEG_LIBRARY -DSK_VULKAN_HEADER=\"../../skia/config/SkVulkanConfig.h\" -DSK_VULKAN=1 -DSK_SUPPORT_GPU=1 -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DVK_NO_PROTOTYPES -I../.. -Igen -Igen/shim_headers/libwebp_shim -I../../third_party/vulkan/include -I../../third_party/libdrm/src -I../../third_party/libdrm/src/include -I../../third_party/libdrm/src/include/drm -I../../third_party/minigbm/src -I../../third_party/ced/src -I../../third_party/icu/source/common -I../../third_party/icu/source/i18n -I../../third_party/protobuf/src -I../../skia/config -I../../skia/ext -I../../third_party/skia/include/c -I../../third_party/skia/include/config -I../../third_party/skia/include/core -I../../third_party/skia/include/docs -I../../third_party/skia/include/effects -I../../third_party/skia/include/encode -I../../third_party/skia/include/gpu -I../../third_party/skia/include/pathops -I../../third_party/skia/include/ports -I../../third_party/skia/include/utils -I../../third_party/vulkan/include -I../../third_party/skia/third_party/vulkanmemoryallocator -I../../third_party/skia/include/codec -I../../third_party/skia/src/gpu -I../../third_party/skia/src/sksl -I../../third_party/skia/modules/skottie/include -I../../third_party/vulkan/include -I../../third_party/khronos -I../../gpu -I../../third_party/mesa_headers -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -mfpu=neon-vfpv4 -mthumb -Wall -Wextra -Wimplicit-fallthrough -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-null-pointer-arithmetic -Wno-ignored-pragma-optimize -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wno-\#pragma-messages -Wno-enum-conversion -std=c++14 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden  -O2 -pipe  -feliminate-unused-debug-types -fmacro-prefix-map=/media/data/home/mastag/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/72.0.3626.121-r0=/usr/src/debug/chromium-ozone-wayland/72.0.3626.121-r0                      -fdebug-prefix-map=/media/data/home/mastag/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/72.0.3626.121-r0=/usr/src/debug/chromium-ozone-wayland/72.0.3626.121-r0                      -fdebug-prefix-map=/media/data/home/mastag/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/72.0.3626.121-r0/recipe-sysroot=                      -fdebug-prefix-map=/media/data/home/mastag/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/72.0.3626.121-r0/recipe-sysroot-native= -g1 --stdlib=libc++ -c ../../ui/ozone/common/linux/gbm_wrapper.cc -o obj/ui/ozone/common/linux/gbm/gbm_wrapper.o
../../ui/ozone/common/linux/gbm_wrapper.cc:173:9: error: no matching function for call to 'gbm_bo_map'
        gbm_bo_map(bo_, 0, 0, gbm_bo_get_width(bo_), gbm_bo_get_height(bo_),
        ^~~~~~~~~~
../../third_party/minigbm/src/gbm.h:356:1: note: candidate function not viable: requires 9 arguments, but 8 were provided
gbm_bo_map(struct gbm_bo *bo,
^
../../ui/ozone/common/linux/gbm_wrapper.cc:302:40: error: variable has incomplete type 'struct gbm_import_fd_modifier_data'
    struct gbm_import_fd_modifier_data fd_data;
                                       ^
../../ui/ozone/common/linux/gbm_wrapper.cc:302:12: note: forward declaration of 'gbm_import_fd_modifier_data'
    struct gbm_import_fd_modifier_data fd_data;
           ^
../../ui/ozone/common/linux/gbm_wrapper.cc:320:33: error: use of undeclared identifier 'GBM_BO_IMPORT_FD_MODIFIER'
    bo = gbm_bo_import(device_, GBM_BO_IMPORT_FD_MODIFIER, &fd_data, gbm_flags);

msisov, 2019-03-18:
you'll have to wait until m73 is released (it's ongoing process). I just remember the m72 had only been able to be built against system gbm. The patches to make both minigbm and upstream gbm apis compatible are being pushed to m73

MastaG, 2019-03-20:
thanks it's fixed :)

closed 2019-03-20 by MastaG