OSSystems/meta-browser #237: chromium: Decide which toolchain(s) to support

rakuco, 2019-03-18:
So far, we've supported building Chromium (both chromium-x11 and chromium-ozone-wayland) with whatever GCC (or clang, via meta-clang) was in a given Yocto/OE branch.

This is becoming increasingly difficult:
* GCC is not officially supported upstream, which means @msisov and I need to spend a considerable amount of time submitting build fixes upstream. Some of them are actual code fixes, but most of them are just workarounds for GCC bugs, and depending on the kind of workaround they are not very welcome upstream (and sometimes are rejected altogether). This is becoming untenable: fixing Chromium is time-consuming, but reporting bugs and reaching a wall in GCC's Bugzilla is frustrating. And even when the bugs do get fixed there's the problem of getting the fixes to the users of stable GCC versions. Additionally, I know that there are some clang-specific ifdefs in the Chromium code that leverage some builtins that can offer performance improvements that are just not used with GCC.
* clang via meta-clang is a better alternative, but being stuck with an old (by Chromium's standards) version in a branch doesn't help either. For example, I'm testing an update to M73 with @msisov and right now the code just doesn't build with clang6 in the sumo branch. This was fixed in clang7, but cherry-picking the revision that fixed things isn't enough (I reach an ICE if I do that), and adding some upstream-inappropriate patch to work around the issue can be difficult and result in new bugs or performance regressions.

Officially, Chromium only really supports building a certain release with a specific LLVM/clang revision. There are prebuilt binaries that can be downloaded, but they can also be built locally. There are no Chromium-specific patches, Chromium's clang is essentially a snapshot from LLVM's svn repository.

I'd like to hear what your opinions are here. If we were to follow upstream's advice, we'd build a copy of Chromium's clang as a native recipe and it to build the code (I have no idea how this would work in terms of all the toolchain-related variables that get set in e.g. `bitbake.conf` though).

rakuco, 2019-03-18:
@otavio @kraj @msisov

otavio, 2019-03-18:
There are a few hidden things here:

1. number of branches
2. GCC
3. clang

I think we should reduce the number of branches we support instead. I see no reason for `master` to be tested against `sumo` for example. In fact, we could test it on `thud` and `master` and it would be enough, from my point of view.

GCC vs clang is more difficult from my perspective. The biggest problem in "dropping" GCC support is it will add a hard dependency on `meta-clang`.

So I'd propose we keep supporting GCC but limit the number of branches to `master` and current stable branches only. We can branch for old releases and keep them locked in, for specific versions, for the users to have a know good release to use.

kraj, 2019-03-18:
I think @otavio’s idea is fine we should limit the testing matrix but I think we should be also well served if we switch to using clang from meta-clang since it has become very usable since this release I would be glad to help with any backports required but in essence we can start with 2.7 release as clang being primary compiler 

If upstream has dropped supporting Gcc then it is just a climb that will keep getting steeper

msisov, 2019-03-19:
@kraj by saying 
> If upstream has dropped supporting Gcc then it is just a climb that will keep getting steeper
do you mean Chromium mainline repository? The GCC support has already been dropped quite a long time ago, and we have always had to send gcc fixes to the repository. You can see the patches from this bug: https://bugs.chromium.org/p/chromium/issues/detail?id=819294

otavio, 2019-03-19:
So guys, what are your view? should we try to keep with GCC for some more time or remove it?

rakuco, 2019-03-19:
Thanks every one for chiming in. I'll try to address the topics separately:

## Supported branches

I am fine with reducing the number of branches we test and officially support, whichever they may be. I'm not so sure about creating branches for e.g. `sumo` and `rocko` because we'd be essentially creating branches tracking old releases with multiple known security vulnerabilities. I'd rather have the recipes not build for those branches than letting users ship old, insecure software.

And if we change the branches we supported, I'd like to agree on a policy for which branches we should regularly support and when to drop support for older ones.

## Supported compilers

By far, supporting GCC is the most demanding option: as @msisov mentioned, upstream Chromium hasn't officially supported GCC for [many years](https://groups.google.com/a/chromium.org/d/msg/chromium-dev/6OwjiZXGTVs/W_VURycXuREJ), and clang is currently the sole compiler tested and used upstream to produce all their official builds for Linux, ChromeOS, Android, Windows and Mac. Like I said, We've been spending a lot of time landing GCC build fixes upstream, but it stops making sense when we are working around actual GCC bugs that do not get fixed and which may require us to make the code uglier or more inefficient upstream.

Supporting only clang would definitely help here, but if I understood @kraj correctly meta-clang only became really stable starting with 2.7 (i.e. master)? We'd need to have good support on at least thud for now, and our branch support policy would also be influenced by the clang versions we have for each branch, as I suspect even clang 7 may become too old for Chromium in a not-so-distant future.

Did anyone think if bootstrapping Chromium's own clang would be a viable option as well?

rakuco, 2019-03-19:
> Like I said, We've been spending a lot of time landing GCC build fixes upstream, but it stops making sense when we are working around actual GCC bugs that do not get fixed and which may require us to make the code uglier or more inefficient upstream.

And also like I said, this can end up in plain wrong code being produced -- see https://bugs.chromium.org/p/chromium/issues/detail?id=941696, for example.

otavio, 2019-03-19:
> ## Supported branches
>
> I am fine with reducing the number of branches we test and officially support, whichever they may be. I'm not so sure about creating branches for e.g. `sumo` and `rocko` [...]

I understand the security implications but some users are Ok with it; the README might express this concern but providing branches allow users to have a known baseline for testing. We can even add, on bitbake output, a warning saying the version is old and has known security flaws.

> And if we change the branches we supported, I'd like to agree on a policy for which branches we should regularly support and when to drop support for older ones.

As I said earlier, I think `master` and current `stable` (currently `thud`) are good enough. We might provide `oldstable` (currently `sumo`) support for some time (as a transition period for customers) but we should drop official support so people are pushed to upgrade.

So, I'd be fine with:

* `master`
* `stable`
* `oldstable` (for 2-3 months)

> ## Supported compilers
> [...]
> Supporting only clang would definitely help here, but if I understood @kraj correctly meta-clang only became really stable starting with 2.7 (i.e. master)? We'd need to have good support on at least thud for now, and our branch support policy would also be influenced by the clang versions we have for each branch, as I suspect even clang 7 may become too old for Chromium in a not-so-distant future.

I suspect @kraj would be fine in backporting the newer clang to old branches if needed. @kraj ?

> Did anyone think if bootstrapping Chromium's own clang would be a viable option as well?

This is not a good option; this requires more build time and in the end, we'd need to keep up with foreign toolchain issues.

kraj, 2019-03-19:
@rakuco

We should be able to use clang for thud branch as well however I have not tested thud very much for meta-clang with chromium especially but others probably have and if we have issues I can fix them no issues there

kraj, 2019-03-19:
Let’s work on making thud work and switch to meta-clang and let’s see where we end up with I can provide fixes for both chromium and clang as needed and if I can :)

msisov, 2019-03-19:
@kraj awesome! hopefully, it will work :)

rakuco, 2019-03-21:
Thanks for the input, everyone. I'm a bit swamped with non-Yocto work at the moment, but will try to drive this forward as soon as I can.

cpboyd, 2019-05-23:
With regards to @octavio's https://github.com/OSSystems/meta-browser/issues/237#issuecomment-474029765, what was the logic for making a hard dependency upon meta-rust in order to include firefox rather than using dynamic-layers?

Most developers are probably only including meta-browser for one of the browsers.

Could it be possible to use dynamic-layers of `meta-clang` for Chromium and `meta-rust` for Firefox?

This would remove the hard requirement for both, should a developer only want one of the browsers.

otavio, 2019-05-24:
I am not against to use the dynamic layer mechanism. Please prepare a PR and we can all review it.

kraj, 2019-05-24:
rust is required for modern firefox and clang is their default compiler which is default for chromium too. so for consistency we should add hard dependencies on these layers. So we can get consistent behavior across

cpboyd, 2019-05-29:
Per https://github.com/OSSystems/meta-browser/pull/264#issuecomment-496909824 and #265, it seems like Firefox 68 will also require meta-clang so it makes more sense to have meta-clang as a hard-requirement once either the Firefox or Chromium recipes update to use clang.

ashie, 2019-05-30:
> GCC is not officially supported upstream

> Officially, Chromium only really supports building a certain release with a specific LLVM/clang revision. There are prebuilt binaries that can be downloaded, but they can also be built locally. [...]

Modern Firefox is also facing very similar situation.
At least dropping GCC support in the near feature is preferred to stop wasting our time.

> Did anyone think if bootstrapping Chromium's own clang would be a viable option as well?

I'm also considering both options for building firefox by clang:

* Use pre-built clang that is provided by Mozilla
* Use meta-clang

But currently I confirme only later way to ensure respecting Yocto's philosophy.
meta-clang seems fine enough for Firefox at this moment.

otavio, 2019-06-19:
Are we set on this? I think we ought to require `meta-clang` sooner than later, so we can get any fixes in.

cpboyd, 2019-06-19:
I'd say requiring `meta-clang` seems the most Yocto-friendly option.

If there are issues requiring a lot of effort to patch, then maybe fallback to the pre-built clang versions.

otavio, 2019-06-19:
@cpboyd using pre-built versions is not an option IMO as it is hard to test, fix and maintain. 

I am asking because we are still keeping GCC working and I'd like to know from @rakuco if it has been a challenge or not. If it has been working well, keeping GCC for now is good.

As I mentioned in past, I'd rather reduce the supported branches than drop GCC support, at least for near future.

rakuco, 2019-06-19:
It's been quite painful whenever we're moving from one major version to another. There are multiple compiler bugs we need to work around: some of which were fixed in more recent GCC releases that never make it to the older branches we support, and some which are still open in GCC's bug tracker. Overall, clang seems to be a lot better and faster at supporting C++11 and C++14, and the fact that GCC is never tested upstream even leads to [wrong code being generated](https://bugs.chromium.org/p/chromium/issues/detail?id=941696).

I'd like to require meta-clang, I just haven't had enough time in a day to test all branches with clang and verify everything also runs on actual hardware for at least one branch.

kraj, 2019-06-19:
I think we should start with master branch for using clang for all arches we currently care for and maybe drop one of older combination that is tested 

With my limited testing I have seen clang compiled chromium work on rpi3 and I have not yet runtime tested other machines but build succeeds fully for qemux86-64 and qemuarm

closed 2019-07-22 by rakuco