OSSystems/meta-browser #238: Missing X11 folder and gdk/gdkx.h for chromium

Yann-le-Chevoir, 2019-03-18:
Hello,

I first tried to report this issue to GNOME developers, but I did not reach the yocto-GNOME community (https://gitlab.gnome.org/GNOME/gtk/issues/1736).
Let me try with yocto-browser community :)

With distribution Poky, branch **Thud**, machine **imx6qdlsabreauto**, recipe gtk+3 does not export X11 folder and gdk/gdkx.h. I tried to compile the recipe with machine qemuarm to compare and these files are well present with this machine (recipe-sysroot/usr/include/gtk-3.0/gdk)
1. Any idea to do come these files that Chromium is looking for?

Actually, I try to compile the **Master** branch of meta-browser/chromium with all other meta from **Thud** branch.
2. Is there a plan to release a Thud branch for this meta?

I think that my machine should use X11. Note that I also tried chromium-wayland recipe and I face other issues, related to libgbm.
3. Same as here: https://community.nxp.com/thread/485492 (libgbm and libgbm-dev already exist when I add DEPENDS to libgbm).

Thanks for your help,

 4. Note that I also added a DEPENDS to at-spi2-atk to chromium to go ahead.

Yann

rakuco, 2019-03-18:
> 1. Any idea to do come these files that Chromium is looking for?

Do you have any Wayland-related settings in your `local.conf`? It really sounds like you're doing a Wayland build, so GTK+ doesn't install X11-related files like `gdkx.h`.

>  2. Is there a plan to release a Thud branch for this meta?

We don't have separate branches because master should build fine with any branch going as far back as rocko at the moment.

> Note that I also tried chromium-wayland recipe and I face other issues, related to libgbm.
> 3. Same as here: https://community.nxp.com/thread/485492 (libgbm and libgbm-dev already exist when I add DEPENDS to libgbm).

That's another issue specific to the chromium-ozone-wayland recipe; if you're doing a Wayland build you'll need to get this fixed to get the recipe to build.

> 4. Note that I also added a DEPENDS to at-spi2-atk to chromium to go ahead.

This is already done in `chromium-gn.inc`. That looks like another indication that you're doing a Wayland-based build.

Yann-le-Chevoir, 2019-03-19:
Hello,
Thanks for quick reply!

1. You are right! Before to ask, I checked that X11 was well here in my DISTRO_FEATURES, but I did not check that Wayland was not here... I think it comes by default with poky conf.
Actually, I have to do a DISTRO_FEATURES_remove = "wayland" in my local.conf.
Chromium is still building but I confirm the problem with gtk is solved.
Note that I also had to do a "ulimit -n 4096" to go further (https://github.com/OSSystems/meta-browser/issues/194) (librenderer.a: error adding symbols: Malformed archive).

2. Ok for the Master branch, good.

4. Ok, I removed it, you were right.

Thanks again.

closed 2019-03-19 by Yann-le-Chevoir